dsmrreader/dsmr-reader #634: Verbinden met HASS.io Mosquitto MQTT broker vanuit DSMR in docker

JBS5, 2019-05-03:
Binnen mijn Home Assistant (Hass.io) 0.92.2 omgeving draai ik de [Mosquitto broker add-on](https://www.home-assistant.io/addons/mosquitto/), daarin heb ik meerdere logins gedefinieerd, onder andere eentje voor DSMR-Reader.

DSMR Reader v2.0.2 draai in Docker, specifiek deze https://github.com/xirixiz/dsmr-reader-docker opgestart via de docker-compose: https://github.com/xirixiz/dsmr-reader-docker/blob/master/docker-compose.yaml

Via Django-beheer heb ik de broker/verbinding-configuratie ingesteld:

[image]

Het lijkt er echter op dat er geen daadwerkelijke verbinding met de MQTT broker wordt gemaakt, ik zie in de log van de Mosquitto broker namelijk geen nieuwe verbinding op basis van de username 'dsmr' voorbij komen en de topics blijven leeg.

Onder 'MQTT: Uitgaande berichten' zie ik:

[image]

Iemand eenzelfde omgeving waarbij dit wel werkt, of wellicht iemand die weet waar dit probleem kan zitten?

dennissiemensma, 2019-05-03:
* Blijven de berichten in `MQTT: Uitgaande berichten` staan?
* Draait het `dsmr_mqtt` proces in Supervisor? (als sudo/root: `sudo supervisorctl status`)

JBS5, 2019-05-03:
Die indruk heb ik wel:
[image]

Aangezien ik deze docker https://github.com/xirixiz/dsmr-reader-docker gebruik, vermoed ik dat het checken of het dsmr_mqttt proces draait anders moet?

dennissiemensma, 2019-05-03:
Wellicht is het dan gewoon `docker exec -t supervisorctl status` of `docker exec -t sudo supervisorctl status`?

JBS5, 2019-05-03:
Vond het 'docker top' commando https://docs.docker.com/engine/reference/commandline/container_top/
Dat geeft voor de dsmr docker:

[image]

Heb jij hier wat aan?

dennissiemensma, 2019-05-03:
Ik zie dat het proces er wel is, dan is het de vraag waarom het niets doet. Kun je proberen in de logfile te kijken?

Wellicht werkt dit:
```
docker exec -it sudo tail -f /var/log/supervisor/dsmr_mqtt.log
```

En anders op een andere manier proberen om `/var/log/supervisor/dsmr_mqtt.log` te bekijken.

JBS5, 2019-05-04:
Het is me nog niet gelukt om `dsmr_mqtt.log` om te vinden.
Hieronder een screenshot met resultaat van `docker exec -t -i dsmr /bin/bash` gevolgd door het tonen van de inhoud van de mappen dsmr_mqtt en logs.

[image]

Inhoud van /var/log:

[image]

Heb je wellilcht nog een tip waar ik de gevraagde log zou kunnen vinden?

dennissiemensma, 2019-05-04:
Het lijkt erop dat de aftakking van Xirixiz geen logfiles heeft in de Supervisor config. 

Mogelijk worden alle logs naar één locatie gestuurd, maar ik raad je aan om dan nu alleen voor `dsmr_mqtt` de logs aan te zetten.

* https://github.com/xirixiz/dsmr-reader-docker/blob/master/app/supervisord.ini#L48
```
### xirixiz

[program:dsmr_mqtt]
command=/bin/nice -n 15 python3 -u /dsmr/manage.py dsmr_mqtt
pidfile=/var/tmp/dsmrreader--%(program_name)s.pid
autostart=true
autorestart=true
startsecs=1
startretries=100
stopwaitsecs=5
```

* https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmrreader/provisioning/supervisor/dsmr-reader.conf#L61
```
### dsmr-reader

[program:dsmr_mqtt]
environment=PYTHONUNBUFFERED=1
command=/usr/bin/nice -n 15 /home/dsmr/.virtualenvs/dsmrreader/bin/python3 -u /home/dsmr/dsmr-reader/manage.py dsmr_mqtt
directory=/home/dsmr/dsmr-reader/
user=dsmr
group=dsmr
autostart=true
autorestart=true
startsecs=1
startretries=100
stopwaitsecs=5

redirect_stderr=true
stdout_logfile=/var/log/supervisor/%(program_name)s.log
stdout_logfile_maxbytes=5MB
stdout_logfile_backups=5
```

Waarschijnlijk ben je er al als je de onderste paar regels in je lokale kopie zet. Gevolgd door het reloaden binnen de box:
```
supervisorctl reread
supervisorctl update
```

JBS5, 2019-05-05:
Ben nog iets verder gedoken in 't feit dat er überhaupt geen verbinding werd opgezet, en ook een issue aangemaakt op de Github van xirixiz https://github.com/xirixiz/dsmr-reader-docker/issues/52.
Opgelost, docker network aangelegenheid.

Dank voor het meedenken @dennissiemensma !

closed 2019-05-05 by JBS5

dennissiemensma, 2019-05-05:
Fijn om te horen, dank voor je terugkoppeling

heystge, 2020-03-30:
Ik heb Hass.io draaien op de ene Raspberry Pi en DSRM-reader op een andere Pi.
DSMR laat zijn eigen data goed zien in het dashboard.
Ik heb de MQTT functie aangezet en het lijkt erop de dat de connectie ook werkt.
In Home Assistant zie ik in de logging ook data binnenkomen.....

bijv:
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity2
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity1_returned
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity2_returned
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity_merged
2020-03-29 22:28:44 DEBUG (MainThread) [homeassistant.components.mqtt] Subscribing to dsmr/day-consumption/electricity_returned_merged

Ik heb echter nu al alle mogelijke configuraties geprobeerd in de YAML file, echter op geen enkele wijze krijg ik data te zien. 
sensor:
  - platform: dsmr_reader

maakt netjes alle sensoren aan, deze zie ik ook bij 'toestanden' in de lijst staan, echter allemaal met 'unknown' als binnenkomende data: 
sensor.dsmr_day_consumption_electricity2 | **unknown** | unit_of_measurement: kWh friendly_name: High tariff usage icon: mdi:counter

Welke essentiele stap lijk ik nu te missen om de data netjes op een LoveLace pagina te kunnen afbeelden?

-- | -- | --