dsmrreader/dsmr-reader #636: Unable to connect via ip and Home Assistant

passie, 2019-05-11:
* DSMR-reader versie v2.0.2
* Hardware Raspberry Pi 3 Model B Rev 1.2
* DSMR-protocol v5
* Home Assistant 0.92.2

### Issue
Ik probeer DSMR uit te lezen via de DSMR component op de HA site.

In mijn config heb ik het volgende staan.
```
- platform: dsmr
    host: 192.169.0.143
    port: 2001
    dsmr_version: 5
```

Als ik een telnet doe naar het ip adres krijg ik de volgende berichten terug.

```
1-3:0.2.8(50)
8)
1-0:1.8.1(000291.654*kWh)
1-0:1.8.2(000268.872*kWh)
1-0:2.8.1(000322.203*kWh)
1-0:2.8.2(000629.022*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.328*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00016)
0-0:96.7.9(00003)
1-0:99.97.0(2)(0-0:96.7.19)(180719112926S)(0000000963*s)(180719113458S)(0000000324*s)
1-0:32.32.0(00009)
1-0:32.36.0(00005)
0-0:96.13.0()
1-0:32.7.0(237.3*V)
1-0:31.7.0(001*A)
1-0:21.7.0(00.328*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1.0(003)
0-1:96.1.0(4730303339303031383432383335333138)
0-1:24.2.1(190511232003S)(00285.813*m3)
!4F57
F0000443345905
```
Voor ser2net gebruik ik de volgende oneliner.
```
ser2net -C 2001 raw 600 /dev/ttyUSB0 115200 8DATABITS:NONE:1STOPBIT
```

Echter in Home Assistant krijg ik de volgende melding te zien en lukt het mij niet om via ser2net de data uit te lezen.
```
Error connecting to DSMR
Traceback (most recent call last):
  File "/usr/local/lib/python3.7/site-packages/homeassistant/components/dsmr/sensor.py", line 199, in connect_and_reconnect
    reader_factory())
  File "uvloop/loop.pyx", line 1904, in create_connection
  File "uvloop/loop.pyx", line 1883, in uvloop.loop.Loop.create_connection
TimeoutError: [Errno 110] Operation timed out
```
Via mqtt uitlezen werkt alles perfect na wat tweaken.

Enige suggesties voor het  uitlezen via ser2net?
Alvast bedankt.

dennissiemensma, 2019-05-12:
Bedankt voor je vraag. Ik zit helaas niet in de HA/ser2net-materie, wellicht kun je deze vraag stellen op een HA-forum. Het lijkt er in ieder geval op dat de verbinding geweigerd wordt en het eerst wat in mij opkomt is een configuratiefout, gezien het je handmatig wel lukt. 

Kun je dubbelchecken of je hetzelfde IP/poort hebt ingesteld als waarmee je getest hebt qua `telnet`?

passie, 2019-05-13:
I have created a topic on the HA forum.

https://community.home-assistant.io/t/dsmr-via-tcp/116503

dennissiemensma, 2019-06-12:
@passie ben je hier zelf nog verder mee gekomen imiddels?

passie, 2019-06-13:
@dennissiemensma helaas heb ik geen antwoord gekregen op mijn vraag op het forum van HA. Mogelijk ben ik de enige met dit probleem. 
Als je het vervelend vindt wil ik best dit issue sluiten, omdat zoals je zelf al eerder aangaf dit waarschijnlijk te maken heeft met HA en niet met DSMR. 
Laat maar weten wat je wilt met het issue.

dennissiemensma, 2019-06-13:
Ik vind het niet vervelend hoor, maar als het geen issue van dit project is en ik je er niet mee verder kan helpen, houdt het helaas op. Ik heb alleen wat tips voor je, gezien het iets kleins lijkt:

* Had je nog gekeken of de poort wel klopt?
* Doe je de telnet vanaf dezelfde machine en user als waarop HA draait?
* Je zou nog op het Tweakers forum om hulp kunnen vragen, daar zitten aardig wat experts op dit gebied.

passie, 2019-06-16:
Je hebt gelijk ook. Helaas ben ik niet verder geraakt met de oplossing. 
Voor nu gebruik ik MQTT om de meter uit te lezen via DSMR, Dit werkt prima maar toch.....
Bedankt voor je hulp en de tips.

closed 2019-06-16 by passie

frankhommers, 2020-04-12:
Late reactie, maar volgens mij zit er een typo in je ip address van home assistant.

192.169.0.143 moet waarschijnlijk zijn 192.168.0.143