dsmrreader/dsmr-reader #637: Live gas gebruik per MQTT

joep2000, 2019-05-13:
Zoals verzocht in issue #562 hierbij een nieuwe.

De huidige meters (in ieder geval de mijne) elke 5 minuten de gasstand door. Met deze frequentie kan je al aardige analyses maken, wat ik dan ook graag zou doen.
Graag zou ik een het live gasverbruik willen (dus niet het cumulatief) willen ontvangen via MQTT. Dennis, zou je dit kunnen toevoegen in toekomstige release?
dank!

megapearl, 2019-08-25:
Ik zocht ook al een tijdje naar deze feature, zie hem ook graag in dsmr-reader verschijnen, goed bezig in ieder geval.

kaosmagix, 2019-10-05:
+1 I would love to have this information in MQTT as well

dennissiemensma, 2019-10-09:
Zie de screenshot:

[image]

Is deze informatie afdoende straks? JSON en/of gesplitst per topic.

dennissiemensma, 2019-10-09:
Gaat mee in volgende release.

closed 2019-10-09 by dennissiemensma

dennissiemensma, 2019-10-10:
Gereleased in [v2.7](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.7.0)

dataegg, 2019-10-13:
Hallo, ik vind dit ook interessant en heb het net aangezet. Is currently_delivered in m3 per uur ofzo of begrijp ik het verkeerd?

dennissiemensma, 2019-10-13:
Het geeft (bij gas) het verschil aan t.o.v. de vorige gasmeting. De tijd is variabel en hangt af van je slimme meter. 

* Voor DSMR v4 meters komt er 1x per uur een nieuwe gasmeting binnen en is het dus per uur. 
* Voor DSMR v5 meters is die interval veel korter (ik weet niet precies hoelang) en kun je kijken naar de waarde in ``read_at`` om te zien op welk tijdstip het gemeten is (en vergelijken met de vorige).

dataegg, 2019-10-13:
Ok mooi. Ik heb een DSMR v5 meter en ik zie dat er exact om de 5 minuten een bericht is:

mosquitto_sub -h localhost -p 1883 -t dsmr/consumption/gas/json
{"read_at": "2019-10-13T11:40:00+02:00", "delivered": "141.343", "currently_delivered": "0.000"}
{"read_at": "2019-10-13T11:45:00+02:00", "delivered": "141.343", "currently_delivered": "0.000"}
{"read_at": "2019-10-13T11:50:00+02:00", "delivered": "141.362", "currently_delivered": "0.019"}
{"read_at": "2019-10-13T11:55:00+02:00", "delivered": "141.413", "currently_delivered": "0.051"}

Om een indicatie in homeassistant weer te geven zou ik currently_delivered x 12 moeten doen om een verbruiksindicatie m3/h te krijgen

Giel538, 2019-10-13:
Hoi Dennis,

Top dat de actuele gaswaarde en de spanningsmeting er inmiddels inzitten. Ook de MQTT berichten retained maken voegt echt iets toe. Ik ga hem vanaf nu gebruiken als logger. Via MQTT stuur ik de waarden door naar Home Assistant. Tot nu toe las ik het hier direct in en logte ik het om met Grafana te kijken.

Nog 2 kleine bugs. 

1) Je dsmr/consumption/gas/read_at (split topic) waarde laat alleen het jaartal zien. Deze geeft alleen 2019 door. Als je de gas consumption als json uitleest in MQTT staat bij read_at wel de complete date/time. 

2) een kleinigheidje: Bij Telegram: Split topic en ook bij alle andere split topics staan standaard alle metingen aan. Bij Telegram moet je alleen de 3 spanningen eerst zelf toevoegen.

dennissiemensma, 2019-10-13:
Bedankt voor je terugkoppeling. Krijg je de tijden wel goed door in de split-topic-variant voor telegrammen?

```
timestamp = dsmr/reading/timestamp
extra_device_timestamp = dsmr/reading/extra_device_timestamp
```

Wat betreft voltages. Ik zie dat ze inderdaad nog niet in de standaard config staan. Ik zal ze later meenemen.

Giel538, 2019-10-14:
> Bedankt voor je terugkoppeling. Krijg je de tijden wel goed door in de split-topic-variant voor telegrammen?
>
> ```
> timestamp = dsmr/reading/timestamp
> extra_device_timestamp = [...]

Sorry het was mijn fout. Ik had payload decoding aanstaan op JSON, daardoor snapte die het formaat niet :) Alles werkt dus correct.