dsmrreader/dsmr-reader #639: Toegang tot Postgres database

Kruijd, 2019-05-14:
hoi Dennis,
Complimenten voor dsmrreader, het draait als een zonnetje op mijn Raspbery Pi 3b+
Nu zou ik graag toegang tot de postgres database willen hebben (read only) om de data bijvoorbeeld in een ander tool te visualiseren. 
Is dat mogelijk of is dat een slecht idee?
Ik heb al geprobeerd om met psql aan te melden maar krijg de melding dat er geen password is voor user dsmrreader en alsnog een wachtwoord erop zetten zal de applicatie niet leuk vinden.

Kan je me adviseren?
Alvast bedankt.
Jan

dennissiemensma, 2019-05-14:
Er staat standaard wel degelijk een (simpel) wachtwoord op. De inloggevens kun je hier vinden: https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmrreader/provisioning/django/postgresql.py#L6

Echter, de database is standaard dusdanig geinstalleerd dat je er **niet buiten de Pi bij kan** (wegens veiligheid). Als je Pi alleen in je thuisnetwerk draait en je al het verkeer daarop vertrouwt, kun je het volgende doen:

1) Open config met `sudo vi /etc/postgresql/9.6/main/postgresql.conf` (versienummer kan verschillen) en zoek:
```
#listen_addresses = 'localhost'		# what IP address(es) to listen on;
					# comma-separated list of addresses;
					# defaults to 'localhost'; use '*' for all
					# (change requires restart)
```
En maak er dit van:
```
listen_addresses = '*'
```
Sla het bestand op.

2) Open daarna `sudo vi /etc/postgresql/9.6/main/pg_hba.conf` en voeg onderaan een regel toe:
```
host    dsmrreader      all             0.0.0.0/0                md5
```

3) Sla ook dit bestand op en herstart nu de database:
```
sudo systemctl restart postgresql
```

4) Controleer of je geen typefouten hebt gemaakt en de DB nu goed draait:
```
sudo systemctl status postgresql
```

Je kunt ook DSMR-reader zelf even checken.

Als het goed is kun je nu binnen je netwerk verbinden naar de database met een client, zoals `pgAdmin`.

dennissiemensma, 2019-05-14:
Je kunt overigens ook de API gebruiken als je de data in een andere tool wil laden. Dat is wat robuuster dan via SQL.

Kruijd, 2019-05-15:
Bedankt voor je uitleg Dennis, ik ga het proberen.

closed 2019-05-15 by Kruijd

Kruijd, 2019-05-15:
Beste Dennis, Ik heb de aanpassingen gemaakt en het is nu inderdaad gelukt. Alles draait ook nog steeds zoals verwacht. Bedankt voor de support 😄