dsmrreader/dsmr-reader #640: Foutmelding in webbrowser

peterbakker82, 2019-05-19:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y` 
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3 B+
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Er doet zich een foutmelding voor als ik de gegevens probeer op te vragen in de webbrowser. Ik ben zelf geen programmeur en heb zo goed mogelijk de stappen van de installatie doorlopen. Dit verliep succesvol op twee punten na: 1. bij het downloaden van de applicatie was de internetverbinding verbroken. Nadat deze hersteld was, heb ik de applicatie nogmaals geinstalleerd. 2.  Bij de installatie van virtualinv start de map niet automatisch op, ondanks ik de stappen volgens de instructie doorlopen heb.
De volgende foutmelding krijg ik te zien als ik het ipadres in een webbrowser open:

Server Error
Sorry, something unexpected happened.
Exception:

ProgrammingError: relation "dsmr_consumption_electricityconsumption" does not exist
LINE 1: SELECT (1) AS "a" FROM "dsmr_consumption_electricityconsumpt...
                               ^

Traceback:

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 97, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/dashboard.py", line 29, in get_context_data
    context_data['capabilities'] = dsmr_backend.services.get_capabilities()

  File "/home/dsmr/dsmr-reader/dsmr_backend/services.py", line 32, in get_capabilities
    'electricity': ElectricityConsumption.objects.exists(),

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 766, in exists
    return self.query.has_results(using=self.db)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/query.py", line 522, in has_results
    return compiler.has_results()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1067, in has_results
    return bool(self.execute_sql(SINGLE))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1097, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 67, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 76, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

Alvast heel erg bedank!

Met vriendelijke groet,

Peter Bakker

dennissiemensma, 2019-05-19:
Het lijkt erop dat de database migraties niet goed zijn uitgevoerd. Je zou het nogmaals kunnen proberen vanaf stap 7: https://dsmr-reader.readthedocs.io/nl/latest/installation/explained.html#application-configuration-setup

peterbakker82, 2019-05-19:
Heel erg bedankt. Na nogmaals vanaf stap 7 alles doorlopen te hebben, is het operationeel

closed 2019-05-20 by dennissiemensma