dsmrreader/dsmr-reader #641: Hoe te update uit te voeren i.c.m. docker?

bikeymouse, 2019-05-19:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

bikeymouse, 2019-05-19:
Hi, ik gebruik DSMR-reader via de Docker image van [xirixiz](url=https://github.com/xirixiz/dsmr-reader-docker/blob/master/update_images.sh). Momenteel gebruik ik de image met DSMR 1.28 op een RPI2. E.a. gestart met Docker-compose met koppeling naar een volume voor behoud van de data, en ik gebruik Portainer voor het Docker-beheer.

Ik zou graag upgraden, maar lijkt me niet verstandig om gewoon de nieuwste Docker-image op te starten aangezien de database mogelijk van de nieuwe versie mogelijk niet backwards compatible is.

Hoe kan ik dit het beste aanpakken? Eerst op een of andere manier zorgen dat deploy.sh binnen de huidige Docker-container wordt uitgevoerd?

(excuus als dit niet de goede plek is voor dit soort vragen en mijn beperkte kennis van Docker, maar ik kon geen andere zinvolle plek vinden).

dennissiemensma, 2019-05-19:
Ik denk dat je dit het beste bij xirixiz zelf kunt vragen. Als het goed is kun je daar gewoon een issue aanmaken en helpt hij je verder. In deze specifieke upgrade heb je namelijk ook minimaal Python 3.5 nodig binnen Docker.

bikeymouse, 2019-05-20:
Hoi, dank voor je antwoord, ik heb de vraag ook aan xirixiz gesteld. Hij gaf aan in praktijk gewoon altijd de nieuwste Docker image te starten wat tot nu toe altijd goed gaat.
(De upgrade naar de nieuwste Python zit gewoon in de nieuwe Docker image, dus dat zal het probleem niet zijn. )

Waar ik echter bang voor ben is dat als je een nieuwe versie van DSMR-reader opstart en laat verbinden met de oude database dit tot problemen kan leiden als de nieuwe DSMR-reader niet backwards compatible is met de vorige database.

Dat probleem heb je natuurlijk ook met gewoon upgraden, of wordt dit afgehandeld via het deploy.sh script wat je volgens de documentatie voor een reguliere update (buiten Docker om) moet gebruiken?

Is 2.02 overigens backwards compatible met de database van 1.28?

dennissiemensma, 2019-05-20:
De database is compatible, in de zin dat het deploy-script altijd een migratie moet uitvoeren. Hoe dat exact werkt in de image van xirixiz weet ik dus niet. Het is wel nodig om de database migraties uit te voeren.

bikeymouse, 2019-05-22:
Dank, ik ben er met xirxiz uitgekomen, zijn Docker-image voert ook de activiteiten van het deploy script uit, dus dat gaat altijd goed.

closed 2019-05-22 by bikeymouse