dsmrreader/dsmr-reader #642: Installatie: Set password for database user:

Robert126, 2019-05-21:
### Jouw omgeving
*(Indien van toepassing)*
* Probeer een nieuwe installatie uit te voeren
* Hardware *(RaspberryPi 3)*: Nieuwste Raspberian installatie vandaag uit gevoerd en ge-update

- Set password for database user:
sudo sudo -u postgres psql -c "alter user dsmrreader with password 'd
Krijg geen vraag om password in te voeren.
melding:
sudo sudo -u postgres psql -c
/usr/lib/postgresql/9.6/bin/psql: option requires an argument -- 'c'
Try "psql --help" for more information.

De voorgaande stappen zijn uitgevoerd zonder problemen.

dennissiemensma, 2019-05-21:
Kan het zijn dat je niet het volledige command hebt gebruikt? Let erop dat je alle commado's volledig overneemt:

```
sudo sudo -u postgres psql -c "alter user dsmrreader with password 'dsmrreader';"
```

Je kunt het makkelijk kopieren door **drie keer** met je muis op een command te klikken. Je browser selecteert dan de hele regel.

Robert126, 2019-05-27:
Hi Dennis,

Dank voor je mail.
Het heeft even geduurd. Ik heb meerdere malen een nieuwe installatie gedaan. Nu is het gelukt!

Ik heb nog wel een vraag:
- Ik krijg de DSMR Reader dash board te zien
- Om in te loggen Configuration heb ik een username/ password nodig

Tijdens de installatie stap 
# Create application user
https://dsmr-reader.readthedocs.io/en/latest/installation/quick.html <https://dsmr-reader.readthedocs.io/en/latest/installation/quick.html>
Heb ik een password ingevoerd.
Username: dsmrreader in combinatie met door mij aangemaakte password kan ik niet inloggen??

Enig advies?

Alvast dank! Groet,
Robert

Robert126, 2019-05-27:
Hi Dennis,

Ik heb het opgelost!

Username: admin en password werken nu.

Hartelijk dank voor je hulp!

Hartelijke groet,
Robert

dennissiemensma, 2019-05-27:
Fijn om te horen, bedankt voor je terugkoppeling!

closed 2019-05-27 by dennissiemensma