dsmrreader/dsmr-reader #644: Daystatistics gebruikt UTC overige local times?

bikeymouse, 2019-05-22:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

bikeymouse, 2019-05-22:
Hi,
Ik lees de databasegegevens van DSMR ook uit met Grafana om dat te combineren met andere meter-gegevens in overzichten. 
Ik krijg daarbij de indruk dat in alle tabellen de meetgegevens worden weggeschreven o.b.v. de Unix local-time (=UTC+2 voor Amsterdam), behalve voor de tabel dsmr_stats_daystatistics. 
Deze lijken te worden opgeslagen op basis van de UTC=00:00 uur van de dag, wat dus overeenkomt met 02:00 uur Amsterdam-tijd. 
Klopt dat en is dat op "by design", of vergis ik mij?

dennissiemensma, 2019-05-22:
Voor zover ik weet bevat `dsmr_stats_daystatistics` geen tijdstippen en daarmee ook geen tijdzones, alleen datums. Bedoel je wellicht `dsmr_stats_hourstatistics`?

----

Alle tijdstippen die weggeschreven worden bevatten een tijdzone en voor de database maakt het niet uit welke variant het is, zolang ze relatief maar kloppen.  Het is namelijk aan de applicatie om pas de juiste tijdzone te gebruiken wanneer de gegevens getoond of verwerkt worden.

Je kunt het beste in de verbinding naar de DB opgeven in welke tijdzone je de gegevens wilt hebben, de database zet ze dan om naar die tijdzone.

Voorbeeld over hetzelfde record, zonder voorkeur op te geven:
```
dsmrreader=# select id, hour_start from dsmr_stats_hourstatistics limit 1;
  id  |       hour_start       
------+------------------------
 4974 | 2016-03-08 00:00:00+00
```
En nu in de gewenste tijdzone:

```
dsmrreader=# set time zone 'Europe/Amsterdam' ;
SET
```
```
dsmrreader=# select id, hour_start from dsmr_stats_hourstatistics limit 1;
  id  |       hour_start       
------+------------------------
 4974 | 2016-03-08 01:00:00+01
```

Ben je hier voldoende mee geholpen?

dennissiemensma, 2019-05-22:
Zie trouwens ook #227

bikeymouse, 2019-05-28:
Hartelijk dank voor je uitgebreide antwoord. Moest er even tijd voor nemen om het te checken.
Het lijkt er op dat e.a. door Grafana wordt gedaan.
Als ik dit in Psql doe:
```
SELECT
date_trunc('day', (read_at AT TIME ZONE 'Europe/Amsterdam')) AS Time,
max(delivered_1) - min(delivered_1) AS "Kwh Dal verbruk",
max(delivered_2) - min(delivered_2) AS "Kwh Piek verbruik"
FROM dsmr_consumption_electricityconsumption
GROUP BY 1
ORDER BY 1 DESC
LIMIT 10;
```
 ```
time         | Kwh Dal verbruk | Kwh Piek verbruik
---------------------+-----------------+-------------------
 2019-05-28 00:00:00 |           2.790 |             3.687
 2019-05-27 00:00:00 |           2.922 |             5.401
etc.
`
```
En als ik deze doe:
```
SELECT
date_trunc('day', read_at) AS time,
max(delivered_1) - min(delivered_1) AS "Kwh Dal verbruik",
max(delivered_2) - min(delivered_2) AS "Kwh Piek verbruik"
FROM dsmr_consumption_electricityconsumption
GROUP BY 1
ORDER BY 1 DESC
LIMIT 10
```
```
          time          | Kwh Dal verbruik | Kwh Piek verbruik
------------------------+------------------+-------------------
 2019-05-28 00:00:00+02 |            2.790 |             4.034
 2019-05-27 00:00:00+02 |            2.922 |             5.401
```

Ziet er goed uit en verbruikscijfers kloppen ook met zoals DSMR-reader die in dsmr_stats_daystatistics heeft staan. 
Echter in Grafana wordt dit weergegeven als timestamps zoals:
`2019-05-28 02:00:00|           2.790 |             3.687`

Dus ook in de grafieken zie je dan steeds de tijd op 02:00 uur staan. 
Begrijp niet helemaal hoe dat zit, maar het zit in ieder geval niet fout in de DSMR database ;-)

dennissiemensma, 2019-05-28:
Bedankt voor je update en het delen van je bevindingen. Wellicht kun je dan in Grafana een tijdzone of iets opgeven of meegeven aan de DB-verbinding naar DSMR-reader.

Mocht je desondanks nog meer informatie nodig hebben, dan hoor ik het graag.

closed 2019-05-28 by dennissiemensma