dsmrreader/dsmr-reader #645: Datatlogger setup with virtual port

marceltrapman, 2019-05-23:
Setup
* DSMR-reader v2.1
* RaspberryPi 3B+
* DSMR-protocol v2.2

I have a RaspberryPi 3B+ with a datalogger cable attached.
I have DSMR-reader installed as well as ser2net.

I have Home Assistant with the DSMR plugin setup on an Udoo(single board pc) connecting to the RaspberryPi using a remote connection.

Question
Because the serial port is used by ser2net it is not available for DSMR-reader anymore so I created a virtual port using socat. Problem is that, whatever I do to change the user and/or group I only see one or two telegrams come in and back are the settings to root:root.

I don't know how to solve this so what I would like to know is if there is an alternative way to achieve what I want. Or would it be possible to add an option to connect to the tcp/ip connection setting up the ip and port?

Thanks in advance,

Marcel

dennissiemensma, 2019-05-24:
Bedankt voor je vraag. 

Je zou nog even kunnen kijken in issues: #598, #599, #310 en #521. Mogelijk staan daar nog wat tips voor je in.

drvdijk, 2019-05-26:
I'm just getting started with this `dsmr-reader` project, and trying to figure out how I want to eventually hook it all up together. @marceltrapman rather interested in how you set up `ser2net` and even more so the virtual port with `socat`.

My current setup is as follows:
* Home Assistant just being home assistant, no dsrm integration at all. Runs in docker on a Synology NAS.
* Raspberry Pi 3B+ reading serial port and putting data in an InfluxDB instance, also running on the pi.
This is not ideal for multiple reasons. I do have history in the InfluxDB, but have no nice integration. Worse, the script it runs to read & parse the telegrams doesn't use the timestamp in the telegram itself for gas measurements, but writes a value for every telegram. Both the Home Assistant dsrm component as well as this project seem to handle that better (at least, I hope).

What I had done before discovering this project, is indeed integrating the dsrm home assistant component, reading from the network. However, I have only used `socat` to make this work:

```
socat tcp-listen:8000,reuseaddr,fork file:/dev/ttyUSB0,nonblock,waitlock=/var/run/tty0.lock,b115200,raw,echo=0
```

With that, Home Assistant correctly pics up the smart meter messages. The idea was to have Home Assistant then write datapoints into a new InfluxDB instance, but at that point I found this project, which looks pretty good :)

However, I too would like to _also_ have Home Assistant display the meters in its UI, and possibly still have it write to an InfluxDB as backup, while still actually having _this_ project running on the Pi and collecting all data.

So I'm too going to struggle further with `socat`, but hope you can share a bit more information on your setup @marceltrapman :)

marceltrapman, 2019-05-28:
OK, I have come a bit further.
Upon first start of the service (or a reboot) I receive in dsmr-reader exactly one telegram and after that the fun is over again :(
Home Assistant works like a charm.

The changes I made are the number of connections allowed for ser2net (3 was just an arbitrairy number, I need only 2).
In the service I added the User and Group...

This is my setup:
**/etc/ser2net.conf** 
```
2001:raw:600:/dev/ttyUSB0:9600 EVEN 1STOPBIT 7DATABITS XONXOFF LOCAL -RTSCTS max-connections=3
```

**/etc/systemd/system**
```
[Unit]
Description=Virtual DSMR port

[Service]
Type=simple
ExecStart=/usr/bin/socat -U pty,link=/dev/dsmr-port,raw,group-late=dialout,mode=660 tcp:127.0.0.1:2001&
User=root
Group=dialout
Restart=always
RestartSec=10

[Install]
WantedBy=multi-user.target
Alias=dsmr-port.service
```

I am now going to try pushing data as per this [suggestion](https://github.com/dennissiemensma/dsmr-reader/issues/598).

marceltrapman, 2019-05-28:
So, I think I got it running.

I moved from the virtual port to pushing telegrams through the api.
I changed the code of the file '/home/dsmr/dsmr_datalogger_api_client.py' to work with ser2net.
Not the best code that I ever made but you are free to improve it.
Otherwise I will see when I can clean it up.

```
import socket
from time import sleep

import requests

API_SERVER = ('http://HOST-OR-IP-ONE/api/v1/datalogger/dsmrreading', 'APIKEY')
HOST = ('HOST-OR-IP-ONE', 2001)

failed = 0

def main():
	api_url, api_key = API_SERVER

	print( 'Starting...' )

	for telegram in read_telegram():
		# print( telegram )
		# print( 'Sending telegram to:', api_url )

		send_telegram( telegram, api_url, api_key )

		sleep( 1 )

def read_telegram():
	global failed

	host, port = HOST

	server = None
	telegram_start_seen = False
	telegram = ''

	try:
		server = socket.socket()

		server.settimeout( 60 )
		server.connect( (host, port) )
	except socket.timeout as error:
		stop( server, str(error) )
	except IOError as error:
		stop( server, str(error) )

	# Just keep fetching data until we got what we were looking for.
	while True:
		data = None

		try:
			data = server.recv( 1024 )
		except socket.timeout:
			pass
		except IOError:
			pass

		if not data or data is None:
			failed += 1

			if failed == 10:
				stop( server, 'Data is none!' )

			continue

		try:
			# Make sure weird characters are converted properly.
			data = str( data, 'utf-8' )
		except TypeError:
			pass

		for index in range( 0, len( data ) ):
			character = data[index]

			if '/' in character:
				# Start the flow.
				telegram_start_seen = True
				telegram = ''

			telegram += data[index]

			if '!' in character and telegram_start_seen:
				yield telegram

				# Reset the flow again.
				telegram_start_seen = False
				telegram = ''

def send_telegram( telegram, api_url, api_key ):
	global failed

	# Register telegram by simply sending it to the application with a POST request.
	response = requests.post(
		api_url,
		headers={'Authorization': 'Token ' + api_key},
		data={'telegram': telegram},
		)

	# Old versions of DSMR-reader return 200, new ones 201.
	if response.status_code not in (200, 201):
		failed += 1

		if failed == 10:
			exit()

		# Or you will find the error (hint) in the reponse body on failure.
		print( 'API error: {}'.format( response.text ) )

def stop( server, error ):
	if server:
		server.close()

	print( 'Socket connection failed ' + str( failed ) + ' time(s) with error: ' + error )

	exit()

if __name__ == '__main__':
	main()

```

dennissiemensma, 2019-05-28:
Bedankt voor je updates. Als het script voor je werkt met je aanpassingen, zou ik het vooral lekker zo laten eerlijk gezegd.

dennissiemensma, 2019-06-12:
Ik ga er vanuit dat jouw bovenstaande script het voor je heeft opgelost. Mocht je desondanks nog hulp nodig hebben, dan hoor ik het graag.

closed 2019-06-12 by dennissiemensma