dsmrreader/dsmr-reader #648: Logger blijft actief ondanks dat deze niet gebruikt wordt / is uitgezet

MarsWarrior, 2019-05-30:
### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.1`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 2 met logger, NUC met DSMR.`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4`

### Bug, wens of iets anders
* Ik heb de logger los op mijn RPI draaien die de data via http naar dsmr doorzet op een NUC.
* De logger op de NUC blijft echter ook draaien, ondanks dat ik P1 polling uit heb staan

```
2019-05-30 18:37:17,065 INFO exited: dsmr_datalogger (exit status 1; not expected),
2019-05-30 18:37:15,904 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs),
2019-05-30 18:37:14,901 INFO spawned: 'dsmr_datalogger' with pid 2451,
2019-05-30 18:37:13,897 INFO exited: dsmr_datalogger (exit status 1; not expected),
2019-05-30 18:37:11,715 INFO spawned: 'dsmr_datalogger' with pid 2450,
2019-05-30 18:37:10,710 INFO exited: dsmr_datalogger (exit status 1; not expected),
2019-05-30 18:37:09,537 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs),
2019-05-30 18:37:08,532 INFO spawned: 'dsmr_datalogger' with pid 2449,
2019-05-30 18:37:06,357 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs),
2019-05-30 18:37:05,355 INFO spawned: 'dsmr_datalogger' with pid 2448,
2019-05-30 18:37:04,349 INFO exited: dsmr_datalogger (exit status 1; not expected),
2019-05-30 18:37:03,173 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs),
2019-05-30 18:37:07,526 INFO exited: dsmr_datalogger (exit status 1; not expected),
2019-05-30 18:37:13,232 INFO success: dsmr_datalogger entered RUNNING state, process has stayed up for > than 1 seconds (startsecs)
```

dennissiemensma, 2019-06-03:
Ja dat 'klopt', in de zin van dat het vinkje niets doet met het proces zelf. Het is eigenlijk een instelling die ik ooit bedacht had voor migraties, maar die in de praktijk nooit uit staat (van mijn kant).

Als je het datalogger-proces wilt uitschakelen, dan mag je `/etc/supervisor/conf.d/dsmr-reader.conf` openen (als root) en het hele blok van `[program:dsmr_datalogger]` (inclusief die regel) uitcommenten met `#` voor elke regel.

Vervolgens kun je het doorvoeren met `sudo supervisorctl reread`, gevolgd door `sudo supervisorctl reload`. Als het goed is verdwijnt dan het datalogger-proces.

closed 2019-06-03 by dennissiemensma

MarsWarrior, 2019-08-24:
In de alpine docker omgeving is dit het bestand:
cd /etc/supervisor.d/supervisord.ini

Heb restart voor de datalogger op false gezet en daarna de docker container herstart...

dennissiemensma, 2019-08-24:
Heb je het hele blok in de config voor de `dsmr_datalogger` uitgezet, zoals hierboven aangegeven? Het proces zou dan uberhaupt niet meer moeten bestaan.

d0mmels, 2022-02-14:
Ik loop hier weer tegenaan sinds de upgrade naar v5. Alleen is er geen supervisor meer? Ik kan de config niet vinden.
Het betreft de docker container.