dsmrreader/dsmr-reader #649: Wat wensen

sideriusj, 2019-06-01:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(2.1`
* Hardware *(RaspberryPi  3 
* DSMR-protocol versie van slimme meter *(v2, 

### Bug, wens of iets anders
Ik gebruik geen gas meer, graag zou ik zien dat je de keuze kan maken in DSMR reader voor alleen electriciteit.

Ik heb ook wat moeite met de termen ""verbruikt en teruggeleverd"", mooier zou het zijn als de termen ""import en export gebruikt zouden worden"", net als bij pvoutput.org. 
Als je zonnepanelen hebt is je verbruik niet gelijk aan import, maar is een optel som van import en opwek min teruglevering. Ik upload met DSMR, mijn slimme meter gegevens naar pvoutput, de opbrengst van mijn zonnepanelen upload ik ook. Het zou mooi zijn als de gegevens die in PVoutput staan, weer gebruikt kunnen worden als input voor DSMR reader.

dennissiemensma, 2019-06-03:
Bedankt voor je input. Wat betreft het uitschakelen van gas, dit is een wat diepere instelling die niet vaak gebruikt wordt, zie: https://dsmr-reader.readthedocs.io/nl/latest/settings.html#dsmrreader-disabled-capabilities

Als het goed is kun je het uitzetten met:
```
sudo su - dsmr
vi dsmrreader/settings.py

# Voer nu onderaan de file een regel toe met:
DSMRREADER_DISABLED_CAPABILITIES = ['gas']

# Sla het bestand op.

# Herlaad de applicatie met:
./reload.sh
```

Als het goed is is gas dan uitgeschakeld in de interface.

dennissiemensma, 2019-06-03:
Wat betreft de woordkeuze binnen de applicatie. Ik begrijp helemaal je punt en er is zeker wat voor te zeggen. Echter, niet iedereen gebruikt PVOutput en daarmee ook niet de termen import/export, gezien die veel technischer klinken. Daarom houd ik de termen liever consistent zoals ze al waren.

dennissiemensma, 2019-06-03:
Het importeren van gegevens die niet afgeleid zijn van een slimme meter, of simpelweg telegrammen zijn, valt helaas buiten de scope van het project. 
Dit project focust zich alleen op het uitlezen van de slimme meter, het plotten van die gegevens en je helemaal in staat te stellen die gegevens overal naar wens heen te exporteren.

closed 2019-06-03 by dennissiemensma