dsmrreader/dsmr-reader #650: Energy Used waarde te hoog in pvoutput

Merlin1201, 2019-06-04:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.1`
* Hardware *(RaspberryPi 3 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5` (denk ik)

### Bug, wens of iets anders
* vreemde energy waarde in pvoutput. energy used 177.301, deze waarde is veel te hoog, zoveel verbruik ik niet.

04/06/19 | 22:05 | 19.900kWh | 4.146kWh/kW | 0W | 0W | 0.000kW/kW | 28.9C | 137.3V | 177.301kWh | 239,324W

dennissiemensma, 2019-06-05:
Bedankt voor je melding. Ik denk dat het goed is dat je even in je database kijkt waar die waardes vandaan komen. Wellicht zie je iets vreemds:

```
sudo su - postgres
psql dsmrreader

select read_at, delivered_1, delivered_2 from dsmr_consumption_electricityconsumption where read_at between '2019-06-04 22:00:00+02:00' and '2019-06-04 22:10:00+02:00';
```

Merlin1201, 2019-06-05:
heb hier ff de output daarvan neergezet:  hopelijk kan jij er wat mee :-)
Ook vandaag zat ik al op 192.454kWh     en zoveel energie gebruik ik niet,   mogelijk lees ik het verkeerd.

mischien kun je op mijn pvoutput kijken:  https://pvoutput.org/intraday.jsp?id=74451&sid=66030

  read_at         | delivered_1 | delivered_2 
------------------------+-------------+-------------
 2019-06-04 22:00:00+02 |     770.861 |     872.763
 2019-06-04 22:01:00+02 |     770.861 |     872.772
 2019-06-04 22:02:00+02 |     770.861 |     872.781
 2019-06-04 22:03:00+02 |     770.861 |     872.789
 2019-06-04 22:04:00+02 |     770.861 |     872.798
 2019-06-04 22:05:00+02 |     770.861 |     872.807
 2019-06-04 22:06:00+02 |     770.861 |     872.816
 2019-06-04 22:07:00+02 |     770.861 |     872.824
 2019-06-04 22:08:00+02 |     770.861 |     872.833
 2019-06-04 22:09:00+02 |     770.861 |     872.842
 2019-06-04 22:10:00+02 |     770.861 |     872.850

Ook ff een gedeelte van de reading vandaag in pvoutput:

05/06/19 | 19:00 | 11.000kWh | 2.292kWh/kW | 417W | 1,200W | 0.250kW/kW | 32.1C | 301.9V | 192.454kWh | - |  
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --
05/06/19 | 18:55 | 10.900kWh | 2.271kWh/kW | 739W | 0W | 0.000kW/kW | 32.2C | 303.0V | 192.454kWh | - |  
05/06/19 | 18:50 | 10.900kWh | 2.271kWh/kW | 495W | 1,200W | 0.250kW/kW | 32.2C | 302.2V | 192.454kWh | 130,572W |  
05/06/19 | 18:45 | 10.800kWh | 2.250kWh/kW | 0W | 0W | 0.000kW/kW | 0.0C | 0.0V | 181.573kWh | 0W |  
05/06/19 | 18:40 | 10.800kWh | 2.250kWh/kW | 512W | 1,200W | 0.250kW/kW | 32.6C | 307.5V | 181.573kWh | 932W |  
05/06/19 | 18:35 | 10.700kWh | 2.229kWh/kW | 549W | 0W | 0.000kW/kW | 32.7C | 301.3V | 181.495kWh | 0W |  
05/06/19 | 18:30 | 10.700kWh | 2.229kWh/kW | 516W | 1,200W | 0.250kW/kW | 32.9C | 308.2V | 181.495kWh | 128,146W |  
05/06/19 | 18:25 | 10.600kWh | 2.208kWh/kW | 0W | 0W | 0.000kW/kW | 0.0C | 0.0V | 170.816kWh | 0W |  
05/06/19 | 18:20 | 10.600kWh | 2.208kWh/kW | 510W | 1,200W | 0.250kW/kW | 33.2C | 309.8V | 170.816kWh | 126,981W |  
05/06/19 | 18:15 | 10.500kWh | 2.188kWh/kW | 0W | 0W | 0.000kW/kW | 0.0C | 0.0V | 160.234kWh | 0W |  
05/06/19 | 18:10 | 10.500kWh | 2.188kWh/kW | 607W | 1,200W | 0.250kW/kW | 33.9C | 305.9V | 160.234kWh | 938W |  
05/06/19 | 18:05 | 10.400kWh | 2.167kWh/kW | 927W | 0W | 0.000kW/kW | 34.0C | 305.7V | 160.156kWh | 0W |  
05/06/19 | 18:00 | 10.400kWh | 2.167kWh/kW | 798W | 2,400W | 0.500kW/kW | 34.3C | 307.0V | 160.156kWh | 124,481W

Merlin1201, 2019-06-05:
Hier een voorbeeld wat het ongeveer zou moeten zijn:
Dus rond de 9.000 kWh is mijn dagelijkse verbruik

05/06/19 | 19:35 | 8.369kWh | 1.550kWh/kW | 298W | 300W | 0.056kW/kW | - | - | 9.691kWh | 507W
-- | -- | -- | -- | -- | -- | -- | -- | -- | -- | --

dennissiemensma, 2019-06-05:
Ik weet eerlijk gezegd niet heel goed waar ik naar kijk. Het helpt niet meer dat komma's en punten als decimalen door elkaar heen gebruikt worden in PVOutput.

Hoe dan ook, wil je eens je dagstatistieken bekijken in het Archief van DSMR-reader? Kloppen de dagtotalen qua kWh wel voor de dagen dat je afwijkingen hebt?

dennissiemensma, 2019-07-29:
@Merlin1201 ben je ondertussen in de gelegenheid geweest om het Archief te controleren?

Merlin1201, 2019-07-29:
Hoi Dennis,  ik heb er niet meer naar gekeken maar ook met domotica had ik in een Lua script een soortgelijke waarde dus ik heb geen idee wat daar aan de hand is.
Ik heb dsmr momenteel niet in gebruik maar ik zal hem nog eens installeren eens kijken of die waarde nu weer fout is.

dennissiemensma, 2019-09-03:
Ik zie je aanvulling helaas nu pas. Ik krijg geen meldingen van bijgewerkte comments, dus je kunt het beste aanvullingen in losse comments zetten en alleen editen wanneer er fouten in je eerder comment zit.

dennissiemensma, 2019-09-03:
Wat betreft je vraag. Je kunt het beste even inhoudelijk in ticket #9 kijken. Ik gebrui kzelf geen zonnepanelen, dus ik kan je niet met garantie zeggen wat klopt en hoe je het moet lezen in PVOutput. Als ik het me goed herinner, geeft de slimme meter (en dus dsmr-reader) de waarde terug na aftrek van eventuele opgewerkte elektriciteit. 
Dit hangt bijvoorbeeld af of de opgewekte elektriciteit op dezelfde groepen/fasen zitten als je verbruik.

closed 2019-09-03 by dennissiemensma