dsmrreader/dsmr-reader #651: Webinterface hangt

DennisIOT, 2019-06-05:
### DSMR V2.0.2

* DSMR-reader versie : v2.0.2
* Hardware : RaspberryPi 2
* DSMR-protocol versie v4 

### Bug
* De webinterface laat alleen maar gateway time out en of fout zien, na diverse issues doorlopen te hebben en alles vers opnieuw geïnstalleerd te hebben kom ik op hetzelfde punt uit. 
De dsmr reader vergaard netjes zijn data gezien het met de hand starten van de webinterface dit laat zien. Wat nu gereproduceerd kan worden is het opnieuw starten van de pi de webinterface niet start ook niet na enige tijd. 
Op het moment dat ik de volgende actie uitvoer werkt hij direct.

#403 stukje ....
Stop eerst dsmr_webinterface in Supervisor (ander krijg je sowieso een foutmelding): sudo supervisorctl stop dsmr_webinterface.
sudo su - dsmr
/home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--dsmr_webinterface.socket --pid /var/tmp/gunicorn--dsmr_webinterface.pid dsmrreader.wsgi

[image]

Hoe kan ik dit oplossen Dennis?

dennissiemensma, 2019-06-05:
Het zou kunnen zijn dat dit komt doordat de Pi2 net te traag is om alles simultaan in 'lage prio' te draaien, dit is een standaard-config die ik gemaakt heb. Helemaal gezien het wel lijkt te werken als je het handmatig uitvoert.

Wat je kunt proberen is om het webinterface-proces de 'lage prio' weg te halen, zodat die meer resources voor het proces gebruikt:
```
sudo vi /etc/supervisor/conf.d/dsmr-reader.conf
```

* Zoek onderaan in de file het blok met: "[program:dsmr_webinterface]"

* Zoek de regel die begint met: `command=/usr/bin/nice -n 15 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60..............`

* Haal daarin `/usr/bin/nice -n 15 ` weg en sla het bestand op.

* Herlaad nu de config en webinterface en kijk of het beter gaat:
```
sudo supervisorctl reread
sudo supervisorctl reload
```

DennisIOT, 2019-06-05:
Tikken en lopen, super en met oprecht een fantastische applicatie.!
Dank voor je input en snelle reactie

dennissiemensma, 2019-06-06:
Fijn om te horen, enjoy!

closed 2019-06-06 by dennissiemensma