dsmrreader/dsmr-reader #653: "It has been a while since the latest telegram reading." - direct na compleet nieuwe install

janwijbrand, 2019-06-05:
### Jouw omgeving
* DSMR-reader versie: `v2.1`
* Hardware: `RaspberryPi 3 model B+`
* DSMR-protocol versie van slimme meter: `v5`

### Bug, wens of iets anders
Sinds de plaatsing van een nieuwe slimme meter bij de aanpassing van onze aansluiting naar 3x25A, geeft de status pagina 3 waarschuwingen:

"It has been a while since the latest telegram reading.
Most recent reading: 4 months, 2 weeks ago"

"There are too many unprocessed telegrams.
Unprocessed telegrams: 59"

"Data processing is lagging behind: 4 months, 2 weeks ago"

Voor de tweede waarschuwing geldt dat het aantal unprocessed telegram naar ongeveer 60 oploopt en daarna weer "reset" naar een lager aantal. Ik heb grouping van readings op de default van "by minute" staan.

De eerste en derde melding echter, verbazen met. Omdat ik op een gegeven het vermoeden kreeg dat het SD kaartje mss niet lekker meer was, heb ik zojuist een hele nieuwe install gedaan op een verse SD. Maar dit hielp dus niet.

Enige idee waar ik moet zoeken?

Alvast bedankt!

dennissiemensma, 2019-06-05:
Begrijp ik het goed dat de melding `Data processing is lagging behind` nooit weg gaat? Ook niet na een dag?

Zou je eens een telegram willen plaatsen? Sinds de nieuwe versie van DSMR-reader, die je al draait, kan dat in de admin-interface. Deze staat onderaan de pagina `/admin/dsmr_datalogger/meterstatistics/1/change/`

Vergeet niet om de serienummer van je meter weg te halen uit de telegram voordat je hem hier plaatst. Dit is doorgaans de vierde en twee-na-laatste regel. Deze beginnen met: `0-0:96.1.1` en `0-1:96.1.0`.

janwijbrand, 2019-06-06:
> Begrijp ik het goed dat de melding `Data processing is lagging behind` nooit weg gaat? Ook niet na een dag?
>
> Zou je eens een telegram willen plaatsen? Sinds de nieuwe versie van DSMR-reader, die je [...]

Idd is "4 maanden, 2 weken" niet weg. Zou het kunnen dat de interne klok van de meter verkeerd staat? Te meer dat die meter statistic view in DSMR reader het ook over een timestamp van januari 2019 heeft.

latest telegram:

```
/ISK5\2M550T-1012

1-3:0.2.8(50)
0-0:1.0.0(190123021403W)
0-0:96.1.1 [REDACTED]
1-0:1.8.1(000225.956*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.212*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00004)
0-0:96.7.9(00003)
1-0:99.97.0(1)(0-0:96.7.19)(181219002514W)(0000000191*s)
1-0:32.32.0(00000)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(231.1*V)
1-0:52.7.0(233.1*V)
1-0:72.7.0(232.6*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.109*kW)
1-0:41.7.0(00.047*kW)
1-0:61.7.0(00.054*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!B39A
```

dennissiemensma, 2019-06-06:
Wil je voor de zekerheid even direct de datalogger-poort uitlezen, om te dubbelchecken dat je inderdaad de verkeerde datum ziet? Dat kan met:
```
sudo supervisorctl stop dsmr_datalogger
sudo su - dsmr
cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q

# Zodra je een telegram ziet, typ een "q" een punt (.) en ENTER. ALs het goed is stopt het uitlezen daarna.

CTRL + D
sudo supervisorctl start dsmr_datalogger
```

Als de datum niet klopt, verklaart dat een hoop. In dat geval kun je het het beste melden bij je netbeheerder denk ik. Wellicht kunnen zij het op afstand goedzetten.

janwijbrand, 2019-06-09:
Dit is een direct uitgelezen telegram:

```
1-3:0.2.8(50)
0-0:1.0.0(190126055848W)
0-0:96.1.1[REDACTED]
1-0:1.8.1(000250.076*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000000.000*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.210*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00004)
0-0:96.7.9(00003)
1-0:99.97.0(1)(0-0:96.7.19)(181219002514W)(0000000191*s)
1-0:32.32.0(00000)
1-0:52.32.0(00000)
1-0:72.32.0(00000)
1-0:32.36.0(00001)
1-0:52.36.0(00001)
1-0:72.36.0(00001)
0-0:96.13.0()
1-0:32.7.0(229.9*V)
1-0:52.7.0(229.4*V)
1-0:72.7.0(230.8*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
1-0:21.7.0(00.035*kW)
1-0:41.7.0(00.113*kW)
1-0:61.7.0(00.059*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!1405
/ISK5\2M550T-101
```

Welk veld hierin bevat de timestamp?

In de tussentijd ook contact opgenomen met netbeheerder. Die gaf aan dat de communicatie over CDMA niet lekker werkt. De meter wordt vervangen door een andere die over GPRS communiceert. Ik hoop een beetje dat daarmee het beschreven probleem ook verholpen wordt.

Bedankt zover voor je reacties trouwens!

dennissiemensma, 2019-06-09:
Dat is het veld bijna bovenaan met: `190126055848W` > `2019-01-26`

Ik vermoed dat je issues opgelost worden met een nieuwe meter. Helaas heb ik geen makkelijke fix voor alle data die tot nu toe op een verkeerd tijdstip staat.

dennissiemensma, 2019-06-12:
Voor nu sluit ik dit issue, er vanuit gaande dat het een issue is met de interne klok of config van je slimme meter. Mocht je na vervangen/fixen van je slimme meter door de netbeheerder alsnog issues hebben, dan assisteer ik je weer graag.

closed 2019-06-12 by dennissiemensma

janwijbrand, 2019-06-28:
Inderdaad: nieuwe slimme meter (de derde inmiddels, dus deze zal wel de slimste zijn dan ;)  ) en probleem opgelost. Bedankt voor je support tot zover!!

dennissiemensma, 2019-06-28:
Top, fijn om te horen. Dank voor je update.