dsmrreader/dsmr-reader #654: Restore Database Backup

megapearl, 2019-06-07:
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `latest`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3b`
* DSMR-protocol versie van slimme meter *v5)*: `v5`

Nadat de sdkaart voor de 2de keer gecrasht is probeer ik een oude database terug te zetten, ondertussen al een paar uur bezig, maar een hoop foutmeldingen, wat te doen?

`root@power:/home/dsmr/dsmr-reader/backups# ls -l
total 1479464
-rw-r--r-- 1 root root 262687712 Jun  7 15:26 dsmrreader-postgresql-backup-Friday.sql.gz
-rw-r--r-- 1 root root 103845930 Jun  7 15:26 dsmrreader-postgresql-backup-Monday.sql.gz
-rw-r--r-- 1 root root 278550486 Jun  7 15:27 dsmrreader-postgresql-backup-Saturday.sql.gz
-rw-r--r-- 1 root root 291771131 Jun  7 15:28 dsmrreader-postgresql-backup-Sunday.sql.gz
-rw-r--r-- 1 root root 246209101 Jun  7 15:28 dsmrreader-postgresql-backup-Thursday.sql.gz
-rw-r--r-- 1 root root 104245804 Jun  7 15:28 dsmrreader-postgresql-backup-Tuesday.sql.gz
-rw-r--r-- 1 root root 227633152 Jun  7 15:29 dsmrreader-postgresql-backup-Wednesday.sql.gz
root@power:/home/dsmr/dsmr-reader/backups# sudo sudo -u postgres dropdb dsmrreader
root@power:/home/dsmr/dsmr-reader/backups# sudo sudo -u postgres createdb -O dsmrreader dsmrreader
root@power:/home/dsmr/dsmr-reader/backups# zcat dsmrreader-postgresql-backup-Sunday.sql.gz | sudo sudo -u postgres psql dsmrreader
SET
SET
SET
SET
SET
 set_config
------------

(1 row)

SET
SET
SET
CREATE EXTENSION
COMMENT
SET
SET
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
[... 561 lines cut ...]
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
ALTER TABLE
`

dennissiemensma, 2019-06-07:
Ik denk dat je het beste even kunt zoeken in je dump naar:
* `hourstatistics`
* `daystatistics`

Dat zijn de belangrijkste tabellen, vooral de laatste, want die bevatten dat dagtotalen. Als je die nog hebt, al is het voor een deel, dan heb je wellicht nog je geschiedenis.

Ik hoor graag wat je kunt vinden.

megapearl, 2019-06-08:
Ik heb de complete database terug kunnen zetten zonder eerst de database te verwijderen.

1 week aan data totaal kwijt, maar dat is verder niet erg.

Alleen ik kan niets meer aanpassen in het admin panel, zodra ik op save of 'save and continue editing' van elke instelling klik krijg ik steeds een duplicate error:

Is dit nog aan te passen?

```
Server Error
Sorry, something unexpected happened.
Exception:
IntegrityError: duplicate key value violates unique constraint "django_admin_log_pkey"
DETAIL:  Key (id)=(20) already exists.
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/decorators.py", line 142, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/sites.py", line 223, in inner
    return view(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/solo/admin.py", line 68, in change_view
    extra_context=extra_context,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1637, in change_view
    return self.changeform_view(request, object_id, form_url, extra_context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/decorators.py", line 45, in _wrapper
    return bound_method(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/utils/decorators.py", line 142, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1522, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/options.py", line 1568, in _changeform_view
    self.log_change(request, new_object, change_message)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/options.py", line 828, in log_change
    change_message=message,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/contrib/admin/models.py", line 35, in log_action
    change_message=change_message,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 422, in create
    obj.save(force_insert=True, using=self.db)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/base.py", line 741, in save
    force_update=force_update, update_fields=update_fields)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/base.py", line 779, in save_base
    force_update, using, update_fields,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/base.py", line 870, in _save_table
    result = self._do_insert(cls._base_manager, using, fields, update_pk, raw)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/base.py", line 908, in _do_insert
    using=using, raw=raw)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 1186, in _insert
    return query.get_compiler(using=using).execute_sql(return_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1335, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 67, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 76, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

```

dennissiemensma, 2019-06-08:
Je zou kunnen kijken of dit het oplost:
```
sudo su - postgres
psql dsmrreader

# Voer dit als queries uit.
BEGIN;
SELECT setval(pg_get_serial_sequence('"django_admin_log"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "django_admin_log";
COMMIT;
```

megapearl, 2019-06-08:
That did the trick! Alles is weer aan te passen en op te slaan nu. super!
DSMR reageert nog wel erg traag, misschien dat hij op de achtergrond nog data aan het verwerken is?

Onder Status:

 P1 DSMR TELEGRAMS 
 It has been a while since the latest telegram reading. 

Most recent reading: 12 hours ago
Data processing is lagging behind: May 30, 2019

dennissiemensma, 2019-06-08:
Staat er ook een (hoog) aantal bij `Unprocessed telegrams`?

dennissiemensma, 2019-06-08:
Je zou de andere tabellen ook nog even kunnen fixen, wellicht hangen die op hetzelfde.

```
BEGIN;
SELECT setval(pg_get_serial_sequence('"dsmr_api_apisettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_api_apisettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backend_backendsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backend_backendsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backend_emailsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backend_emailsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backend_scheduledprocess"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backend_scheduledprocess";
SELECT setval(pg_get_serial_sequence('"dsmr_backup_backupsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backup_backupsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backup_dropboxsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backup_dropboxsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_backup_emailbackupsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_backup_emailbackupsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_electricityconsumption"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_electricityconsumption";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_gasconsumption"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_gasconsumption";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_energysupplierprice"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_energysupplierprice";
SELECT setval(pg_get_serial_sequence('"dsmr_consumption_consumptionsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_consumption_consumptionsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_dataloggersettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_dataloggersettings";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_retentionsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_retentionsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_dsmrreading"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_dsmrreading";
SELECT setval(pg_get_serial_sequence('"dsmr_datalogger_meterstatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_datalogger_meterstatistics";
SELECT setval(pg_get_serial_sequence('"dsmr_frontend_frontendsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_frontend_frontendsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_frontend_notification"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_frontend_notification";
SELECT setval(pg_get_serial_sequence('"dsmr_mindergas_mindergassettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mindergas_mindergassettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_mqttbrokersettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_mqttbrokersettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_jsondaytotalsmqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_jsondaytotalsmqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_splittopicdaytotalsmqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_splittopicdaytotalsmqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_rawtelegrammqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_rawtelegrammqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_jsontelegrammqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_jsontelegrammqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_splittopictelegrammqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_splittopictelegrammqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_splittopicmeterstatisticsmqttsettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_splittopicmeterstatisticsmqttsettings";
SELECT setval(pg_get_serial_sequence('"dsmr_mqtt_message"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_mqtt_message";
SELECT setval(pg_get_serial_sequence('"dsmr_notification_notificationsetting"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_notification_notificationsetting";
SELECT setval(pg_get_serial_sequence('"dsmr_notification_statusnotificationsetting"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_notification_statusnotificationsetting";
SELECT setval(pg_get_serial_sequence('"dsmr_pvoutput_pvoutputapisettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_pvoutput_pvoutputapisettings";
SELECT setval(pg_get_serial_sequence('"dsmr_pvoutput_pvoutputaddstatussettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_pvoutput_pvoutputaddstatussettings";
SELECT setval(pg_get_serial_sequence('"dsmr_stats_note"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_stats_note";
SELECT setval(pg_get_serial_sequence('"dsmr_stats_daystatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_stats_daystatistics";
[... cut ...]
SELECT setval(pg_get_serial_sequence('"dsmr_stats_electricitystatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_stats_electricitystatistics";
SELECT setval(pg_get_serial_sequence('"dsmr_weather_weathersettings"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_weather_weathersettings";
SELECT setval(pg_get_serial_sequence('"dsmr_weather_temperaturereading"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_weather_temperaturereading";
COMMIT;
```

megapearl, 2019-06-08:
> Staat er ook een (hoog) aantal bij `Unprocessed telegrams`?

Nee, niets...

 TELEGRAM PROCESSING 
 Unprocessed telegrams: -

megapearl, 2019-06-08:
> Je zou de andere tabellen ook nog even kunnen fixen, wellicht hangen die op hetzelfde.
>
> ```
> BEGIN;
> SELECT setval(pg_get_serial_sequence('"dsmr_api_apisettings"','id'), coalesce(max("id"), 1), [...]

Dat werkt inderdaad, alleen bij "DAY & HOUR STATISTICS" staat nog  Data processing is lagging behind: May 30, 2019

[image]

dennissiemensma, 2019-06-08:
Wellicht even kijken of het morgen hersteld is.

megapearl, 2019-06-09:
Het is inderdaad hersteld, super, bedankt voor je snelle support!

closed 2019-06-09 by megapearl