dsmrreader/dsmr-reader #655: suggesties voor installatie handleiding

Cornmeister, 2019-06-09:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.1`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3B`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v2.2+`

### Bug, wens of iets anders
In de handleiding staat bij meerdere stappen een dubbel 'sudo'  commando,
bijv:
`sudo sudo -u postgres createdb -O dsmrreader dsmrreader`

ik moest hier een enkele van maken, anders slikte raspbian dit niet.

Tevens dient 'cu'  geinstalleerd te worden, anders komt er na installatie geen data binnen.
`sudo apt install cu`

dennissiemensma, 2019-06-09:
Bedankt voor je melding.

De dubbele `sudo` zou niet uit moeten maken. De reden dat het er dubbel in staat overigens is omdat sommige gebruikers direct als `root` de commands uitvoeren, maar andere weer als een (sub) sudo-user. Krijg je een foutmelding of iets van Raspbian? En welke versie van het OS heb je draaien?

Verder maakt DSMR-reader voor zover ik weet geen gebruik van `cu`. Het is puur een tooltje om handmatig metingen uit te lezen via de command line. DSMR-reader zelf gebruikt hier `pyserial` voor, die direct de seriele poort uitleest. Als ik `cu` verwijder en de applicatie herstart, blijft het uitlezen werken. Krijg je een foutmelding?

dennissiemensma, 2019-07-29:
@Cornmeister zie bovenstaande comment. Enig idee of je die informatie nog hebt?

Cornmeister, 2019-07-29:
Ik denk dat ik nog eea heb liggen, half aug. Heb ik weer toegang tot de desbetreffende systemen.

Verstuurd vanaf mijn iPhone

Cornmeister, 2019-08-17:
@dennissiemensma ik heb cu gemoved naar een andere dir, en het werkt nog steeds. Ik vermoed dat bij het initieel starten van de tool het even tijd kostte voordat de data tevoorschijn kwam.
Dubbele sudo heb ik nog eens getest, en dit lijkt nu wel te werken :$ kan dus gesloten worden, geen verdere actie noodzakelijk.

closed 2019-08-17 by Cornmeister

dennissiemensma, 2019-08-17:
Oke dankjewel voor de terugkoppeling. Fijn dat het nu wel werkt!