dsmrreader/dsmr-reader #656: Oude versie blijft terugkomen bij updaten naar 2.1

microsaft82, 2019-06-09:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...
Ik update van 2.0.2 naar 2.1 en dat lukt zonder fouten en door jouw exacte instructie op te volgen. Werkt ook in webui maar na reboot heb ik keer op keer weer de 2.0.2 versie. 
Heb jij een idee?
Ik draai hem op een rpi3b.
Groet,
Mike

dennissiemensma, 2019-06-10:
Bedankt voor je vraag. Blijven de rest van de gegevens wel bewaard? Als in, mis je geen verbruiksdata na de reboot? 

Je zou eens kunnen kijken of andere schrijfacties wel persistent zijn:
```
sudo su - dsmr
touch ~/test-file
CTRL + D
sudo reboot
```

Na de reboot, kijk of de file er nog staat:
```
sudo su - dsmr
stat ~/test-file
```

microsaft82, 2019-06-10:
Thanks. Jij hij is read only. Is al de tweede die nu in korte tijd defect
raakt. Ik blijf ze niet vervangen :(
Misschien toch maar eens ergens anders op gaan draaien...

dennissiemensma, 2019-06-11:
Helaas is dat een zwakte van de SD-kaartjes, waar ik geen oplossing voor heb. Soms gaan ze een maand mee, soms meerdere jaren, het is erg willekeurig. Je zou nog kunnen overwegen om de database op een USB-stick-mount te draaien. Wellicht houdt die het wat langer vol dan een SD-kaartje.

microsaft82, 2019-06-12:
Ja ik ben er al door een aantal heen in korte tijd. Ik ga op zoek naar een
andere oplossing. Ben bang dat een USB stick ook snel stuk gaat dan.
Kan de applicatie niet beter veel langer cachen in mem en bv maar eens per
uur naar SD schrijven?

dennissiemensma, 2019-06-12:
Ik weet eerlijk gezegd niet of SD-kaartjes slijten door de hoeveelheid writes of de grootte van de writes.

Onderwater gebruikt de applicatie (standaard) PostgreSQL als database. Deze is wellicht te tunen met andere instellingen, zodat die minder vaak schrijft. Ik heb er zelf alleen geen ervaring mee. Je zou kunnen kijken naar het onderwerp "Checkpointing" in dit blog: https://severalnines.com/blog/tuning-io-operations-postgresql

Een ander onderwerp is Vacuuming. Mocht je DSMR-reader hebben geconfigureerd dat oude metingen na X-periode weggegooid worden, dan komen er "gaten" in de database en trekt die af en toe dat recht. Ook dat kan ongewenst zijn en voor extra I/O zorgen. Je kunt ofwel retentie in DSMR-reader dan uitzetten of kijken of je de Autovacuuming van PostgreSQL anders kunt instellen.

closed 2019-06-12 by dennissiemensma

microsaft82, 2019-06-12:
Thanks voor de tips Dennis. Ik heb maar weer een kaartje besteld want was
door de voorraad heen. Als die binnen is zal ik de db verplaatsen naar USB
en ik zal ook even kijken of de io te verminderen is inderdaad. Thanks.

Mooi project hoor. Heb er al veel plezier van gehad. Baalde nu alleen want
iedere keer opnieuw alles installeren is niet leuk en gebruik de pi ook
voor andere zaken dus ik blijf aan de gang :)

jeroenpeters1986, 2019-06-12:
@microsaft82 een andere tip die ik nog kan geven is bijvoorbeeld het hosten van de database op een vpsje die je nog "hebt liggen". Daarmee schrijf je naar "de cloud" en niet naar je SD kaartje. Je kunt de backuptaken daarom ook uitzetten en er hoeft niets gecomprimeerd te worden voor backups e.d. op de schijf :-)

Heb je geen VPSje ergens over, dan zijn er altijd nog betaalde diensten (die een gratis beginplan hebben). Voorbeelden zijn https://www.elephantsql.com/plans.html en https://www.heroku.com/postgres maar er zijn er ongetwijfeld meer, al is een VPS bij Hertzner of OVH dan misschien alsnog goedkoper uiteindelijk.