dsmrreader/dsmr-reader #657: dropbox backup eenmalig?

RezzZ, 2019-06-11:
gister is mijn usbstick van mn rpi gecrasht. Nu hoopte ik op een recente backup bij dropbox maar nu is er slechts eenmalig een backup weggeschreven (maart 2019).. de dag dat ik het had geconfigureerd. Bug of was er bij mij een probleem. DSMR Reader gaf alleen geen melding dat de uploads mislukten..

dennissiemensma, 2019-06-11:
Bedankt voor je melding en spijtig om te horen. Dit is een specifiek probleem met Docker dat bekend is, zie ook issue #613.

Hoewel ik officieel geen ondersteuning geef voor Docker, wil ik wel graag dat het zonder issues werkt. Ik raad je daarom aan om #613 te volgen. Eventueel kun je ook nog (in de laatste versie van DSMR- reader) een extra back-up per e-mail aanzetten.

dennissiemensma, 2019-06-11:
Hoh, ik haal nu even twee issues door elkaar heen zie ik. Voor zover ik wist, kwam dit alleen voor op Docker. Welke versie van DSMR-reader gebruik je?

RezzZ, 2019-06-12:
Hoi Dennis, ook ik draai DSMR in docker, sorry, had ik moeten vermelden. Dus ik ga 613 volgen! Backup per email aanzetten ga ik zeker doen en hopen op onedrive support in de toekomst ;)

closed 2019-06-12 by RezzZ