dsmrreader/dsmr-reader #658: cu gives weird output

jeroen369, 2019-06-14:
Hi, 

Ik probeer de slimme meter uit te lezen middels de Raspberry Pi. Als eerste stap via su, maar daar loop ik al tegen problemen aan. Met het commando "cu -l /dev/ttyUSB0 -s 115200 --parity=none -E q" krijg ik live resultaat die steeds verder aanvult, maar dat resultaat ziet er uit als "...|~||||||~~>|~|||~~~|~~|||~~~|~|||||||||~~~|~~|||||||||~~||~|||~>||~|||~~||~|||||~|||||~|~||||~>||~~~||~|||||~~||~|||||~~||~||||~>||~||||~>||~||||~~||~...". 

Hoe zorg ik er voor dat dit leesbaar wordt? Mijn slimme meter is een Merk: ISRA AM 550
DSM/ESMR: ESMR 5.0. P1 header: /ISK5\

dennissiemensma, 2019-06-14:
Wellicht gaat er iets mis met het omzetten van de data in de kabel. Gebruik je een kant-en-klare kabel of heb je er zelf eentje in elkaar gezet?

jeroen369, 2019-06-14:
Deze kant en klare kabel https://m.nl.aliexpress.com/item/32986693564.html?spm=a2g0n.orderlist-amp.item.32986693564&aff_trace_key=e67cd14e41564760a977b7c22554f2d0-1559995668165-00570-UneMJZVf&aff_platform=msite&m_page_id=3454amp-xOB3IzEySvBuwsEAWj2uAg1560534935628

dennissiemensma, 2019-06-14:
Ik heb zelf helaas alleen ervaring met de kabel van SOS-solutions en ik durf niet met zekerheid te zeggen of het aan je kabel ligt. Het is een goed teken dat er dus steeds data over de kabel heenkomt, dat geeft volgens mij aan dat je met de goede instellingen verbindt. Het zou kunnen zijn dat er nog een weerstandje niet goed zit in de kabel, maar ik heb daar zelf geen inhoudelijke kennis van. Misschien is het simpelweg een kwestie van het signaal inverten, maar dat weet ik dus niet zeker (of je dan uberhaupt data binnen zou moeten krijgen).

Ik denk dat je dan het beste even op het Tweaker-forum kunt rondvragen. Wellicht kun je je probleem in dit topic voorleggen en herkent iemand direct wat er mis is: https://gathering.tweakers.net/forum/list_messages/1578510

jeroen369, 2019-06-16:
tnxvoor je vlotte reactie Dennis. ik ga verder kijken...

closed 2019-06-16 by jeroen369