dsmrreader/dsmr-reader #659: Q: dsmr_datalogger_api_client.py

marceltrapman, 2019-06-17:
Hi,

Ik wil het script 'dsmr_datalogger_api_client.py' copieren naar een docker container maar tot nu toe start het script niet op van /home/dsmr in de container.

Kan/moet ik hem zelf op een bepaalde plek plaatsen?
Of is er iets anders dat ik doen kan/moet?

Thanks,

Marcel

dennissiemensma, 2019-06-18:
Hallo Marcel, ik denk dat je dit het beste even kunt vragen aan de ontwikkelaar van de docker-installatie die je gebruikt. Mijn kennis van Docker is vrij beperkt en ik wil graag dat je meteen zo goed mogelijk geholpen wordt.

Weet je toevallig welke handleiding, repo of URL je hebt gebruikt om de DSMR-docker-container te downloaden/inrichten? Is het een van deze drie? https://dsmr-reader.readthedocs.io/en/latest/installation/docker.html

dennissiemensma, 2019-06-24:
@marceltrapman is het nog gelukt met contact leggen?

marceltrapman, 2019-06-24:
Sorry, ik ben een paar dagen 'uit de lucht geweest' (dwz mijn laptop) :(
Ik gebruik de image van ualex (ualex73/dsmr-reader-docker:arm32v6).
Zal idd kijken of ik hem kan vragen.

marceltrapman, 2019-06-24:
Hmm, ik kan geen email adres vinden en ook geen issue aanmaken :(
Alternatief kan ik een service aanmaken om het script te runnen maar dat doe ik liever niet...

dennissiemensma, 2019-06-24:
Je zou hem nog kunnen proberen te mentionen. 

@ualex73, zie bovenstaande vraag, weet jij hier een antwoord op? :]

ualex73, 2019-06-25:
Op dit moment is er geen nette manier om de datalogger te gebruiken in de docker image, want deze wordt elke keer wegblazen zodra je de container opnieuw aanmaakt.

Dit zeggend, het python script kan worden geplaats in "/dsmr" (dus niet in "/home/dsmr") en dan moet de "/etc/supervisor.d/supervisord.ini" worden aangepast met het onderstaande (wel eerst de andere [program:dsmr*] weghalen).

BTW - Ik zal even kijken of ik dit een switch kan maken op de docker commandline, zodat het met 1 image kan.

[program:dsmr_client_datalogger]
command=/usr/bin/nice -n 5 python3 -u /dsmr/dsmr_datalogger_api_client.py
pidfile=/var/tmp/dsmrreader--%(program_name)s.pid
user=dsmr
group=dsmr
autostart=true
autorestart=true
startsecs=1
startretries=100
stopwaitsecs=20

marceltrapman, 2019-06-25:
OK, het lukt mij om de container opnieuw te builden maar ik moet nog even goed uitzoeken hoe ik het script toegang kan geven tot de serieele poort van de host. 
priviledged en pid host helpen me kennelijk niet.

Overigens build ik de container op en voor een raspberry en daar is /usr/bin/nice /bin/nice...

Wordt vast vervolgd.

ualex73, 2019-06-25:
Het supervisor.ini voorbeeld is niet perfect sorry, maar ik uiteindelijk morgen een test build beschikbaar stellen voor de Raspberry Pi. Dan kan met ENV variabelen de client aansturen, zonder dat je een Docker build hoeft te doen - indien je daar interesse in hebt ;-)

marceltrapman, 2019-06-25:
@ualex73 dat zou natuurlijk super zijn :)

dennissiemensma, 2019-06-25:
Klinkt goed! Mocht ik nog wijzigingen moeten doen in [het script](https://github.com/dennissiemensma/dsmr-reader/blob/master/dsmr_datalogger/scripts/dsmr_datalogger_api_client.py), zoals env vars, dan hoor ik het graag.

ualex73, 2019-06-25:
@dennissiemensma - Thanks, ik zal een pull inschieten zodra we weten dat het werkt ;-)

@marceltrapman:
ik heb net een arm32v6-beta gepusht naar hub.docker.com, dus als je deze kan proberen?

Dus:
docker pull ualex73/dsmr-reader-docker:arm32v6-beta

Met docker run moet je dan het volgende doen:
docker run -d \
  --name dsmr \
  --restart unless-stopped \
  -e DSMR_HOST=192.168.1.1 \
  -e DSMR_SERIALPORT=/dev/ttyUSB0 \
  -e DSMR_APIKEY=APIKEY-BLABLABLA-ABCDEFGHI \
  -v /dev:/dev \
  --device /dev/ttyUSB0:/dev/ttyUSB0 \
  ualex73/dsmr-reader-docker

DSMR_HOST - Waar je DSMR draait (IP of hostnaam)
DSMR_SERIALPORT - je seriele device
DSMR_APIKEY - De DSMR v1 API key

marceltrapman, 2019-06-26:
Hi @ualex73,

Sorry mijn communicatie was weer te kort door de bocht :(

Ik had moeten schrijven dat ik mijn aangepaste script wilde gebruiken...
Dit script leest de data niet van een serieele port maar een netwerk poort om deze vervolgens wel naar de server te sturen via een api call.

Ik kan deze opzet dus helaas niet testen want die werkt niet bij/voor mij.

Wel een mooie oplossing trouwens.

Inmiddels weet ik ook wat ik fout deed dus ik denk dat ik de container toch maar zelf ga builden...

Nogmaals excuus, hopelijk is het handig voor anderen!

ualex73, 2019-06-27:
Np ;-) Als de aanpassingen voor netwerk poort niet extreem zijn, zou ik die ook nog in het script kunnen meenemen (en de docker).

marceltrapman, 2019-06-28:
Mijn script ging uit van een netwerk poort (ser2net) maar ik heb het weer veranderd.
De basis is het originele script maar ik maak gebruik van dsmr-parser en heb het meer universeel gemaakt (zowel met de api versie als met de versie van de metere) met mijn eigen MQTT JSON.
Het is echter nog niet af maar zal het op github zetten als het af is...

marceltrapman, 2019-07-02:
Hier istie https://github.com/marceltrapman/dsmr-datalogger

Ik heb hem inmiddels in een 'eigen' container draaien en het werkt goed dus ik weet niet of het het waard is om (veel) energie in de integratie te stoppen.

dennissiemensma, 2019-07-03:
Top. Ik ga er vanuit dat je hier dan voor nu mee geholpen bent. Mochten anderen nog zoeken op Docker, dan komen ze wellicht ook hier uit. Bedankt voor het delen!

closed 2019-07-03 by dennissiemensma