dsmrreader/dsmr-reader #661: na nieuwe installeatie worden telegrammaen niet meer verwerkt

newtech2015, 2019-06-19:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v1.X.Y`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi X...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

dennissiemensma, 2019-06-19:
Zou je eens willen kijken of er een melding in de logs voorbij komt?

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html#logging

newtech2015, 2019-06-19:
na een aantal regels opnieuw te hebben uitgevoerd van de setup, werkt het opeens weer

closed 2019-06-19 by newtech2015