dsmrreader/dsmr-reader #662: Update niet van 2.0.2

peterjohan52, 2019-06-20:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.0.2`
* Hardware *(RaspberryPi 2 3 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Bug, wens of iets anders
* Bijvoorbeeld stappen hoe een probleem te reproduceren is...
* Of simpelweg een beschijving van een feature die je graag wenst te zien...

Na,

sudo su - dsmr
./deploy.sh

en het lopen van het script.

en een reboot blijft de versie. v2.0.2.

Waarom? wat gaat er mis.

dennissiemensma, 2019-06-20:
Bedankt voor je vraag. 

* Krijg je na upgraden wel te zien dat die bijgewerkt is? Er staat dan een bericht in de terminal in de trend van: ` --- Deployed version: v2.2.0`

* Kan het overigens zijn dat je SD-kaartje niet meer goed is? Helaas komt dat regelmatig voor en een van de symptonen daarbij kan zijn dat gegevens niet goed weggeschreven worden, zodat na een reboot de oude versie weer staat.

Je zou in het Archief kunnen kijken of je toevallig al wat data mist.

dennissiemensma, 2019-06-24:
@peterjohan52 ben je hier nog uit gekomen?

peterjohan52, 2019-06-26:
Hallo Dennis,

momenteel ben ik in het buitenland. Dank voor je reactie. Als ik terug ben, ga ik proberen om de update alsnog te installeren.

Alvast hartelijk dank.

peterjohan52, 2019-07-06:
--- Merging/updating checkout.
Updating 9daf68ca..fbd4c0b7
error: Your local changes to the following files would be overwritten by merge:
	dsmrreader/config/base.py
Please commit your changes or stash them before you merge.
Aborting

0 static files copied to '/var/www/dsmrreader/static', 440 unmodified.

jeroenpeters1986, 2019-07-06:
Het ziet er naar uit dat je wijzigingen hebt gemaakt aan de base config en
niet aan de lokale configuratie. Het beste kun je je wijzigingen noteren en
‘git reset —hard’ doen. Daarna moet het upgraden werken.

Je wijzigingen terugzetten kun je doen in dsmrreader/settings.py

peterjohan52, 2019-07-06:
import os

from dsmrreader.config.production import *

# Default settings, adjust if you use other ones, or use environment variables.
DSMRREADER_NAME = 'dsmrreader'
DSMRREADER_USER = 'dsmrreader'
DSMRREADER_PASSWORD = 'dsmrreader'
DSMRREADER_HOST = 'localhost'
DSMRREADER_CONN_MAX_AGE = 60
DSMRREADER_BACKEND_SLEEP = 1
DSMRREADER_DATALOGGER_SLEEP = 0.5
DSMRREADER_MQTT_SLEEP = 1

# Change me when exposing your application to the outside world using the Internet!
DSMRREADER_SECRET_KEY = 'heb ik even weggehaald'

# Settings are read from environment or, when not found, default to the values above.
DATABASES = {
    'default': {
        'ENGINE': 'django.db.backends.postgresql',
        'NAME': os.environ.get('DB_NAME', DSMRREADER_NAME),
        'USER': os.environ.get('DB_USER', DSMRREADER_USER),
        'PASSWORD': os.environ.get('DB_PASS', DSMRREADER_PASSWORD),
        'HOST': os.environ.get('DB_HOST', DSMRREADER_HOST),
        'CONN_MAX_AGE': os.environ.get('CONN_MAX_AGE', DSMRREADER_CONN_MAX_AGE),
    }
}

SECRET_KEY = os.environ.get('SECRET_KEY', DSMRREADER_SECRET_KEY)

"""
    Enable and change the logging level below to alter the verbosity of the (backend) command(s).
    - DEBUG:             Log everything.
    - INFO:              Log most things.
    - WARNING (default): Log only warnings and errors.

    Restart the commands in Supervisor to apply any changes.
"""
# LOGGING['loggers']['commands']['level'] = 'DEBUG'

Bovenstaande is de settings.py

dennissiemensma, 2019-07-06:
Je kunt met `git diff` zien welke wijzigingen je hebt gedaan. Als die allemaal weg kunnen, voer je het commando van Jeroen uit: `git reset —hard`

peterjohan52, 2019-07-07:
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ git diff
diff --git a/dsmrreader/config/base.py b/dsmrreader/config/base.py
index de42424a..af91cbf3 100644
--- a/dsmrreader/config/base.py
+++ b/dsmrreader/config/base.py
@@ -77,7 +77,7 @@ MIDDLEWARE = (
     'django.middleware.csrf.CsrfViewMiddleware',
     'django.contrib.auth.middleware.AuthenticationMiddleware',
     'django.contrib.messages.middleware.MessageMiddleware',
-    'django.middleware.clickjacking.XFrameOptionsMiddleware', (rood)
+#    'django.middleware.clickjacking.XFrameOptionsMiddleware', (groen)
     'django.middleware.security.SecurityMiddleware',
 
     # Local.
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ 

is resultaat na git diff

dennissiemensma, 2019-07-07:
Ik zie dat je een hack hebt gedaan om de applicatie toe te staan in een frame. Hier is een aparte instelling voor. Zie https://github.com/dennissiemensma/dsmr-reader/issues/344#issuecomment-315523564 voor meer informatie.

Je kunt dan verder gewoon `git reset —hard` doen en opnieuw proberen te update.

peterjohan52, 2019-07-07:
Dat was de oplossing. Draai nu ook Buster. Om DSMR weer aan de praat te krijgen, moet je wel de installatie stappen opnieuw doorlopen. Bedankt voor de hulp.

dennissiemensma, 2019-07-07:
Top, fijn om te horen

closed 2019-07-07 by dennissiemensma