dsmrreader/dsmr-reader #663: problemen met installatie

eterpstra, 2019-06-26:
Hallo Dennis,

Inmiddels heeft een vriend van mij ook interesse in jouw software. Helaas gaat het installeren weer niet goed. Ik heb de handleiding step bij step gevolgd.

Ik heb ubuntu server 19 geinstalleerd.
Ik loop vast op het volgende stukje:

virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3

Ik ben ingelogd als DSMR user maar krijg dan deze error:

```
Running virtualenv with interpreter /home/dsmr/.virtualenvs/dsmrreader/bin/python3
Please use the *system* python to run this script
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2375, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 946, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/posixpath.py", line 378, in abspath
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType
(dsmrreader) dsmr@winkelserver:~/dsmr-reader$ source ~/.virtualenvs/dsmrreader/bin/activate
(dsmrreader) dsmr@winkelserver:~/dsmr-reader$ cd ~/dsmr-reader
(dsmrreader) dsmr@winkelserver:~/dsmr-reader$ virtualenv ~/.virtualenvs/dsmrreader --no-site-packages --python python3
Running virtualenv with interpreter /home/dsmr/.virtualenvs/dsmrreader/bin/python3
Please use the *system* python to run this script
Traceback (most recent call last):
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 2375, in <module>
    main()
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 724, in main
    symlink=options.symlink)
  File "/usr/lib/python3/dist-packages/virtualenv.py", line 946, in create_environment
    site_packages=site_packages, clear=clear, symlink=symlink))
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.6/posixpath.py", line 378, in abspath
    path = os.fspath(path)
TypeError: expected str, bytes or os.PathLike object, not NoneType
```

Heb jij enige idee wat er fout gaat?

dennissiemensma, 2019-06-26:
Ik heb deze foutmelding nog niet eerder gezien. Het lijkt iets met Python of de virtualenv. 

Welke versie van Ubuntu 19 gebruik je? De laatste minor? Kun je `lsb_release -a` doen en terugkoppelen wat die aangeeft?

Ik zal dan eens kijken of ik het kan reproduceren, want ik zou niet zo 1-2-3 weten waar dit door komt.

eterpstra, 2019-06-27:
Hi Dennis,

Ik zie dat het toch nog om versie 18 gaan:

Distributor ID: Ubuntu                                                                                                                                                                                                                    
Description:    Ubuntu 18.04.2 LTS                                                                                                                                                                                                        
Codename:       bionic

eterpstra, 2019-06-27:
Overigens lees ik hier ook nog iets over op een ander stukje van het forum:

https://github.com/aws/chalice/issues/275

eterpstra, 2019-06-28:
Dennis,

Hij mag dicht. Ik heb per ongeluk de virtual env als sudo user gedaan en niet als dsmr.

dennissiemensma, 2019-06-28:
Bedankt voor je update en fijn dat je het zo snel gevonden hebt!

closed 2019-06-28 by dennissiemensma