dsmrreader/dsmr-reader #664: API v1 doet 't niet?

ThaStealth, 2019-06-27:
### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.1.0`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `Docker (xirixiz/dsmr-reader-docker:latest)'
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Bug, wens of iets anders
Als ik naar http://IP:PORT/api/**v1**/datalogger/dsmrreading ga krijg ik een HTTP405 (method not allowed) http://IP:PORT/api/**v2**/datalogger/dsmrreading doet het wel. Het lijkt erop dat de API v1 versie niet luisterd? Hoe kan ik dit verder debuggen?

ThaStealth, 2019-06-27:
Nvm, ik moet posten ipv getten...

closed 2019-06-27 by ThaStealth

ThaStealth, 2019-06-27:
Ik gebruikte de GET ipv POST

dennissiemensma, 2019-06-28:
Bedankt voor je terugkoppeling en fijn dat je het zelf al zo snel gevonden had!