dsmrreader/dsmr-reader #665: Django security releases issued: 2.2.3

mjanssens, 2019-07-01:
Enkel in geval van django achter een reverse proxy die connect met https, dus kleine kans dat dsmr-reader eindgebruikers een dergelijke setup hebben. Maar toch reden om te melden.
CVE-2019-12781

dennissiemensma, 2019-07-01:
Dankje, ik zal hem deze week bijwerken.

dennissiemensma, 2019-07-01:
Bijgewerkt in `development`. Ik zal hem even een of twee dagen draaien voor de zekerheid.

dennissiemensma, 2019-07-03:
Uitgebracht in nieuwe patch release.

closed 2019-07-03 by dennissiemensma

mjanssens, 2019-07-03:
Thanks, draait hier zonder probleem

mjanssens, 2019-12-07:
> Enkel in geval van django achter een reverse proxy die connect met https, dus kleine kans dat dsmr-reader eindgebruikers een dergelijke setup hebben.

Lees net mijn eigen commentaar, dit klopt natuurlijk niet. Nginx draait in de standaard installatie als reverse-proxy.