dsmrreader/dsmr-reader #668: 502 Bad Gateway / nginx/1.10.3

T1rnanog, 2019-07-15:
DSMR-reader 2.0.2

Geen idee wat er mis is gegaan maar hij doet het niet meer. Iemand een idee hoe ik hem weer aan de praat krijg? Alvast bedankt.

2019/07/15 21:42:53 [crit] 628#628: *35 connect() to unix:///var/tmp/gunicorn--dsmr_webinterface.socket failed (2: No such file or directory) while connecting to upstream, client: 192.168.2.1, server: _, request: "GET /favicon.ico HTTP/1.1", upstream: "http://unix:///var/tmp/gunicorn--dsmr_webinterface.socket:/favicon.ico", host: "192.168.2.1", referrer: "http://192.168.2.1/"

dsmr_backend                     RUNNING   pid 9669, uptime 0:22:25
dsmr_datalogger                  RUNNING   pid 9678, uptime 0:22:18
dsmr_mqtt                        RUNNING   pid 11599, uptime 0:00:15
dsmr_webinterface                FATAL     Exited too quickly (process log may have details)

pi@raspberrypi:~ $ cd /home/dsmr/dsmr-reader/
pi@raspberrypi:/home/dsmr/dsmr-reader $ sudo -u dsmr /usr/bin/nice -n 15 /home/d                                                  smr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind                                                   unix:/var/tmp/gunicorn--TEST.socket --pid /var/tmp/gunicorn--TEST.pid dsmrreader                                                  .wsgi

[2019-07-15 21:29:57 +0200] [10712] [INFO] Starting gunicorn 19.9.0
[2019-07-15 21:29:57 +0200] [10712] [INFO] Listening at: unix:/var/tmp/gunicorn-                                                  -TEST.socket (10712)
[2019-07-15 21:29:57 +0200] [10712] [INFO] Using worker: sync
[2019-07-15 21:29:57 +0200] [10716] [INFO] Booting worker with pid: 10716
[2019-07-15 21:30:01 +0200] [10712] [INFO] Handling signal: winch

dennissiemensma, 2019-07-15:
@T1rnanog wat probeerde je te doen? Heb je iets geupdate of is het zomaar stuk gegaan?

Ik zie verder iets met `--TEST`, is dat van jezelf?

dennissiemensma, 2019-07-15:
Het lijkt erop dat de datalogger en backend het nog wel doen, maar dat de applicatie-server stuk is. Wat zegt de log in supervisor? `sudo tail -n 50 /var/log/supervisor/dsmr_webinterface.log`

T1rnanog, 2019-07-16:
Hallo Dennis, 
Vandaag een reboot gedaan mijn Pi wil niet meer opstarten. 
Even terug gedacht wat ik gister uitgevoerd heb.

sudo apt-get update
sudo apt-get upgrade
sudo apt-get dist-upgrade
sudo do-release-upgrade
sudo reboot

Nu moet ik zeggen mijn kennis op dit gebied gaat niet zo ver. Kan alleen handleidingen lezen en kopieeren en plakken uit je topics.
Mijn doel was de Pi weer helemaal up te date te brengen, wellicht kun je dit ooit nog eens verwerken in de FAQ voor mij ^^
Ik heb wel de dropbox backups ga zo een verse installatie doen.
Bedankt voor de hulp.

dennissiemensma, 2019-07-16:
Oke prima, in dat geval sluit ik deze voor nu. Mocht je er niet meer uitkomen dan hoor ik het graag.

closed 2019-07-16 by dennissiemensma

Ernst79, 2019-07-19:
@dennissiemensma Ik denk dat dit komt door de dist-upgrade. Ik heb eenzelfde issue na de dist-upgrade van Stretch naar Buster. Daarbij wordt ook postgresql geupgrade naar versie 11, wat ook weer voor problemen kan zorgen, zie https://wiki.postgresql.org/wiki/Locale_data_changes. Ook ik ga een verse installatie doen, maar het lijkt er dus op dat de upgrade van Stretch naar Buster niet probleemloos (voor DSMR-reader) uitgevoerd kan worden.

dennissiemensma, 2019-07-19:
Bedankt voor je terugkoppeling! 

Mijn ervaring met dist-upgrades is hetzelfde. Doorgaans installeer ik zelf liever een heel nieuw schoon OS, maar wel jammer dat upgraden dus niet zo 1-2-3 kan.

helmo, 2019-08-04:
Zelf net ook de upgrade naar buster gedaan.
Tijdens de upgrade gaf Debian me netjes een melding over de nieuwe postgres versie, met de onderstaande twee commando's die vermeld werden was de database weer in orde.
```sh
 pg_dropcluster --stop 11 main
 pg_upgradecluster 9.6 main
```
Mijn ervaring is juist dat Debian upgrades een hoop tijd schelen. Tuurlijk moet je dat niet op een belangrijke productie server proberen zonder test en backup ... maar door op te letten wat er gebeurd en het uitzoeken leer ik elke keer wat bij... volgende upgrade gaat weer een stukje sneller  :)

helmo, 2019-08-06:
Nog een aanvulling...

Na het draaien van `apt autoremove` was de oude python 3.5 verwijderd, maar faalde dsmr.
Voor het gebruik van python 3.7 die er al wel op stond was het opnieuw bouwen van het virtual env nodig.

Onderstaande stappen uit de installatie handleiding waren de oplossing:
```
mv /home/dsmr/.virtualenvs/dsmrreader /home/dsmr/.virtualenvs/dsmrreader-backup-old-python

virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3

/home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/base.txt -r /home/dsmr/dsmr-reader/dsmrreader/provisioning/requirements/postgresql.txt
```

Daarna kon ik met `sudo supervisorctl restart dsmr_webinterface` het webinterface process weer starten.

dennissiemensma, 2019-08-06:
Bedankt voor de info. Bij het wisselen van Python-versies is dat helaas iets wat moet gebeuren. Fijn dat je het tot zover zelf hebt kunnen oplossen. :]

Nvanderbeek, 2020-04-28:
> Nog een aanvulling...
>
> Na het draaien van `apt autoremove` was de oude python 3.5 verwijderd, maar faalde dsmr.
> Voor het gebruik van python 3.7 die er al wel op stond was het opnieuw bouwen van het [...]

top dit is nodig na een dist upgrade voor de web interface weer up en running te krijgen,