dsmrreader/dsmr-reader #669: Can't import my db

raoulteeuwen, 2019-07-17:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: latest (can't boot :-()
* Hardware *(RaspberryPi 2, 3 of iets anders)*: RaspberryPi 3b
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: v4

### Bug, wens of iets anders
* Installeer Raspbian Lite, installeer DSMR, import/restore backup > new telegrams don't import
* Ik had een situatie waarin mijn Rpi3 met DSMR niet meer wilde starten. Ik dacht: ik heb een backup, dus ik ga gewoon alles opnieuw installeren. Maar nu loop ik bij die import tegen een probleem op. Het zou kunnen dat ik niet vaak genoeg een dsmr update heb gedaan. Maar ik zou heel graag mijn backup kunnen gebruiken. Na veel uur klooien kom ik niet verder en trek ik toch maar de stoute schoenen aan en vraag of je me kunt en wil helpen.

in backend.log tref ik

"The current logging level only logs warnings and errors, to reduce I/O. More information can be found here: https://dsmr-reader.readthedocs$
[2019-07-17 20:05:07,384] ERROR    column dsmr_notification_notificationsetting.telegram_api_key does not exist
LINE 1: ...ication_notificationsetting"."pushover_user_key", "dsmr_noti...
                                                             ^
NoneType: None
[2019-07-17 20:05:08,651] ERROR    column dsmr_datalogger_meterstatistics.latest_telegram does not exist
LINE 1: ...datalogger_meterstatistics"."rejected_telegrams", "dsmr_data...
                                                             ^
NoneType: None
[2019-07-17 20:05:08,651] ERROR    column dsmr_notification_notificationsetting.telegram_api_key does not exist
LINE 1: ...ication_notificationsetting"."pushover_user_key", "dsmr_noti...
                                                             ^
NoneType: None
[2019-07-17 20:05:09,848] ERROR    column dsmr_datalogger_meterstatistics.latest_telegram does not exist
LINE 1: ...datalogger_meterstatistics"."rejected_telegrams", "dsmr_data...
"

(ik heb nog even overwogen te proberen zelf met psql die column toe proberen te voegen, maar ik krijg vanaf de CLI psql al niet werkend, ik strand op "psql: FATAL:  Peer authentication failed for user".

dennissiemensma, 2019-07-17:
Weet je zeker dat de applicatie de laatste versie is? 

Als ik de melding `column dsmr_datalogger_meterstatistics.latest_telegram does not exist` zie, dan lijkt er een kolom te missen. Deze is vrij recent toegevoegd, in DSMR-reader v2.1 twee maanden terug.

Welke versie staat er in `dsmrreader/__init__.py` op je Pi?

raoulteeuwen, 2019-07-17:
Dank voor de snelle reactie weer ... v2.2.1

(ik heb ondertussen alles opnieuw geinstalleerd en ga er van uit dat ik in het ergste geval alle historie kwijt ben (wat jammer zou zijn ivm vergelijken van jaar op jaar, maar geen wereldramp ;-)) 

Ik heb https://dsmr-reader.readthedocs.io/nl/latest/installation/quick.html gewoon gevolgd, en neem aan dat ik dan altijd de laatste versie zou moeten hebben  (ik installeerde nu al een keer of 3 op een versie SD met versie Raspbian Lite image)...

(Overigens had ik problemen met het (op basis van de docu) verwijderen van DSMR als ik probeerde eea te herstellen zonder een heel nieuw image op die SD te moeten zetten; kan het zijn dat die verwijder-instructies niet meer actueel zijn (want mss niet vaak gebruikt)?)

dennissiemensma, 2019-07-17:
Ik kan me niet voorstellen dat je data kwijt bent. In het ergste geval moeten we de belangrijkste data er even handmatig uithalen.

Kun je eens proberen om de back-up op een tijdelijke DB te restoren. Om te kijken of je dezelfde fouten krijgt:
```
sudo su - postgres
createdb temp_dsmrreader

# Afhankelijk van of je backup gecomprimeerd is of niet:
psql temp_dsmrreader -f <PATH-TO-POSTGRESQL-BACKUP.sql>

# Of
zcat <PATH-TO-POSTGRESQL-BACKUP.sql.gz> | psql temp_dsmrreader
```

Krijg je dezelfde fout?

raoulteeuwen, 2019-07-17:
Die import gaat goed, geen foutmeldingen ... is er nu een volgende stap om die data in de 'lopende omgeving' te krijgen?

dennissiemensma, 2019-07-17:
Ik denk dat je de applicatie moet stoppen, de DB verwijderen, opnieuw aanmaken en vervolgens daar in importeren. Zou zoiets moeten zijn:
```
sudo supervisorctl stop all

sudo su - postgres
dropdb dsmrreader
dropdb temp_dsmrreader

# Er vanuitgaande dat bij je eerdere installatie al een dsmrreader DB-user is.
createdb -O dsmrreader dsmrreader

# Let op, andere doel-DB:
psql dsmrreader -f <PATH-TO-POSTGRESQL-BACKUP.sql>
# Of
zcat <PATH-TO-POSTGRESQL-BACKUP.sql.gz> | psql dsmrreader

# Logout
CTRL + D

# Update nu de applicatie voor de zekerheid.
sudo su - dsmr
./deploy.sh

# Al dat goed gaat, alles weer starten.
CTRL + D
sudo supervisorctl start all
```

Wellicht kom je daar al verder mee.

raoulteeuwen, 2019-07-17:
De ./deploy.sh eindigt met

`Applying dsmr_backend.0003_scheduled_processes... OK
  Applying dsmr_backup.0004_custom_backup_path...Traceback (most recent call last):
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
psycopg2.errors.DuplicateColumn: column "folder" of relation "dsmr_backup_backupsettings" already exists

The above exception was the direct cause of the following exception:

Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/__init__.py", line 375, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 323, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 364, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/base.py", line 83, in wrapped
    res = handle_func(*args, **kwargs)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/management/commands/migrate.py", line 234, in handle
    fake_initial=fake_initial,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 117, in migrate
    state = self._migrate_all_forwards(state, plan, full_plan, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 147, in _migrate_all_forwards
    state = self.apply_migration(state, migration, fake=fake, fake_initial=fake_initial)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/executor.py", line 245, in apply_migration
    state = migration.apply(state, schema_editor)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/migration.py", line 124, in apply
    operation.database_forwards(self.app_label, schema_editor, old_state, project_state)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/migrations/operations/fields.py", line 112, in database_forwards
    field,
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/schema.py", line 447, in add_field
    self.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/base/schema.py", line 137, in execute
    cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 67, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 76, in _execute_with_wrappers
    return executor(sql, params, many, context)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)
django.db.utils.ProgrammingError: column "folder" of relation "dsmr_backup_backupsettings" already exists

 --- Checking & synchronizing static file changes.

0 static files copied to '/var/www/dsmrreader/static', 2046 unmodified.

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)   [??] PID file does not exist
 * Reloading process: dsmr_backend                   [??] PID file does not exist
 * Reloading process: dsmr_datalogger                [??] PID file does not exist
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist

 --- Clearing cache...

 --- Deployed version: 
2.2.1

 >>> Deployment complete. <<<
`

Ik heb daarom nog geen sudo supervisorctl start all gedaan. Wel proberen?

dennissiemensma, 2019-07-17:
Wat krijg je te zien bij dit command, staat er overal een kruisje voor?

```
sudo su - dsmr
./manage.py showmigrations
```

raoulteeuwen, 2019-07-17:
Zeker niet. Niet voor:

 [ ] 0004_custom_backup_path
 [ ] 0005_email_backup_settings
 [ ] 0006_scheduled_email_backup
 [ ] 0011_raw_telegram_insight
 [ ] 0015_notification_meta
 [ ] 0016_v200_release
 [ ] 0017_v201_release
 [ ] 0018_v202_release
 [ ] 0019_v210_release
 [ ] 0007_support_for_telegram

rest wel

raoulteeuwen, 2019-07-17:
(ik weet niet hoe laat jij normaal gaat slapen, maar het kan wachten hoor ... maar als jij door gaat, blijf ik ook online)

dennissiemensma, 2019-07-17:
Ik ga er zo vandoor, maar ben even aan het denken wat dit is. Normaal gesproken zou de DB-tabel waar alle migratie-statussen in staan in sync moeten zijn met de rest van de DB (een backup bevat die tabel dan ook). 
Maar het lijkt erop dat de DB-structuur nieuwer is dan de migratie-statussen. Dus moet je eigenlijk per migratie kijken of ze allemaal doorgevoerd zijn.

Ik denk dat je even de ouput van de volgende commands moet dumpen hier:

```
sudo su - postgres
psql dsmrreader

# 0004_custom_backup_path (kolom "folder" moet bestaan)
\dS dsmr_backup_backupsettings

# 0005_email_backup_settings (moet bestaan)
\dS dsmr_backup_emailbackupsettings

# 0006_scheduled_email_backup (moet 1 record hebben)
select * from dsmr_backend_scheduledprocess;

# 0011_raw_telegram_insight (kolom "latest_telegram" moet bestaan)
\dS dsmr_datalogger_meterstatistics

# 0007_support_for_telegram (kolommen "telegram_api_key" en "telegram_chat_id" moeten bestaan)
\dS dsmr_notification_notificationsetting
```

dennissiemensma, 2019-07-17:
Als alle wijzigingen al doorgevoerd zijn (zie comments hierboven), kun je alle migraties markeren als voltooid:

```
sudo su - dsmr
./manage.py migrate --fake
```

En daarna nogmaals proberen te updaten (hij zou niet meer moeten falen op de migraties).

Tot slot dan alles starten in supervisor en kijken of die draait. Je hebt dan nog steeds een database die "mwah" is, maar wellicht werkt dit al. Kijk in ieder geval of je al je data in het Archief nog hebt.

raoulteeuwen, 2019-07-17:
# 0004_custom_backup_path (kolom "folder" moet bestaan)
\dS dsmr_backup_backupsettings

                                        Table "public.dsmr_backup_backupsettings"
    Column     |           Type           | Collation | Nullable |                        Default                         
---------------+--------------------------+-----------+----------+--------------------------------------------------------
 id            | integer                  |           | not null | nextval('dsmr_backup_backupsettings_id_seq'::regclass)
 daily_backup  | boolean                  |           | not null | 
 compress      | boolean                  |           | not null | 
 backup_time   | time without time zone   |           | not null | 
 latest_backup | timestamp with time zone |           |          | 
 folder        | character varying(512)   |           | not null | 
Indexes:
    "dsmr_backup_backupsettings_pkey" PRIMARY KEY, btree (id)

# 0005_email_backup_settings (moet bestaan)
\dS dsmr_backup_emailbackupsettings

Did not find any relation named "dsmr_backup_emailbackupsettings".

# 0006_scheduled_email_backup (moet 1 record hebben)
select * from dsmr_backend_scheduledprocess;

 id | name | module | planned 
----+------+--------+---------
(0 rows)

# 0011_raw_telegram_insight (kolom "latest_telegram" moet bestaan)
\dS dsmr_datalogger_meterstatistics

                                              Table "public.dsmr_datalogger_meterstatistics"
          Column          |           Type           | Collation | Nullable |                           Default                           
--------------------------+--------------------------+-----------+----------+-------------------------------------------------------------
 id                       | integer                  |           | not null | nextval('dsmr_datalogger_meterstatistics_id_seq'::regclass)
 timestamp                | timestamp with time zone |           | not null | 
 electricity_tariff       | integer                  |           |          | 
 power_failure_count      | integer                  |           |          | 
 long_power_failure_count | integer                  |           |          | 
 voltage_sag_count_l1     | integer                  |           |          | 
 voltage_sag_count_l2     | integer                  |           |          | 
 voltage_sag_count_l3     | integer                  |           |          | 
 voltage_swell_count_l1   | integer                  |           |          | 
 voltage_swell_count_l2   | integer                  |           |          | 
 voltage_swell_count_l3   | integer                  |           |          | 
 rejected_telegrams       | integer                  |           | not null | 
 dsmr_version             | character varying(2)     |           |          | 
Indexes:
    "dsmr_datalogger_meterstatistics_pkey" PRIMARY KEY, btree (id)

# 0007_support_for_telegram (kolommen "telegram_api_key" en "telegram_chat_id" moeten bestaan)
\dS dsmr_notification_notificationsetting

                                            Table "public.dsmr_notification_notificationsetting"
        Column        |           Type           | Collation | Nullable |                              Default                              
----------------------+--------------------------+-----------+----------+-------------------------------------------------------------------
 id                   | integer                  |           | not null | nextval('dsmr_notification_notificationsetting_id_seq'::regclass)
 notification_service | character varying(20)    |           |          | 
 next_notification    | timestamp with time zone |           |          | 
 prowl_api_key        | character varying(64)    |           |          | 
 pushover_api_key     | character varying(64)    |           |          | 
 pushover_user_key    | character varying(64)    |           |          | 
Indexes:
    "dsmr_notification_notificationsetting_pkey" PRIMARY KEY, btree (id)

dennissiemensma, 2019-07-17:
Oke, het probleem lijkt dat in `0004_custom_backup_path` te zitten. Ik kijk even of het makkelijker is om alleen die migratie te markeren als voltooid, of om de kolom te verwijderen. Ik denk het eerste.

raoulteeuwen, 2019-07-17:
./deploy.sh (nav je 2de bericht zojuist) eindigt met

`Requirement already satisfied: MarkupSafe>=0.23 in /home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages (from jinja2->coreschema->coreapi==2.3.3->-r dsmrreader/provisioning/requirements/base.txt (line 2)) (1.1.1)

 --- Checking & synchronizing database changes/migrations.
Operations to perform:
  Apply all migrations: admin, auth, contenttypes, dsmr_api, dsmr_backend, dsmr_backup, dsmr_consumption, dsmr_datalogger, dsmr_frontend, dsmr_mindergas, dsmr_mqtt, dsmr_notification, dsmr_pvoutput, dsmr_stats, dsmr_weather, sessions
Running migrations:
  No migrations to apply.

 --- Checking & synchronizing static file changes.

0 static files copied to '/var/www/dsmrreader/static', 2046 unmodified.

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)   [??] PID file does not exist
 * Reloading process: dsmr_backend                   [??] PID file does not exist
 * Reloading process: dsmr_datalogger                [??] PID file does not exist
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist

 --- Clearing cache...

 --- Deployed version: 
2.2.1

 >>> Deployment complete. <<<
`

raoulteeuwen, 2019-07-17:
Ga jij er lekker vandoor, ga ik ook op 1 oor ... dank weer tot zover

dennissiemensma, 2019-07-17:
Ik denk dat je wat te snel was. Je mist nu migraties, want er was slechts 1 doorgevoerd. Maar dat komt morgen dan weer

dennissiemensma, 2019-07-17:
```
sudo su - postgres

update django_migrations set applied = null where app = 'dsmr_backup' and name = '0004_custom_backup_path';
update django_migrations set applied = null where app = 'dsmr_backup' and name = '0005_email_backup_settings';
update django_migrations set applied = null where app = 'dsmr_backup' and name = '0006_scheduled_email_backup';

update django_migrations set applied = null where app = 'dsmr_datalogger' and name = '0011_raw_telegram_insight';
update django_migrations set applied = null where app = 'dsmr_frontend' and name = '0015_notification_meta';
update django_migrations set applied = null where app = 'dsmr_frontend' and name = '0016_v200_release';
update django_migrations set applied = null where app = 'dsmr_frontend' and name = '0017_v201_release';
update django_migrations set applied = null where app = 'dsmr_frontend' and name = '0018_v202_release';
update django_migrations set applied = null where app = 'dsmr_frontend' and name = '0019_v210_release';
update django_migrations set applied = null where app = 'dsmr_notification' and name = '0007_support_for_telegram';
```

Daarna opnieuw updaten, zodat die deze migraties doorvoert.

raoulteeuwen, 2019-07-17:
Moest na de "sudo su - postgres" een "psql dsmrreader"? Als ik dat niet doe krijg ik een melding over een niet gevonden update commando. Als ik het wel doe krijg ik bij het 1ste statement

ERROR:  null value in column "applied" violates not-null constraint
DETAIL:  Failing row contains (99, dsmr_backup, 0005_email_backup_settings, null).

Weet niet of door gaan dan zin heeft?

dennissiemensma, 2019-07-18:
Ik heb wat foutjes gemaakt, de migratie-records moeten gewoon weggegooid worden bedenk ik me.

```
sudo su - postgres
psql dsmrreader

delete from django_migrations where app = 'dsmr_backup' and name = '0005_email_backup_settings';
delete from django_migrations where app = 'dsmr_backup' and name = '0006_scheduled_email_backup';
delete from django_migrations where app = 'dsmr_datalogger' and name = '0011_raw_telegram_insight';
delete from django_migrations where app = 'dsmr_frontend' and name = '0015_notification_meta';
delete from django_migrations where app = 'dsmr_frontend' and name = '0016_v200_release';
delete from django_migrations where app = 'dsmr_frontend' and name = '0017_v201_release';
delete from django_migrations where app = 'dsmr_frontend' and name = '0018_v202_release';
delete from django_migrations where app = 'dsmr_frontend' and name = '0019_v210_release';
delete from django_migrations where app = 'dsmr_notification' and name = '0007_support_for_telegram';
```

dennissiemensma, 2019-07-18:
Ik kan dit overigens niet zo makkelijk testen hier, maar als het nog niet werkt zal ik vanavond even de situatie nabootsen.

raoulteeuwen, 2019-07-18:
Dank. Na bovenstaande deletes eindigt deploy met

  Applying dsmr_notification.0007_support_for_telegram... OK

 --- Checking & synchronizing static file changes.

0 static files copied to '/var/www/dsmrreader/static', 2046 unmodified.

 --- Reloading running apps...

 * Reloading process: dsmr_webinterface (Gunicorn)   [??] PID file does not exist
 * Reloading process: dsmr_backend                   [??] PID file does not exist
 * Reloading process: dsmr_datalogger                [??] PID file does not exist
 * Reloading process: dsmr_mqtt                      [??] PID file does not exist

 --- Clearing cache...

 --- Deployed version: 
2.2.1

 >>> Deployment complete. <<< 

Ik heb daarna mijn Pi gereboot, en ik zie mijn archief-gegevens, en telegrams lijken te werken ... Vooralsnog: all goed. Hartelijk dank weer!

dennissiemensma, 2019-07-18:
Oke top, in dat geval zou ik het laten voor wat het is. Zorg er wel even voor dat je weer een tussentijdje backup even ergens wegzet.

closed 2019-07-18 by dennissiemensma