dsmrreader/dsmr-reader #670: Corrupte db? Admin interface opent niet meer.

marcfon, 2019-07-24:
Als ik op mijn admin interface (/admin/login/?next=/admin/) probeer in te loggen krijg ik sinds kort een foutmelding.

Even een snelle check in de postgresql log laat deze twee regels zien.

```2019-07-24 20:23:46 CEST [30621-1] dsmrreader@dsmrreader ERROR:  invalid page in block 0 of relation base/17024/17080```

```2019-07-24 20:23:46 CEST [30621-2] dsmrreader@dsmrreader STATEMENT:  SELECT (1) AS "a" FROM "django_session" WHERE "django_session"."session_key" = 'gl8axdieuv6wjjp92l479z9ahfy7zrqv' LIMIT 1```

Ik draai nog v1.27 en sync mijn backups naar dropbox maar die van de afgelopen week zijn allemaal 61 bytes :-/

Op deze foutmelding na lijkt alles nog perfect te draaien. De data logger loopt nog en de web interface doet ook zijn werk.

Heb al het een en ander zitten googlen maar kom er niet verder mee. Alle tips zijn van harte welkom.

marcfon, 2019-07-24:
Toch kunnen fixen met behulp van deze post.

https://stackoverflow.com/questions/5220344/postgresql-invalid-page-header-in-block

Probleem zat in deze twee tabellen ```public.dsmr_api_apisettings``` en ```public.django_migrations```.

marcfon, 2019-07-24:
Was net iets te snel. Toch nog 1 DB issue...

Ik kom nu wel het admin gedeelte in maar de meeste opties leveren deze foutmelding op.

```2019-07-24 21:16:33 CEST [2050-5] dsmrreader@dsmrreader ERROR:  invalid page in block 0 of relation base/17024/17070```

```2019-07-24 21:16:33 CEST [2050-6] dsmrreader@dsmrreader STATEMENT:  INSERT INTO "django_content_type" ("app_label", "model") VALUES ('dsmr_frontend', 'frontendsettings') RETURNING "django_content_type"."id"```

dennissiemensma, 2019-07-24:
Houd er rekening mee dat dit sowieso nooit hoort voor te komen. Het lijkt inderdaad op corruptie.

 Je kunt nog proberen te herstarten, maar ik kan je aanbevelen om eerst handmatig even een back-up proberen te maken.

Of om via de dropbox site te kijken of je nog een oude versie van een back-up kunt terughalen.

marcfon, 2019-07-24:
Het lijkt erop dat het alleen nog om ```django_content_type``` gaat. Zou ik alleen die tabel kunnen herstellen vanuit een verse install? 

Heb nog even gecheckt maar dropbox heeft geen voorgaande versies meer helaas.

dennissiemensma, 2019-07-24:
Lukt het je wel om een handmatig DB backup te maken dan?

marcfon, 2019-07-24:
Nope backup lukt ook niet.

```pg_dump: [archiver (db)] query failed: ERROR:  invalid page in block 0 of relation base/17024/17070
pg_dump: [archiver (db)] query was: SELECT sequence_name, start_value, increment_by, CASE WHEN increment_by > 0 AND max_value = 9223372036854775807 THEN NULL      WHEN increment_by < 0 AND max_value = -1 THEN NULL      ELSE max_value END AS max_value, CASE WHEN increment_by > 0 AND min_value = 1 THEN NULL      WHEN increment_by < 0 AND min_value = -9223372036854775807 THEN NULL      ELSE min_value END AS min_value, cache_value, is_cycled FROM public.django_content_type_id_seq```

dennissiemensma, 2019-07-24:
En wat als je de foute tabel exclude? Dat kan bij `pg_dump` met `-T django_content_type` volgens mij.

dennissiemensma, 2019-07-24:
En anders eventueel alleen de twee belangrijkste tabellen: 
```
pg_dump -t dsmr_stats_daystatistics -t dsmr_stats_hourstatistics dsmrreader --data-only
```

marcfon, 2019-07-24:
> En anders eventueel alleen de twee belangrijkste tabellen:

Deze dump lijkt wel te wel te werken. Geen errors in ieder geval. Maar heb wel het gevoel dat de dump wat klein is. Kan ik controleren dat het aantal records klopt met wat ik in de dashboard kan vinden?

dennissiemensma, 2019-07-24:
Deze twee tabellen zijn aggregaties van alle metingen en zijn de dag- en uurtotalen. Deze worden gebruikt in het archief.

Voor elke dag zou je 1 record moeten vinden in `dsmr_stats_daystatistics` en 24 of 25 records per dag in `dsmr_stats_hourstatistics`.

dennissiemensma, 2019-07-24:
Je zou dus ook nog kunnen proberen om een omgekeerde dump te maken, waarbij je dus alle tabellen exclude die fouten geeft. Maar bovenstaande twee zijn het belangrijkste voor je historie.

marcfon, 2019-07-24:
> Je zou dus ook nog kunnen proberen om een omgekeerde dump te maken, waarbij je dus alle tabellen uitsluit die fouten geven. Maar bovenstaande twee zijn het belangrijkste voor je historie.

Okay deze twee lijken dus wel correct te zijn. Omgekeerde dump wil ik zeker proberen.

Dit zijn ze?
```                               List of relations
 Schema |                      Name                       | Type  |   Owner    
--------+-------------------------------------------------+-------+------------
 public | auth_group                                      | table | dsmrreader
 public | auth_group_permissions                          | table | dsmrreader
 public | auth_permission                                 | table | dsmrreader
 public | auth_user                                       | table | dsmrreader
 public | auth_user_groups                                | table | dsmrreader
 public | auth_user_user_permissions                      | table | dsmrreader
 public | django_admin_log                                | table | dsmrreader
 public | django_content_type                             | table | dsmrreader
 public | django_migrations                               | table | dsmrreader
 public | django_session                                  | table | dsmrreader
 public | dsmr_api_apisettings                            | table | dsmrreader
 public | dsmr_backup_backupsettings                      | table | dsmrreader
 public | dsmr_backup_dropboxsettings                     | table | dsmrreader
 public | dsmr_consumption_consumptionsettings            | table | dsmrreader
 public | dsmr_consumption_electricityconsumption         | table | dsmrreader
 public | dsmr_consumption_energysupplierprice            | table | dsmrreader
 public | dsmr_consumption_gasconsumption                 | table | dsmrreader
 public | dsmr_datalogger_dataloggersettings              | table | dsmrreader
 public | dsmr_datalogger_dsmrreading                     | table | dsmrreader
 public | dsmr_datalogger_meterstatistics                 | table | dsmrreader
 public | dsmr_datalogger_retentionsettings               | table | dsmrreader
 public | dsmr_frontend_frontendsettings                  | table | dsmrreader
 public | dsmr_frontend_notification                      | table | dsmrreader
 public | dsmr_mindergas_mindergassettings                | table | dsmrreader
 public | dsmr_mqtt_jsondaytotalsmqttsettings             | table | dsmrreader
 public | dsmr_mqtt_jsontelegrammqttsettings              | table | dsmrreader
 public | dsmr_mqtt_message                               | table | dsmrreader
 public | dsmr_mqtt_mqttbrokersettings                    | table | dsmrreader```

dennissiemensma, 2019-07-24:
De output hier heeft er 47, je kunt ze hier vergelijken:

```
\dt
```

```
                               List of relations
 Schema |                      Name                       | Type  |   Owner    
--------+-------------------------------------------------+-------+------------
 public | auth_group                                      | table | dsmrreader
 public | auth_group_permissions                          | table | dsmrreader
 public | auth_permission                                 | table | dsmrreader
 public | auth_user                                       | table | dsmrreader
 public | auth_user_groups                                | table | dsmrreader
 public | auth_user_user_permissions                      | table | dsmrreader
 public | django_admin_log                                | table | dsmrreader
 public | django_content_type                             | table | dsmrreader
 public | django_migrations                               | table | dsmrreader
 public | django_session                                  | table | dsmrreader
 public | dsmr_api_apisettings                            | table | dsmrreader
 public | dsmr_backend_backendsettings                    | table | dsmrreader
 public | dsmr_backend_emailsettings                      | table | dsmrreader
 public | dsmr_backend_scheduledcall                      | table | dsmrreader
 public | dsmr_backend_scheduledprocess                   | table | dsmrreader
 public | dsmr_backup_backupsettings                      | table | dsmrreader
 public | dsmr_backup_dropboxsettings                     | table | dsmrreader
 public | dsmr_backup_emailbackupsettings                 | table | dsmrreader
 public | dsmr_consumption_consumptionsettings            | table | dsmrreader
 public | dsmr_consumption_electricityconsumption         | table | dsmrreader
 public | dsmr_consumption_energysupplierprice            | table | dsmrreader
 public | dsmr_consumption_gasconsumption                 | table | dsmrreader
 public | dsmr_datalogger_dataloggersettings              | table | dsmrreader
 public | dsmr_datalogger_dsmrreading                     | table | dsmrreader
 public | dsmr_datalogger_meterstatistics                 | table | dsmrreader
 public | dsmr_datalogger_retentionsettings               | table | dsmrreader
 public | dsmr_frontend_frontendsettings                  | table | dsmrreader
 public | dsmr_frontend_notification                      | table | dsmrreader
 public | dsmr_mindergas_mindergassettings                | table | dsmrreader
 public | dsmr_mqtt_jsondaytotalsmqttsettings             | table | dsmrreader
 public | dsmr_mqtt_jsontelegrammqttsettings              | table | dsmrreader
 public | dsmr_mqtt_message                               | table | dsmrreader
 public | dsmr_mqtt_mqttbrokersettings                    | table | dsmrreader
 public | dsmr_mqtt_rawtelegrammqttsettings               | table | dsmrreader
 public | dsmr_mqtt_splittopicdaytotalsmqttsettings       | table | dsmrreader
 public | dsmr_mqtt_splittopicmeterstatisticsmqttsettings | table | dsmrreader
 public | dsmr_mqtt_splittopictelegrammqttsettings        | table | dsmrreader
 public | dsmr_notification_notificationsetting           | table | dsmrreader
 public | dsmr_notification_statusnotificationsetting     | table | dsmrreader
 public | dsmr_pvoutput_pvoutputaddstatussettings         | table | dsmrreader
 public | dsmr_pvoutput_pvoutputapisettings               | table | dsmrreader
 public | dsmr_stats_daystatistics                        | table | dsmrreader
 public | dsmr_stats_electricitystatistics                | table | dsmrreader
 public | dsmr_stats_hourstatistics                       | table | dsmrreader
 public | dsmr_stats_note                                 | table | dsmrreader
 public | dsmr_weather_temperaturereading                 | table | dsmrreader
 public | dsmr_weather_weathersettings                    | table | dsmrreader
(47 rows)
```

dennissiemensma, 2019-07-24:
Je kunt dus in ieder geval `-T django_content_type` (exclude) proberen, als dat de enige tabel is waarop die stuk gaat.

marcfon, 2019-07-24:
Een omgekeerde dump lukt mits ik deze 4 tabellen niet meeneem ```django_migrations``` ```django_content_type``` ```dsmr_api_apisettings``` en ```dsmr_backup_backupsettings```.

Rest van de data lijkt dus in orde.

dennissiemensma, 2019-07-24:
Dat is mooi. Je kunt die back-up dan het beste bewaren als fallback, want ik denk dat je hem helaas in deze staat niet kan gebruiken voor een restore. 

De tabel `django_migrations` is namelijk belangrijk voor de applicatie omdat die aangeeft in welke staat de DB is. Bij een nieuwe installatie zal het project alle migraties opnieuw uitvoeren en daarmee ook in sommige tabellen data zetten. 
Als je dan daarna de back-up restore doet, vermoed ik dat je fouten krijg met dubbele of bestaande data. En als je eerste de back-up terugzet, vermoed ik dat de migraties niet uitgevoerd kunnen worden.

Je kunt het sowieso even uitproberen, maar als dat niet lukt, stel ik voor dat je nu eerst al even een tweede back-up maakt van alle data waarvan ik vrijwel zeker weet dat je geen conflicten gaat krijgen. Dat zijn deze tabellen:
```
dsmr_consumption_electricityconsumption
dsmr_consumption_gasconsumption
dsmr_consumption_energysupplierprice
dsmr_datalogger_dsmrreading
dsmr_stats_daystatistics
dsmr_stats_hourstatistics
dsmr_weather_temperaturereading
```

Deze zou je altijd moeten kunnen importeren, *mits je na herinstallatie de applicatie NIET meteen start* (maar pas na de import van deze data). Het importeren doe je dan pas na de herinstallatie en de back-up maak je met ` --data-only`.

Dit betekent wel dat al je instellingen (users, api-keys, externe koppelingen, mqtt) na herinstallatie weer uit/standaard zijn. Maar je hebt dan in ieder geval je data nog.

marcfon, 2019-07-24:
Okay, de tweede backup van de tabellen die je beschrijft is gemaakt.

Moet ik een volledige herinstall doen of eigenlijk alleen het DB gedeelte?

Na het aanamken van de nieuwe DB de backup op deze manier terugzetten?
```sudo sudo -u postgres psql dsmrreader -f <PATH-TO-POSTGRESQL-BACKUP.sql>```

dennissiemensma, 2019-07-25:
Ja op die manier kun je hem terugzetten/inlezen.

Herinstallatie of niet hangt erg af van of de rest van je filesystem ook corrupt is of niet. Je kunt eerst proberen om alleen de DB te vervangen:
```
sudo supervisorctl stop all

# Dubbel check dat je een goede backup hebt (die niet leeg is) voordat je alles weggooit!
sudo sudo -u postgres dropdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader

sudo su - dsmr
./manage.py migrate

# (Terug naar root/sudo user)
CTRL + D

# Importeer nu de backup met het eerdere command.

# Start de applicatie weer en kijk of dat goed is gegaan.
sudo supervisorctl start all
```
----

Werkt dat niet, dan herinstallatie tot **en met** `./manage.py migrate` in stap 8: https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#bootstrapping

Daarna DB dump inlezen en dan verder met de andere stappen uit de docs.

dennissiemensma, 2019-07-25:
Ik doe dit uit mijn hoofd overigens, want ik kan het niet makkelijk reproduceren, dus wellicht mist er ergens nog een command. Mocht je een fout krijgen, dan kijken we even verder.

marcfon, 2019-07-25:
De backup restore loopt nu. Volgens mij bijna 4 miljoen records dus dat gaat niet zo snel op een Pi2. Zal zo een update posten als die klaar is.

marcfon, 2019-07-25:
Import is gelukt en alles lijkt weer te draaien.

Moest alleen nog even de admin user aanmaken. Verder had je alles in 1x goed opgeschreven!
```./manage.py createsuperuser --username admin --email root@localhost```

Super bedankt Dennis!

dennissiemensma, 2019-07-25:
Dat is fijn om te horen!

closed 2019-07-25 by dennissiemensma