dsmrreader/dsmr-reader #671: Dashboard pagina laat geen waarden zien bij "High Tariff"

Rawh, 2019-07-27:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.2.1`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 2`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Bug, wens of iets anders
* Ik maak gebruik van Oxxio als leverancier voor energie + gas.
* Elke dag, bij de overschakeling van "low tariff" naar "high tariff" betekend dat de hoofdpagina van het Dashboard stopt met het laten zien van het overzicht van "high tariff". Wanneer ik naar de "Archive" pagina ga, dan zie ik echter wel de waardes die ik verwacht (en ook het overzicht).
[image]
[image]

Any clue what might be going on here?

dennissiemensma, 2019-07-27:
Bedankt voor je melding. Dit is mij nog niet eerder opgevallen. Ik zal er op een later moment dieper naar kijken.

dennissiemensma, 2019-07-28:
Speelt dit ook doordeweeks? In je screenshot van het dashboard staat zaterdag de 27e en het archief toont 1 dag eerder, de 26e.

Bij mij draait de slimme meter op zaterdag, zondag (en feestdagen) altijd op laag tarief en dan zie ik in het dashboard hetzelfde, want hoogtarief start pas weer op maandag.

Kan het zijn dat je hetzelfde ziet om dezelfde reden? Wellicht kun je doordeweeks even kijken of je dan wel wat ziet, want dat zou het kunnen verklaren.

Rawh, 2019-07-28:
Ah ik dacht dat laag tarief altijd 's avonds speelde, niet ook het hele weekend.

Ik zal a.s. maandag eens afwachten en kijken wat ie door de week doet.

Rawh, 2019-07-29:
@dennissiemensma Het ziet er idd naar uit dat het een niet alleen 'avond' low-tariff dingetje is, maar ook een tariff die voor het weekend geld. Weer wat geleerd :)

closed 2019-07-29 by Rawh