dsmrreader/dsmr-reader #675: New installation with raspbian buster is not working

Robert126, 2019-07-30:
New installation on a RPI3 

Following your great installation guide:
https://dsmr-reader.readthedocs.io/en/latest/installation/quick.html

Unfortunately at first command 
I'm receiving the following error:
sudo apt-get install -y postgresql postgresql-server-dev-all nginx supervisor git python3 python3-pip python3-virtualenv virtualenvwrapper
Reading package lists... Done
Building dependency tree       
Reading state information... Done
git is already the newest version (1:2.20.1-2).
python3 is already the newest version (3.7.3-1).
python3 set to manually installed.
The following package was automatically installed and is no longer required:
  rpi.gpio-common
Use 'sudo apt autoremove' to remove it.
The following additional packages will be installed:
  binfmt-support clang-7 dctrl-tools dh-python fontconfig-config
  fonts-dejavu-core gir1.2-glib-2.0 libclang-common-7-dev libclang1-7
  libexpat1-dev libffi-dev libfontconfig1 libgc1c2 libgd3
  libgirepository-1.0-1 libjbig0 libjs-sphinxdoc libjs-underscore libllvm7
  libncurses-dev libnginx-mod-http-auth-pam libnginx-mod-http-dav-ext
  libnginx-mod-http-echo libnginx-mod-http-geoip
  libnginx-mod-http-image-filter libnginx-mod-http-subs-filter
  libnginx-mod-http-upstream-fair libnginx-mod-http-xslt-filter
  libnginx-mod-mail libnginx-mod-stream libobjc-8-dev libobjc4 libomp-7-dev
  libomp5-7 libpq-dev libpq5 libpython3-dev libpython3.7-dev libsensors-config
  libsensors5 libtiff5 libtinfo-dev libwebp6 libxpm4 libxslt1.1 llvm-7
  llvm-7-dev llvm-7-runtime nginx-common nginx-full postgresql-11
  postgresql-client-11 postgresql-client-common postgresql-common
  postgresql-server-dev-11 python-meld3 python-pbr python-pip-whl
  python-pkg-resources python-setuptools python-six python-stevedore
  python-virtualenv python-virtualenv-clone python3-asn1crypto
  python3-cffi-backend python3-crypto python3-cryptography python3-dbus
  python3-dev python3-distutils python3-entrypoints python3-gi python3-keyring
  python3-keyrings.alt python3-lib2to3 python3-secretstorage
  python3-setuptools python3-wheel python3-xdg python3.7-dev ssl-cert sysstat
  virtualenv virtualenv-clone
Suggested packages:
  clang-7-doc debtags libgd-tools ncurses-doc libomp-7-doc postgresql-doc-11
  lm-sensors llvm-7-doc fcgiwrap nginx-doc postgresql-doc libjson-perl
  python-setuptools-doc python-crypto-doc python-cryptography-doc
  python3-cryptography-vectors python-dbus-doc python3-dbus-dbg gnome-keyring
  libkf5wallet-bin gir1.2-gnomekeyring-1.0 python-secretstorage-doc
  openssl-blacklist supervisor-doc isag
The following NEW packages will be installed:
  binfmt-support clang-7 dctrl-tools dh-python fontconfig-config
  fonts-dejavu-core gir1.2-glib-2.0 libclang-common-7-dev libclang1-7
  libexpat1-dev libffi-dev libfontconfig1 libgc1c2 libgd3
  libgirepository-1.0-1 libjbig0 libjs-sphinxdoc libjs-underscore libllvm7
  libncurses-dev libnginx-mod-http-auth-pam libnginx-mod-http-dav-ext
  libnginx-mod-http-echo libnginx-mod-http-geoip
  libnginx-mod-http-image-filter libnginx-mod-http-subs-filter
  libnginx-mod-http-upstream-fair libnginx-mod-http-xslt-filter
  libnginx-mod-mail libnginx-mod-stream libobjc-8-dev libobjc4 libomp-7-dev
  libomp5-7 libpq-dev libpq5 libpython3-dev libpython3.7-dev libsensors-config
  libsensors5 libtiff5 libtinfo-dev libwebp6 libxpm4 libxslt1.1 llvm-7
  llvm-7-dev llvm-7-runtime nginx nginx-common nginx-full postgresql
  postgresql-11 postgresql-client-11 postgresql-client-common
  postgresql-common postgresql-server-dev-11 postgresql-server-dev-all
  python-meld3 python-pbr python-pip-whl python-pkg-resources
  python-setuptools python-six python-stevedore python-virtualenv
  python-virtualenv-clone python3-asn1crypto python3-cffi-backend
  python3-crypto python3-cryptography python3-dbus python3-dev
  python3-distutils python3-entrypoints python3-gi python3-keyring
  python3-keyrings.alt python3-lib2to3 python3-pip python3-secretstorage
  python3-setuptools python3-virtualenv python3-wheel python3-xdg
  python3.7-dev ssl-cert supervisor sysstat virtualenv virtualenv-clone
  virtualenvwrapper
[... 590 lines cut ...]
dpkg: error processing package nginx-full (--configure):
 installed nginx-full package post-installation script subprocess returned error exit status 1
dpkg: dependency problems prevent configuration of nginx:
 nginx depends on nginx-full (<< 1.14.2-2.1~) | nginx-light (<< 1.14.2-2.1~) | nginx-extras (<< 1.14.2-2.1~); however:
  Package nginx-full is not configured yet.
  Package nginx-light is not installed.
  Package nginx-extras is not installed.
 nginx depends on nginx-full (>= 1.14.2-2) | nginx-light (>= 1.14.2-2) | nginx-extras (>= 1.14.2-2); however:
  Package nginx-full is not configured yet.
  Package nginx-light is not installed.
  Package nginx-extras is not installed.

dpkg: error processing package nginx (--configure):
 dependency problems - leaving unconfigured
Errors were encountered while processing:
 nginx-full
 nginx
E: Sub-process /usr/bin/dpkg returned an error code (1)
pi@raspberrypi:~ $ 

Hopefully some one can help me out!

On forehand thank you for your help!

Robert

dennissiemensma, 2019-07-30:
Bedankt voor je melding. Weet je toevallig ook welk OS en welke versie van het OS je hebt geinstalleerd op de Pi?

Robert126, 2019-07-31:
Goedemorgen Dennis,

Ik heb een nieuwe installatie uitgevoerd nu werkt alles naar behoren.
Ik had ook Pi-Hole geïnstalleerd op dezelfde RPi dit gaf het conflict met poort 80. 

Nu heb ik eerst DSMR Reader geïnstalleerd (de poort aangepast) en daarna Pi-Hole.

Dank voor de snelle reactie! en je mooie programma!

Hartelijke groet,

Robert

dennissiemensma, 2019-07-31:
Fijn om te horen, bedankt voor je terugkoppeling!

closed 2019-07-31 by dennissiemensma

Robert126, 2019-08-05:
Hi Denis,

Sorry voor mijn late reactie!

Je vroeg me nog naar de versies welke ik gebruik:
https://www.raspberrypi.org/downloads/raspbian/
Raspbian Buster Lite

Software Version:
pi@raspberrypi:~ $ cat /etc/os-release
PRETTY_NAME="Raspbian GNU/Linux 10 (buster)"
NAME="Raspbian GNU/Linux"
VERSION_ID="10"
VERSION="10 (buster)"
VERSION_CODENAME=buster
ID=raspbian
ID_LIKE=debian
HOME_URL="http://www.raspbian.org/"
SUPPORT_URL="http://www.raspbian.org/RaspbianForums"
BUG_REPORT_URL="http://www.raspbian.org/RaspbianBugs"

Kernel Version:
pi@raspberrypi:~ $ uname -a
Linux raspberrypi 4.19.58-v7+ #1245 SMP Fri Jul 12 17:25:51 BST 2019 armv7l GNU/Linux

Hardware version:
RPi 3 Model B, 1GB 
pi@raspberrypi:~ $ cat /proc/cpuinfo
processor	: 0
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 1
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 2
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

processor	: 3
model name	: ARMv7 Processor rev 4 (v7l)
BogoMIPS	: 76.80
Features	: half thumb fastmult vfp edsp neon vfpv3 tls vfpv4 idiva idivt vfpd32 lpae evtstrm crc32 
CPU implementer	: 0x41
CPU architecture: 7
CPU variant	: 0x0
CPU part	: 0xd03
CPU revision	: 4

Hardware	: BCM2835
Revision	: a22082
Serial		: 00000000c3c8f6c9

Als ik nog ergens anders mee kan helpen hoor ik het graag.

Hartelijke groet,

Robert

dennissiemensma, 2019-08-05:
Bedankt voor je aanvulling. Ik zal deze versie van de distro binnenkort eens proberen.