dsmrreader/dsmr-reader #676: Tarieven energieleverancier na inlezen backup verdwenen + foutmelding

T1rnanog, 2019-07-30:
Ha, 
Ik heb een verse installatie gedaan en hierin een backup terug gezet. 
Alles werkt alleen zag ik dat mijn energie tarieven niet meer gevuld waren.
Toen ik ze opnieuw probeerde te vullen kreeg ik de volgende melding.

Wat kan ik doen om dit op te lossen? 
Alvast bedankt

http://192.168.2.66/admin/dsmr_consumption/energysupplierprice/add/

Server Error
Sorry, something unexpected happened.

Exception:
DataError: character with byte sequence 0xe2 0x80 0x8b in encoding "UTF8" has no equivalent in encoding "LATIN1"

Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/options.py", line 606, in wrapper
    return self.admin_site.admin_view(view)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/decorators.py", line 142, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/views/decorators/cache.py", line 44, in _wrapped_view_func
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/sites.py", line 223, in inner
    return view(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/options.py", line 1634, in add_view
    return self.changeform_view(request, None, form_url, extra_context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/decorators.py", line 45, in _wrapper
    return bound_method(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/utils/decorators.py", line 142, in _wrapped_view
    response = view_func(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/options.py", line 1522, in changeform_view
    return self._changeform_view(request, object_id, form_url, extra_context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/options.py", line 1561, in _changeform_view
    self.save_model(request, new_object, form, not add)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/contrib/admin/options.py", line 1088, in save_model
    obj.save()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 741, in save
    force_update=force_update, update_fields=update_fields)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 779, in save_base
    force_update, using, update_fields,

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 870, in _save_table
    result = self._do_insert(cls._base_manager, using, fields, update_pk, raw)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/base.py", line 908, in _do_insert
    using=using, raw=raw)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/query.py", line 1186, in _insert
    return query.get_compiler(using=using).execute_sql(return_id)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/models/sql/compiler.py", line 1335, in execute_sql
    cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 67, in execute
    return self._execute_with_wrappers(sql, params, many=False, executor=self._execute)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 76, in _execute_with_wrappers
    return executor(sql, params, many, context)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.7/site-packages/django/db/backends/utils.py", line 84, in _execute
    return self.cursor.execute(sql, params)

dennissiemensma, 2019-07-30:
Bedankt voor je melding. Welke output krijg je te zien bij uitvoeren van:
```
sudo sudo -u postgres psql --list
```

T1rnanog, 2019-08-01:
List of databases
    Name    |   Owner    | Encoding | Collate | Ctype |   Access privileges
------------+------------+----------+---------+-------+-----------------------
 dsmrreader | dsmrreader | LATIN1   | nl_NL   | nl_NL |
 postgres   | postgres   | LATIN1   | nl_NL   | nl_NL |
 template0  | postgres   | LATIN1   | nl_NL   | nl_NL | =c/postgres          +
            |            |          |         |       | postgres=CTc/postgres
 template1  | postgres   | LATIN1   | nl_NL   | nl_NL | =c/postgres          +
            |            |          |         |       | postgres=CTc/postgres
(4 rows)

dennissiemensma, 2019-08-01:
Het lijkt erop dat je systeem gebruik maakt van andere locales en encoding dan de standaard UTF8.

In het gunstigste geval kun je een UTF8-database ernaast maken, maar wellicht is er meer voor nodig. Kun je eens proberen of dit werkt als test?

```
sudo sudo -u postgres createdb -E utf8 -O dsmrreader testdsmr
```

Mocht dat geen fouten geven, check het  overzicht opnieuw met:
```
sudo sudo -u postgres psql --list
```

Vervolgstappen zijn afhankelijk van je bevindingen.

T1rnanog, 2019-08-01:
Bedankt voor je hulp hier zit zeker het probleem, had nl_NL.UTF-8 verwacht maar helaas.

pi@raspberrypi:~ $ locale
LANG=nl_NL
LANGUAGE=nl_NL
LC_CTYPE="nl_NL"
LC_NUMERIC="nl_NL"
LC_TIME="nl_NL"
LC_COLLATE="nl_NL"
LC_MONETARY="nl_NL"
LC_MESSAGES="nl_NL"
LC_PAPER="nl_NL"
LC_NAME="nl_NL"
LC_ADDRESS="nl_NL"
LC_TELEPHONE="nl_NL"
LC_MEASUREMENT="nl_NL"
LC_IDENTIFICATION="nl_NL"
LC_ALL=nl_NL
pi@raspberrypi:~ $ sudo sudo -u postgres psql --list
                              List of databases
    Name    |   Owner    | Encoding | Collate | Ctype |   Access privileges
------------+------------+----------+---------+-------+-----------------------
 dsmrreader | dsmrreader | LATIN1   | nl_NL   | nl_NL |
 postgres   | postgres   | LATIN1   | nl_NL   | nl_NL |
 template0  | postgres   | LATIN1   | nl_NL   | nl_NL | =c/postgres          +
            |            |          |         |       | postgres=CTc/postgres
 template1  | postgres   | LATIN1   | nl_NL   | nl_NL | =c/postgres          +
            |            |          |         |       | postgres=CTc/postgres
(4 rows)

pi@raspberrypi:~ $ sudo sudo -u postgres createdb -E utf8 -O dsmrreader testdsmr                                                                    createdb: database creation failed: ERROR:  encoding "UTF8" does not match locale "nl_NL"
DETAIL:  The chosen LC_CTYPE setting requires encoding "LATIN1".

dennissiemensma, 2019-08-02:
Welke distro gebruik je als OS? Ik denk dat je het beste kan proberen om `LC_CTYPE` aan te passen, zoals in https://askubuntu.com/a/413006. Zelf kan ik niet helaas meer voor je doen dan alleen een beetje mee-googlen.

Zodra je een database kan aanmaken in UTF-8, kan ik je weer verder helpen.

T1rnanog, 2019-08-05:
Bedankt voor je hulp zover, het had helaas geen effect. 
ik had met nano /etc/default/locale het volgende gedaan, daarna een reboot.

LANG=nl_NL.UTF-8
LC_ALL=nl_NL.UTF-8
LANGUAGE=nl_NL.UTF-8

Ik kwam dit nog tegen.

https://stackoverflow.com/questions/5090858/how-do-you-change-the-character-encoding-of-a-postgres-database

Als ik dit volg moet ik dan na deze stappen opnieuw een database backup terug zetten? 

I would rather rebuild everything from the begining properly with a correct local encoding on your debian OS as explained here :

su root
Reconfigure your local settings :

dpkg-reconfigure locales
Choose your locale (like for instance for french in Switzerland : fr_CH.UTF8)

Uninstall and clean properly postgresql :

apt-get --purge remove postgresql\*
rm -r /etc/postgresql/
rm -r /etc/postgresql-common/
rm -r /var/lib/postgresql/
userdel -r postgres
groupdel postgres
Re-install postgresql :

aptitude install postgresql-9.1 postgresql-contrib-9.1 postgresql-doc-9.1
Now any new database will be automatically be created with correct encoding, LC_TYPE (character classification), and LC_COLLATE (string sort order).

T1rnanog, 2019-08-06:
Bovenstaande mag je negeren, heb hem weer helemaal stuk gemaakt ^^

Ik gebruik Raspbian Buster with desktop and recommended software.
Vervolgens zet ik de SSH aan, en ga de DSMR 'Installatie: Toegelicht' volgen zodat ik langs het stukje backup inlezen kom.

Kun je wellicht aangeven waar het fout is gegaan met het volgen van de installatie handleiding, door dat de locale verkeerd stonden. Dan ga ik het vanavond opnieuw proberen. Maar dan moet ik wel in UTF8 modus zitten :D

Alvast bedankt.

dennissiemensma, 2019-08-06:
Ik vind het lastig te zeggen hoe dit exact op te lossen voor Nederlands. Ik heb wel ervaring op Ubuntu met `dpkg-reconfigure locales`, maar toen gebruikte ik iets als `en_us.UTF8`.

Het is niet iets wat je fout hebt gedaan bij de installatie van DSMR-reader zelf, want ik ben dit nog niet eerder tegengekomen. Verder heb ik Buster zelf nog niet geprobeerd.

Je kunt nog proberen met: `sudo raspi-config`. En dan Optie 4 > Optie 1.

T1rnanog, 2019-08-09:
Zojuist een nieuwe installatie gedaan. 

Bij het koppelen van de Pi aan mijn televisie om vervolgens SSH aan te zetten. Wordt gevraagd door een paar instellingen te lopen. Zo ook weer Nederland gekozen etc.
Hier kon ik vervolgens onder dit venster een vinkje zetten voor 'De Engelste taal gebruiken' dit gedaan. Volgens mij heb ik dit de vorige keer niet gedaan. 

Als ik nu locale type zie ik 

pi@raspberrypi:~ $ locale
LANG=en_US.UTF-8
LANGUAGE=en_US.UTF-8
LC_CTYPE="en_US.UTF-8"
LC_NUMERIC="en_US.UTF-8"
LC_TIME="en_US.UTF-8"
LC_COLLATE="en_US.UTF-8"
LC_MONETARY="en_US.UTF-8"
LC_MESSAGES="en_US.UTF-8"
LC_PAPER="en_US.UTF-8"
LC_NAME="en_US.UTF-8"
LC_ADDRESS="en_US.UTF-8"
LC_TELEPHONE="en_US.UTF-8"
LC_MEASUREMENT="en_US.UTF-8"
LC_IDENTIFICATION="en_US.UTF-8"
LC_ALL=en_US.UTF-8

Nu vervolgens ook mijn tarieven ingevuld alles ging goed.

De Tarief energieleverancier 'Energiedirect' is met succes toegevoegd.

Bedankt voor alle hulp ticket mag gesloten worden :D

closed 2019-08-09 by T1rnanog

dennissiemensma, 2019-08-09:
Fijn om te horen, bedankt voor je terugkoppeling!