dsmrreader/dsmr-reader #677: Intergratie met DSMRloggerWS

Wouter1702, 2019-07-30:
Beste Dennis,

Ik heb een DSMR-logger die ik graag wil intergreren met DSMR-reader.
Ik heb de DSMR-logger v4 van Willem Anderwiel. Dit is een hardware logger op basis van een ESP-12 processorboard die jezelf in elkaar kunt zetten. Deze logger ogt alle telegram berichten die uit mijn meter komen. De logger bewaart een aantal statistieken en ik kan de logger via WiFi benaderen. via een WebUI kan ik de statistieken uitlezen. De DSMR-logger snapt een paar restAPI calls en dus kan via een restAPI de actuele data terug sturen in json formaat.

Documentatie over de logger kun je hier vinden:
- https://mrwheel.github.io/DSMRloggerWS
- https://mrwheel.github.io/DSMRloggerWS/restAPI/
- https://opencircuit.nl/Blog/DSMR-logger-V4-Slimme-Meter-uitlezer/

Ik wilde een hardware oplossing hebben voor het loggen van de telegram berichten. Daarnaast heb ik je DSMR-reader bij iemand gezien en vind het een goede presentatie van de gegevens.
Ik wilde deze twee met elkaar laten communiceren echter ik zie dat je enkel input kunt configureren die van een usb port komt.

Is er een mogelijkheid om deze twee produkten met elkaar te laten "kletsen"

met vriendelijke groet,

Wouter

dennissiemensma, 2019-07-30:
Bedankt voor je vraag. Je bent niet gebonden aan input van een USB-poort. Er is tevens een API in DSMR-reader: https://dsmr-reader.readthedocs.io/en/latest/api.html

Ik denk dat de beste optie is om een script te maken die de huidige restAPI van mrWheel uitleest en de waarden vervolgens doorpost naar de v2 REST API van DSMR-reader.

Meer info staat hier: https://dsmr-reader.readthedocs.io/en/latest/api.html#id6

Je zou zo'n scriptje gewoon kunnen laten draaien op dezelfde hardware als waar DSMR-reader op staat (zolang die maar bij beide API's kan).

dennissiemensma, 2019-07-30:
Voorbeeld van een scriptje (die welliswaar USB uitleest, maar kan ook een API zijn), staat hier: https://dsmr-reader.readthedocs.io/en/latest/api.html#script

dennissiemensma, 2019-08-09:
Mocht je hier niet zelf uitkomen, dan hoor ik het graag.

Wouter1702, 2019-08-09:
Beste dennis,

Ik ben nog niet zo’n programmeer held dus ik vind het inderdaad lastig.
Ik kan hier zeker wel wat hulp bij gebruiken.
Op dit moment ben ik niet thuis en geniet lekker van een vakantie maar zodra ik terug ben, dan neem ik weer contact met je op.
Alvast bedankt voor het aanbod

Met vriendelijke groet,

Wouter Iliohan

dennissiemensma, 2019-09-11:
@Wouter1702 ben je hier nog mee verder gekomen?

Wouter1702, 2019-09-11:
Hallo Dennis,

Nee helaas niet maar ik ga het anders doen. De logger ondersteund nu MQTT dus ik wilde het via die weg eerst nog even proberen.

Als ik mij niet vergis ondersteund dsmr-reader ook MQTT dus dat ga ik eerst even proberen.

Nog bedankt dat je aan me dacht.

Met vriendelijke groet,

Wouter Iliohan

Dit bericht is verstuurd vanaf een mobiel device.

dennissiemensma, 2019-09-19:
Mocht je later hulp nodig hebben, dan hoor ik het graag.

closed 2019-09-19 by dennissiemensma