dsmrreader/dsmr-reader #680: DSMR backup fails

tomdh76, 2019-08-03:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie 2.2
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3.`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v...`

### Bug, wens of iets anders
Ik moest vanwege een crash van mijn RPI DSMR opnieuw installeren. Ik dacht met dropbox backups snel weer gegevens te kunnen terugzetten maar helaas werkt het niet. Ik heb neem ik aan de backup's gemaakt met dezelfde versie van DSMR want de backup is tot een dag geleden gemaakt. Ik volg https://dsmr-reader.readthedocs.io/nl/latest/faq.html#how-do-i-restore-a-database-backup. Het begin lijkt goed te gaan ; ik krijg onder elkaar:

CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE
ALTER TABLE
ALTER SEQUENCE
CREATE TABLE
ALTER TABLE
CREATE SEQUENCE

COPY 1
 setval
--------
      1
(1 row)

COPY 240763
 setval
--------
 241241
(1 row)

COPY 1
 setval
--------
      1
(1 row)

COPY 4014
 setval
--------
   4014
(1 row)

COPY 1
 setval
--------
      1
(1 row)

Maar dan foutmelding opeens:

ERROR:  invalid input syntax for type timestamp with time zone: "201"
CONTEXT:  COPY dsmr_datalogger_dsmrreading, line 1096336, column timestamp: "201"

Maar de foutmelding kan ook dit zijn:

psql:/home/pi/Downloads/dsmrreader-postgresql-backup-Tuesday.sql:1458960: ERROR:  invalid input syntax for type numeric: ""
CONTEXT:  COPY dsmr_datalogger_dsmrreading, line 1206034, column electricity_delivered_2: ""

Ik heb nu een stuk of 4 backups geprobeerd en dus telkens een error... 

Ben ik alle data kwijt?

Ik heb ook een keer geprobeerd: 
Mocht de versie toch verschillen, dan kun je proberen om de applicatie te herladen. Log in met: sudo su - dsmr en voer vervolgens ./post-deploy.sh uit.

Maar dit werkte ook niet.

dennissiemensma, 2019-08-03:
Bedankt voor je vraag. Hoe groot is je backup ongeveer? Ik zou er evt een blik op kunnen werpen, misschien staat er nog genoeg leesbare data in.

tomdh76, 2019-08-04:
Dat is een snelle reactie! De backup is 12 mb. Hierbij de link:

https://1drv.ms/u/s!AnF_YnMTNR-d3v50QO1NPckD2Oo07Q?e=qYfsis

Hopelijk kun je er iets mee. Zo niet dan ga ik opnieuw een database aanleggen (maar hopelijk kan ik dan wel een backup terugzetten in de toekomst) Ik vind de maanden dat ik DSMR reader heb gebruik het wel een mooi programma. Gr Tom

dennissiemensma, 2019-08-04:
Bedankt voor het delen. Ik kan je aanraden om de volgende keer even te wachten met het commenten van links hier, want dan kun je die mij beter mailen.

Je moet nu eerst direct in Dropbox de 'app' aanpassen die je gebruikt voor de DSMR-backups. Genereer daar een nieuwe sleutel/token voor in DSMR-reader straks, want de huidige zit in je backup en die heb je nu openbaar gemaakt door de link hier publiekelijk te delen.

dennissiemensma, 2019-08-04:
Ik heb mogelijk wat data gevonden in de backup waarmee een deel weer hersteld kan worden. Heb je een e-mailadres waarop ik je de data kan delen?

dennissiemensma, 2019-08-04:
Ik zag dat de backup de metingen had tot 23 april. Daarna kapte die af, alsof de backup of het bestand onderbroken was. 

Ik zag echter nog wel gegevens die afgeleid waren van alle metingen en die lijken tot 31 juli te gaan. Dus als die gegevens te importeren zijn, is de kans groot dat de applicatie alsnog met terugwerkende kracht het Archief kan hergenereren en dan ben je alleen afgelopen 4 of 5 dagen kwijt.

tomdh76, 2019-08-04:
Dank voor het werk! En voor de tip om niet de link hier te zetten. Ik heb een email maar die zal ik ook maar niet hier zetten. Ik ben niet zo bekend met github. Even kijken hoe ik je mijn email kan doorsturen?

dennissiemensma, 2019-08-04:
Stuur anders een e-mail naar github at dennissiemensma.nl

dennissiemensma, 2019-08-04:
Je kunt nu het beste de applicatie stoppen, je huidige DB backuppen (als er wat in zit), daarna de DB opnieuw aanmaken, initialiseren en dan pas de data-export weer inladen. Tot slot de applicatie starten.

```
sudo supervisorctl stop all
sudo sudo -u postgres pg_dump dsmrreader | gzip --fast > temp_db_dump.sql.gz
sudo sudo -u postgres dropdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader

sudo su - dsmr
./manage.py migrate
CTRL + D

# Hier laad je de gegevens uit de dump in.
zcat data-export.sql.gz | sudo sudo -u postgres psql dsmrreader

# Gelukt zonder fouten? Start dan een deel weer.

sudo supervisorctl start dsmr_webinterface dsmr_backend
```

Als het goed is zie je dan langzaam in het Archief weer iets verschijnen. Kan even duren.

Wanneer dat ook goed is gegaan, kun je de datalogger weer aanzetten.
```
sudo supervisorctl start dsmr_datalogger
```

Het kan zijn dat er toch iets misgaat in al deze stappen. Ben benieuwd.

dennissiemensma, 2019-08-05:
De just de nieuwe dump straks gewoon importeren over je bestaande database. Ik heb wel alle metingen er uitgehaald, maar daarbij heb je wel weer alle uur- en dagtotalen.

```
# Importeer data.
sudo sudo -u postgres psql -f hour_day_statistics.sql dsmrreader

# Ga nu in de DB.
sudo sudo -u postgres psql dsmrreader

# Voer nu deze queries uit:
SELECT setval(pg_get_serial_sequence('"dsmr_stats_daystatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_stats_daystatistics";
SELECT setval(pg_get_serial_sequence('"dsmr_stats_hourstatistics"','id'), coalesce(max("id"), 1), max("id") IS NOT null) FROM "dsmr_stats_hourstatistics";

# Je kunt daarna uit de DB met CTRL +D
```

closed 2019-09-03 by dennissiemensma