dsmrreader/dsmr-reader #682: HOOGTE- EN LAAGTE RECORDS

bramkragten, 2019-08-06:
### Jouw omgeving
* DSMR-reader versie: `v2.2.3`
* Hardware: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter: `v4`
* Aantal fasen: `1`

### Wens / vraag
Vroeger kon ik in statistieken mijn laagste verbruik zien, dit vond ik handig om mijn sluik vebruik te kunnen meten en bijhouden.

Sinds een aantal versies krijg ik de volgende melding te zien:

> Vink "Houd elektriciteitsfasen bij" aan, in de Dataloggerconfiguratie, om deze gegevens bij te houden.

Bij deze vink in de settings staat echter dat ik dit alleen moet aanzetten als ik 3 fases heb, maar ik heb maar 1 fase.

> Geeft aan of we de elektriciteitsfasen moeten bijhouden (wanneer aanwezig). Standaard heeft iedereen één fase thuis, maar sommige slimme meters zijn aangesloten op drie fasen. Bijvoorbeeld voor zonnepanelen of elektrisch koken. Deze feature is alleen nuttig (en werkt tevens alleen) wanneer er daadwerkelijk drie fasen aanwezig zijn. Het dashboard toont deze gegevens wanneer dit ingeschakeld is.

Waarom zijn de hoogte en laagte records gekoppeld aan 3 fases? En is er een manier waarop ik dit wel aankrijg met maar 1 fase?

dennissiemensma, 2019-08-06:
Bedankt voor je melding. Het is meer een technische keuze aan mijn zijde geweest bij het herschrijven hiervan, zonder aan deze situatie gedacht te hebben. 

In dit geval kun je gewoon `Houd elektriciteitsfasen bij` aanzetten, ook al heb je er slechts één. Als het goed is wordt de hoogste en laagste vervolgens bijgehouden.

Ik zal de tekst wat aanpassen, want oorspronkelijk was het niet nuttig om deze optie aan te zetten.

dennissiemensma, 2019-08-09:
Ik heb de omschrijving iets aangepast voor de volgende release. Bedankt voor het melden.

closed 2019-08-09 by dennissiemensma

dennissiemensma, 2019-09-03:
Onderdeel van nieuwe release [v2.3](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.3.0).