dsmrreader/dsmr-reader #683: No PVoutput.org data

lukevredeveld, 2019-08-06:
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.2.3`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3b`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4`

### Bug, wens of iets anders
Ik krijg geen data binnen van PVoutput.org. Mijn zonnepanelen leveren hier wel de data bij aan (ik zie netjes alles gevuld worden gedurende de afgelopen dagen). Ik heb de API en het stationID goed gekopieerd van PVoutput naar DSMR-reader, maar helaas krijg ik niks binnen...

dennissiemensma, 2019-08-06:
Bedankt voor je vraag. Begrijp ik het goed dat je bedoelt dat er vanuit PVOutput naar DSMR-reader geen gegevens binnenkomen?

Zo ja dan klopt dat. De koppeling is helaas alleen de andere kant op. DSMR-reader uploadt je stroomverbruik naar PVOutput. Gecombineerd met je gegevens uit je omvormer, kun je dan e.e.a. gecombineerd weergeven (ik heb zelf geen zonnepanelen).

lukevredeveld, 2019-08-06:
Aha :-) Dan heb ik dat dus verkeerd begrepen :-)
Dan nog even zoeken hoe ik mijn omvormer op DSMR-reader aangesloten krijg dan

dennissiemensma, 2019-08-06:
DSMR-reader leest alleen je slimme meter uit. Daarin kun je wat grafieken en historie zien.

Alle meetgegevens in DSMR-reader zijn echter (optioneel) te exporteren naar andere systemen. Bijvoorbeeld via MQTT voor domatica of via een ingebouwde 'plugin' voor PVOutput. 

De ondersteuning voor PVOutput is overigens beperkt tot het uploaden van je verbruik. Dit is wel het bruto verbruik, want een deel van je opbrengst tussen de omvormer en de slimme meter wordt doorgaans al opgenomen in het huishouden.

Je kunt je omvormer niet aansluiten op DSMR-reader. Je kunt wel zowel gegevens van je omvormer als gegevens uit DSMR-reader (automatisch) uploaden naar PVOutput. Dat eerste heb je dus al voor elkaar. 

Voor de gegevens uit DSMR-reader kun je het beste even de achtergrond lezen in tickets #9 en #467. Ik kan je namelijk inhoudelijk daar niet bij helpen, omdat ik zelf geen zonnepanelen (en dus een werkende omgeving) heb.

Dus:
```
          ------------>  PVOutput  <-----------
        /                                       \
Omvormer ---> Huishouden ---> Slimme meter (en DSMR-reader) ---> Teruglevering
```

lukevredeveld, 2019-08-06:
Thanks! Ik wacht even op mijn kabel en ga dan verder experimenteren!

dennissiemensma, 2019-09-07:
@lukevredeveld is dit nog gelukt?

dennissiemensma, 2019-09-10:
Ik sluit deze voor nu. Reageer gerust als je meer info hebt en/of hulp nodig hebt.

closed 2019-09-10 by dennissiemensma

lukevredeveld, 2019-09-10:
Ja ik had nog niet gereageerd. Ik heb het werkende gekregen maar later alles uitgezet. Het programma was nogal dorstig qua systeembronnen waardoor mijn domoticz niet goed meer reageerde. 

Met vriendelijke groet

======================
Luke Vredeveld

======================