dsmrreader/dsmr-reader #684: Fases niet zichtbaar

Flytezero, 2019-08-06:
De fases worden niet apart getoond zou dit een tekortkoming zijn van het V2 protocol ?
De instelling in de datalogger staan goed, op een ander adres is het me wel gelukt, maar wel een v4 meter.
Output van http://192.168.1.80/xhr/electricity?phases=True
{"errors": {"phases": ["multi_phases"]}}

De meter geeft aan een L1 L2 en L3 te hebben.

* DSMR-reader versie: V 2.2.1
* Hardware RaspberryPi 3b+
* DSMR-protocol: V2
* Meter: mt382-d2a52-m2k0agnz

dennissiemensma, 2019-08-06:
Bedankt voor je vraag. Voor zover ik weet zitten er inderdaad geen fases in het v2 protocol.

Je kunt nog eens dubbelchecken in het telegram dat je meter stuurt. Als het goed is kun je deze onderaan deze pagina inzien in de applicatie: `/admin/dsmr_datalogger/meterstatistics/1/change/`.

Je zou dan zoiets als hieronder moeten zien, maar ik denk dat deze regels ontbreken:
```
1-0:21.7.0(00.016*kW)
1-0:41.7.0(00.142*kW)
1-0:61.7.0(00.000*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
```

Flytezero, 2019-08-06:
Beste Dennis,

Dank je voor je snelle reply.
Het volgende zie ik op de pagina onderaan:

/ISk5\2MT382-1003

0-0:96.1.1(5A424244303035313333323136353133)
1-0:1.8.1(12224.771*kWh)
1-0:1.8.2(13983.472*kWh)
1-0:2.8.1(01630.192*kWh)
1-0:2.8.2(03280.209*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(0000.67*kW)
1-0:2.7.0(0000.00*kW)
0-0:17.0.0(0999.00*kW)
0-0:96.3.10(1)
0-0:96.13.1()
0-0:96.13.0()
!

dennissiemensma, 2019-08-06:
Het lijkt er inderdaad op dat er geen fase-informatie bijzit. Wellicht hopen dat ze je meter ooit vervangen voor een nieuwe? ; -)

Flytezero, 2019-08-06:
Dank je, dat zal nog een hele lange tijd op zich laten wachten vrees ik.

closed 2019-08-06 by Flytezero