dsmrreader/dsmr-reader #685: Maand data publiceren op MQTT

ThaStealth, 2019-08-12:
### Jouw omgeving
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.3.3`

### Bug, wens of iets anders
Kun je de maand data (die op de dashboard page onder het knopje "display current month" verstopt is) ook via MQTT naar buiten sturen?

dennissiemensma, 2019-08-12:
Bedankt voor je suggestie. Ik zet het op de backlog, maar het zal wel een tijd duren voordat ik hier aan toekom.

skank01, 2021-02-11:
+1 
Per jaar zou ook welkom zijn !!
Beetje gelijkgaand ook met https://github.com/dsmrreader/dsmr-reader/issues/989

dennissiemensma, 2021-02-13:
Dit issue is nu opgenomen in ##1291.

closed 2021-02-13 by dennissiemensma

dennissiemensma, 2021-03-07:
Dit gaat mee bij de aanstaande v4.13-release. Naast de gegevens via MQTT worden ze ook zichtbaar in het dashboard. Zie #1291 voor meer informatie en screenshots.