dsmrreader/dsmr-reader #686: Datum staat op 1 jan 2000

bvansluis, 2019-08-14:
* DSMR-reader versie: `v2.2.3`
* Hardware: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter: `v5`

### Bug, wens of iets anders
Ik heb DSMR Reader geinstalleerd en het werkt goed, alleen staat de datum op 1 jan 2000. De datum van mijn PI staat gewoon op 14-08-2019 en de grafiek van het weer staat ook gewoon goed. Enig idee hoe ik dit aan kan passen?

dennissiemensma, 2019-08-14:
Bedankt voor je vraag. DSMR-reader bepaalt zelf geen datums, maar gebruikt die van het systeem en de slimme meter.

Kun je aangeven waar/welke datum niet klopt?

bvansluis, 2019-08-14:
Hoi Dennis, dat kan ik zeker. Als ik op het dashboard kijk, dan staat daar 1 jan 2000:
[cid:image005.png@01D552E3.8C2B00E0]

En bij de statistics ook:
[cid:image006.png@01D552E3.8C2B00E0]

Kunnen het de database instellingen zijn? De PI datum instelling staat goed:
[cid:image007.png@01D552E3.8C2B00E0]

Mocht je nog meer info nodig hebben hoor ik het graag!

Bas

dennissiemensma, 2019-08-14:
Ik zie helaas geen screenshots. Kun je eens kijken wat er in de telegrammen staat? Je kunt ze inzien op: `/admin/dsmr_datalogger/meterstatistics/1/change/`

Ik ben vooral benieuwd naar de regels die beginnen met:
```
0-0:1.0.0(...

0-1:24.2.1(...
```

bvansluis, 2019-08-14:
Dit staat er:
1-3:0.2.8(50)
0-0:1.0.0(000101010000W)
0-0:96.1.1(4530303532303034343634373135343139)
1-0:1.8.1(000859.335*kWh)
1-0:1.8.2(000000.000*kWh)
1-0:2.8.1(000358.892*kWh)
1-0:2.8.2(000000.000*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(00.438*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00011)
0-0:96.7.9(00006)
1-0:99.97.0(4)(0-0:96.7.19)(000101010000W)(0000000320*s)(000101010000W)(0000000673*s)(000101010000W)(0000000474*s)(000101010000W)(0000001600*s)
1-0:32.32.0(00010)
1-0:52.32.0(00008)
1-0:72.32.0(00009)
1-0:32.36.0(00000)
1-0:52.36.0(00000)
1-0:72.36.0(00000)
0-0:96.13.0()
1-0:32.7.0(234.4*V)
1-0:52.7.0(233.9*V)
1-0:72.7.0(235.5*V)
1-0:31.7.0(001*A)
1-0:51.7.0(001*A)
1-0:71.7.0(002*A)
1-0:21.7.0(00.087*kW)
1-0:41.7.0(00.133*kW)
1-0:61.7.0(00.218*kW)
1-0:22.7.0(00.000*kW)
1-0:42.7.0(00.000*kW)
1-0:62.7.0(00.000*kW)
!E6FA

En bovenaan staat dit:
View DSMR Meter statistics (read only)
HISTORY

ID:
1
Timestamp:
Jan. 1, 2000, 1 a.m.
Timestamp indicating when the reading was taken

En dit zijn de screenshots:
[image]
[image]
[image]

dennissiemensma, 2019-08-14:
Ik ben bang dat de interne klok van je slimme meter niet goed staat. Je kunt dit het beste bij je netbeheerder aangeven. Mogelijk kunnen ze het op afstand herstellen.

De datumveld in de telegram is deze:

```
0-0:1.0.0(000101010000W)
```

Ter vergelijking, hij hoort zoiets te zijn:
```
# 2019-08-14-21:39:35-zomertijd
0-0:1.0.0(190814213935S)
```

bvansluis, 2019-08-14:
Ah, dat zou ook nog kunnen. Hij is namelijk ook niet op afstand uit te lezen. De monteur komt volgende week dus dan ga ik eens kijken of het daarna opgelost is. Als ik daarna alle historische gegevens wil wissen zodat ik weer schoon kan beginnen, hoe kan ik dat dan het beste doen?

dennissiemensma, 2019-08-14:
Ik vermoed dat dat het gaat oplossen dan. Ik heb dit veel vaker gehoord/gezien hier en uiteindelijk is het altijd opgelost door de monteur of op afstand.

Je kunt het beste de applicatie stoppen, de database verwijderen, opnieuw aanmaken. Wanneer de meter het weer doet, de applicatie starten.

```
# Stop app:
sudo supervisorctl stop all

# Verwijder DB en maak weer aan:
sudo sudo -u postgres dropdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader

# Maak structuur aan, zodat applicatie later weer gestart kan worden:
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/python3 /home/dsmr/dsmr-reader/manage.py migrate
```

Je kunt de meter na reparatie proberen uit te lezen met:
* https://dsmr-reader.readthedocs.io/en/latest/installation/explained.html#your-first-reading-optional

Als er een goede datum/tijd in staat kun je alles weer starten:
```
sudo supervisorctl start all
```

dennissiemensma, 2019-08-14:
Mocht bovenstaande niet werken dan hoor ik het graag alsnog van je. Je kunt hier dan gewoon weer een reactie achterlaten.

closed 2019-08-14 by dennissiemensma

bvansluis, 2019-08-20:
De nieuwe meter is geinstalleerd en ik krijg nu wel goede readings! Alleen kan ik niet meer inloggen opp de Django admin portal. Het wachtwoord van de dsmr user en de admin user pakt hij niet meer. Kan jij aangeven hoe ik deze kan resetten?

dennissiemensma, 2019-08-20:
Fijn om te horen! Je kunt je wachtwoord resetten met:
```
sudo su - dsmr
./manage.py changepassword admin
```