dsmrreader/dsmr-reader #688: Niet verwerkte telegrammen

jorkzijlstra, 2019-08-15:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.2.3 / v2.1.0`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `Rock Pi 64`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Bug, wens of iets anders
Ik lijk een hoop niet verwerkte telegrammen.

[image]

Enig idee hoe ik dit kan debuggen?

jorkzijlstra, 2019-08-15:
Ik heb de server herstart en ik zie dat er wel data verwerkt wordt.

Bij het uitvoeren van 
```
select *
from dsmr_datalogger_dsmrreading
where processed = false
order by id
limit 100
```

zie ik dat het oudste telegram daadwerkelijk telkens wordt verwerkt, maar de achterstand wordt niet ingehaald. Er komt dus telkens net zoveel telegrammen bij als dat er verwerkt worden

dennissiemensma, 2019-08-15:
Je kunt wellicht even debugging aanzetten en daarna in de log van dsmr_backend kijken of er iets mis is.

https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

dennissiemensma, 2019-08-15:
Aah oke, bedankt voor je toeoeging, dat is een wat lastiger verhaal.

Is er toevallig iets gewijzigd aan je systeem? Applicaties toegevoegd? Draai je DSMR-reader al langere tijd? (weken, maanden)

jorkzijlstra, 2019-08-15:
Debug level stond al aan.

In de stderr log staat er bijvoorbeeld
```
[2019-08-15 22:26:31,203] DEBUG     - Processed reading: 6815374 @ 2019-08-15 13:42:16+02:00 (0.062 kW)
[2019-08-15 22:26:33,292] DEBUG     - Processed reading: 6815375 @ 2019-08-15 13:42:18+02:00 (0.062 kW)
[2019-08-15 22:26:35,371] DEBUG     - Processed reading: 6815376 @ 2019-08-15 13:42:20+02:00 (0.061 kW)
[2019-08-15 22:26:37,465] DEBUG     - Processed reading: 6815377 @ 2019-08-15 13:42:22+02:00 (0.062 kW)
[2019-08-15 22:26:39,553] DEBUG     - Processed reading: 6815378 @ 2019-08-15 13:42:24+02:00 (0.062 kW)
```

de dsmr_backend-stdout blijft leeg.

De applicatie draaide al een tijd (sinds 16 juni) en heb vandaar net de server herstart. Voor de rest is er niks aan de server / applicatie aangepast.

De grafiek op het dashboard loopt ook totaan de tijd die in de processed logs / processed record in db staan.

dennissiemensma, 2019-08-15:
Bedankt voor de aanvulling. Ik doelde er meer op of je het project al lang gebruikt, dat had ik wat beter kunnen verwoorden, maar dat is al minimaal 2 maanden dus. Ik neem aan dat dit niet eerder is voorgekomen?

Je geeft aan een `Rock Pi 64` te gebruiken. Weet jij toevallig hoe die zich verhoudt tot de Pi-hardware? Ik heb alleen referentiepunten qua snelheid voor de Pi 2 en Pi 3.

----

Dingen die je nog kunt proberen zijn:
* In de [Supervisor-config (klik voor voorbeeld)](https://github.com/dennissiemensma/dsmr-reader/blob/v2.2.3/dsmrreader/provisioning/supervisor/dsmr-reader.conf) in `/etc/supervisor` de `nice` prio van `dsmr_backend` verlagen (`command=/usr/bin/nice -n 10`), bijvoorbeeld op `1` zetten of `nice` helemaal weghalen.

```
nice --help
Run COMMAND with an adjusted niceness, which affects process scheduling.
With no COMMAND, print the current niceness.  Niceness values range from
-20 (most favorable to the process) to 19 (least favorable to the process).
```

Vergeet niet `sudo supervisorctl reread` en `sudo supervisorctl reload` te doen.

----

* `DSMRREADER_DATALOGGER_SLEEP` verhogen zodat je iets minder metingen verwerkt (https://dsmr-reader.readthedocs.io/nl/latest/settings.html#dsmrreader-datalogger-sleep)

Hierdoor gaat je nauwkeurigheid wat achteruit, maar staat de datalogger wat minder te drukken qua datadoorvoer. Je kunt de sleep bijvoorbeeld op `3` of `5` of `9` zetten. Even mee experimenteren wellicht wat een goede balans is. Vergeet niet het process te herladen bij wijzigingen.

jorkzijlstra, 2019-08-15:
Hoi Dennis,

De `Rock Pi 64` is een zwaardere versie van de raspberry pi 4 en verbruikt ook wat meer stroom. Het is een iets zwaardere CPU en heb 4Gb geheugen.

Ik zit nu ook eens door te spitten door de code heen en zie dit staan:
`https://github.com/dennissiemensma/dsmr-reader/blob/20f90d4610c56346fe551f6c8c659e10803937a4/dsmr_datalogger/models/reading.py#L8`
Ik heb de retension nog nooit geset / cleanup aangezet en er zitten dus bijna 7 miljoen records in de dsmr_datalogger_dsmrreading tabel.

Nu ken ik Django niet genoeg maar zag de `get_queryset` en dit lijkt volgens Stackoverflow de gehele dataset binnen te halen en daarna te filteren?
Dit zou wel verklaren waarom het steeds maar langzamer en langzamer wordt, maar ik kan niet bepalen of Django ook daadwerkelijk op deze manier de query uitvoert.

Ik zag wel dat er 2 indexen liggen, 1 op processed en 1 op timestamp. In dit geval had ik zelf een combined index verwacht op beide velden. Ik neem even aan dat je de niet processed records wil hebben gesorteerd op de timestamp desc. 

Ik ga morgen eens met die supervisor settings aan de gang en houdt je op de hoogte.

dennissiemensma, 2019-08-16:
Bedankt voor de hardware informatie. 

Wat betreft de queries, het stuk code dat je aangeeft is een alias voor snelle filtering in de code. Alle querysets-verzoeken zijn achter elkaar te chainen en dat gebeurt ook bij het verwerken van de telegrammen. Hij pakt er telkens zo'n 1000 stuks per keer. 
De code die queryset gebruikt staat namelijk hier en zet er een limit op, die vertaald wordt naar het ORM: https://github.com/dennissiemensma/dsmr-reader/blob/v2.2.3/dsmr_consumption/services.py#L25

Ik hoor graag of bovenstaande suggesties iets verbeteren. Anders kan ik wel een branch voor je maken die wat meer debug informatie geeft.

jorkzijlstra, 2019-08-16:
Hoi Dennis,

Ik het de query log in Postgres aangezet en ben er achter gekomen wat het probleem is. Ik denk dat je de fix ook heel erg leuk gaat vinden gezien want de gehele UI is nu daadwerkelijk goed responsive geworden, als in alles laad onder de seconde.

In de query logs stond o.a. die select met 1024 row, maar ook de update en de insert. Het viel mij op dat de Updates met processed = true en de nieuwe Insert een beetje gelijk liepen. Vandaar ook dat het aantal verwerkte diagrammen niet naar beneden toe ging.

Zie 
[postgres.log](https://github.com/dennissiemensma/dsmr-reader/files/3510700/postgres.log)

Ik ben gewoon een voor een de query in een sql editor gaan uitvoeren en kijken welke er langzaam zou zijn.

Nu blijkt dat de met bijna 7 miljoen recods in dsmr_datalogger_dsmrreading de UPDATE ± 2,3 seconden duurt. 
```
UPDATE "dsmr_datalogger_dsmrreading" SET "processed" = true WHERE "dsmr_datalogger_dsmrreading"."id" = 6853436
```

Na wat verder kijken heb ik probleem getraceerd tot de tabel. Deze heeft geen primaire sleutel / index op de primaire sleutel.

Met het typen van dit comment heeft de applicatie alle onverwerkte telegrammen verwerkt. Dit waren er inmiddels al bijna 17000. Als bijzaak is de cpu load van postgres gedropt van 90% naar ±20%.

Benodigde fix (geen idee hoe je dit in Django kan doen): 
```
ALTER TABLE public.dsmr_datalogger_dsmrreading ADD CONSTRAINT dsmr_datalogger_dsmrreading_pk PRIMARY KEY (id);

CREATE UNIQUE INDEX dsmr_datalogger_dsmrreading_id_idx ON public.dsmr_datalogger_dsmrreading USING btree (id);
```

jorkzijlstra, 2019-08-16:
In mijn database zit dus nog alle data sinds 11-02-2019 en de gehele applicatie werkt weer zoals deze in het begin was. Het noodzakelijk opschonen van de data is dus niet nodig.

dennissiemensma, 2019-08-16:
Bedankt voor je uitgebreide onderzoek en de moeite om het terug te koppelen! Ik neem aan dat je heel zeker weet dat de PK index daar miste?

Ik heb het even gedubbelchecked en in mijn eigen database staat de index er wel. Het is sowieso vreemd, want Django maakt (voor zover ik weet) voor [alle PK's ook automatisch een index](https://github.com/dennissiemensma/dsmr-reader/blob/v2.2.3/dsmr_datalogger/migrations/0001_squashed_0005_optional_gas_readings.py#L39).

```
\dS dsmr_datalogger_dsmrreading

...

Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_9d090a40" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1_uniq" btree ("timestamp")
```

Wellicht goed om te dubbelchecken of je misschien een andere output krijgt.

Ook als ik een nieuw, leeg project in een lege database laad. De nieuwere versie van Django gebruikt iets andere namen, maar de indexes betreffen dezelfde velden als mijn 3-jaar oude DB:

```
Indexes:
    "dsmr_datalogger_dsmrreading_pkey" PRIMARY KEY, btree (id)
    "dsmr_datalogger_dsmrreading_processed_add8a852" btree (processed)
    "dsmr_datalogger_dsmrreading_timestamp_1f7020d1" btree ("timestamp")
```

Hoe dan ook ben ik blij dat je het bij jezelf gevonden hebt en dat je er ook nog eens een snellere applicatie van hebt gekregen!

jorkzijlstra, 2019-08-16:
Hoi Dennis,

Heel erg bedankt voor het meedenken.

closed 2019-08-16 by jorkzijlstra