dsmrreader/dsmr-reader #691: Webpagina niet meer bereikbaar

eterpstra, 2019-08-16:
Hoi Dennis,

Door een stroomstoring is mijn DSMR page niet meer te zien. Ik draai Ubuntu server en heb de nieuwste versie draaien .

Wat ik heb gezien is dat
- supervisor processen hebben de status running
- geupdate naar de nieuwste dsmr reader versie

Daarnaast heb ik MQTT ingeschakeld naar mijn domotica pakket (Home-assistant).

In home-asssistant zie ik keurig de data binnenkomen (Stroom verbruik, terugleveren etc)

Het lijkt er dus op dat er iets met de Gui / Webserver is ? Als ik intern er naar navigeer dan krijg ik 404 Not found met de Nginx naam erbij.

dennissiemensma, 2019-08-16:
Heb je toevallig al in de logfiles gekeken of daar iets staat?

`sudo view /var/log/supervisor/dsmr_webinterface.log`

eterpstra, 2019-08-16:
Hoi Dennis,

DIt geeft hij aan:
```

[2019-07-13 12:54:06 +0200] [32198] [INFO] Worker exiting (pid: 32198)
[2019-07-13 10:54:06 +0000] [2531] [INFO] Booting worker with pid: 2531
[2019-07-13 13:06:50 +0200] [2531] [INFO] Autorestarting worker after current request.
[2019-07-13 13:06:50 +0200] [2531] [INFO] Worker exiting (pid: 2531)
[2019-07-13 11:06:50 +0000] [5175] [INFO] Booting worker with pid: 5175
[2019-07-13 13:19:35 +0200] [5175] [INFO] Autorestarting worker after current request.
[2019-07-13 13:19:35 +0200] [5175] [INFO] Worker exiting (pid: 5175)
[2019-07-13 11:19:35 +0000] [7799] [INFO] Booting worker with pid: 7799
[2019-07-13 13:32:22 +0200] [7799] [INFO] Autorestarting worker after current request.
[2019-07-13 13:32:22 +0200] [7799] [INFO] Worker exiting (pid: 7799)
[2019-07-13 11:32:22 +0000] [10354] [INFO] Booting worker with pid: 10354
[2019-07-13 13:42:47 +0200] [10354] [INFO] Autorestarting worker after current request.
[2019-07-13 13:42:47 +0200] [10354] [INFO] Worker exiting (pid: 10354)
[2019-07-13 11:42:47 +0000] [12483] [INFO] Booting worker with pid: 12483
[2019-07-13 13:50:19 +0200] [12483] [INFO] Autorestarting worker after current request.
[2019-07-13 13:50:19 +0200] [12483] [INFO] Worker exiting (pid: 12483)
[2019-07-13 11:50:19 +0000] [14015] [INFO] Booting worker with pid: 14015
```

dennissiemensma, 2019-08-16:
Hmm dat zijn allemaal oude meldingen en zulke meldingen heb ik zelf ook.

Hoe ziet je Supervisor config er uit? Ik neem aan standaard en dat je zelfs niet hebt gewijzigd?

```
cat /etc/supervisor/conf.d/dsmr-reader.conf
```

dennissiemensma, 2019-08-16:
Ik neem aan dat je al een reboot of iets hebt geprobeerd. En anders kun je nog even alle processen herstarten:

```
sudo supervisorctl restart all
sudo systemctl restart nginx
```

dennissiemensma, 2019-08-16:
En als dat niet werkt, kun je nog even in de logs van Nginx kijken. Nginx is de webserver waar je browser tegen praat. Nginx praat vervolgens tegen dsmr_webinterface.

```
Browser ---> Nginx ---> dsmr_webinterface (via Supervisor)
```

Logs tailen en dan even in je browser proberen en zien wat hier uitkomt in de logs:
```
sudo tail -f /var/log/nginx/*.log
```

eterpstra, 2019-08-16:
dsmr reader supervisor config:

```
########################################
# The webinterface of the application  #
########################################

[program:dsmr_webinterface]
environment=PYTHONUNBUFFERED=1
command=/usr/bin/nice -n 15 /home/dsmr/.virtualenvs/dsmrreader/bin/gunicorn --timeout 60 --max-requests 500 --bind unix:/var/tmp/gunicorn--%(program_name)s.socket --pid /var/tmp/gunicorn--%(program_name)s.pid dsmrreader.wsgi
directory=/home/dsmr/dsmr-reader/
user=dsmr
group=dsmr
autostart=true
autorestart=true
startsecs=3
redirect_stderr=true
stdout_logfile=/var/log/supervisor/%(program_name)s.log
stdout_logfile_maxbytes=5MB
stdout_logfile_backups=5

```

sudo tail -f /var/log/nginx/*.log

```
==> /var/log/nginx/access.log <==
64.121.107.83 - - [16/Aug/2019:21:16:19 +0200] "GET /phpmyadmin2011/index.php?la                                                                                    ng=en HTTP/1.1" 404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebK                                                                                    it/537.36 (KHTML, like Gecko) Chrome/72.0.3626.119 Safari/537.36"
64.121.107.83 - - [16/Aug/2019:21:16:19 +0200] "GET /phpmyadmin2012/index.php?la                                                                                    ng=en HTTP/1.1" 404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebK                                                                                    it/537.36 (KHTML, like Gecko) Chrome/72.0.3626.119 Safari/537.36"
64.121.107.83 - - [16/Aug/2019:21:16:20 +0200] "GET /phpmyadmin2013/index.php?la                                                                                    ng=en HTTP/1.1" 404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebK                                                                                    it/537.36 (KHTML, like Gecko) Chrome/72.0.3626.119 Safari/537.36"
64.121.107.83 - - [16/Aug/2019:21:16:20 +0200] "GET /phpmyadmin2014/index.php?la                                                                                    ng=en HTTP/1.1" 404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebK                                                                                    it/537.36 (KHTML, like Gecko) Chrome/72.0.3626.119 Safari/537.36"
64.121.107.83 - - [16/Aug/2019:21:16:20 +0200] "GET /phpmyadmin2015/index.php?la                                                                                    ng=en HTTP/1.1" 404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebK                                                                                    it/537.36 (KHTML, like Gecko) Chrome/72.0.3626.119 Safari/537.36"
64.121.107.83 - - [16/Aug/2019:21:16:20 +0200] "GET /phpmyadmin2016/index.php?la                                                                                    ng=en HTTP/1.1" 404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebK                                                                                    it/537.36 (KHTML, like Gecko) Chrome/72.0.3626.119 Safari/537.36"
64.121.107.83 - - [16/Aug/2019:21:16:21 +0200] "GET /phpmyadmin2017/index.php?la                                                                                    ng=en HTTP/1.1" 404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebK                                                                                    it/537.36 (KHTML, like Gecko) Chrome/72.0.3626.119 Safari/537.36"
64.121.107.83 - - [16/Aug/2019:21:16:21 +0200] "GET /phpmyadmin2018/index.php?la                                                                                    ng=en HTTP/1.1" 404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebK                                                                                    it/537.36 (KHTML, like Gecko) Chrome/72.0.3626.119 Safari/537.36"
64.121.107.83 - - [16/Aug/2019:21:16:21 +0200] "GET /phpmyadmin2019/index.php?la                                                                                    ng=en HTTP/1.1" 404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebK                                                                                    it/537.36 (KHTML, like Gecko) Chrome/72.0.3626.119 Safari/537.36"
64.121.107.83 - - [16/Aug/2019:21:16:21 +0200] "GET /index.php?lang=en HTTP/1.1"                                                                                     404 580 "-" "Mozilla/5.0 (Windows NT 10.0; Win64; x64) AppleWebKit/537.36 (KHTM                                                                                    L, like Gecko) Chrome/72.0.3626.119 Safari/537.36"

==> /var/log/nginx/error.log <==
2019/08/16 06:50:39 [notice] 416#416: using inherited sockets from "8;9;"
```

Niet echt heel veel bijzonders lijkt mij?

En ja ik heb al meerdere malen een herstart gedaan ja. Erg vreemd.

dennissiemensma, 2019-08-16:
* Ik zie in je access logs geen requests naar DSMR-reader toe. Klopt dat? Wellicht kun je de browser even triggeren zodat te zien is of de logs wat doen.

* Heb je nog meer vhosts in Nginx, of alleen voor DSMR-reader?

eterpstra, 2019-08-17:
@dennissiemensma 

Hmmm nee ik heb niet echt een andere host draaien. Wel heb ik Home-assistant in docker draaien maar dat staat er los van denk ik. Het rare is dus wel dat hij draait want ik zie de MQTT berichten gewoon binnenkomen in HA.

Ik heb de browser getriggered om pagina te openen maar hij doet niks helaas.

Arnaud48, 2019-08-17:
Ik had/heb hetzelfde probleem, na een restartall gaat hij even verder, ook erg traag,  als ik dan kijk op de webinterface , die het dan even doet. Zie ik dat  de verwerking achter gaat blijven.
Na restartall zijn de metingen wel weer verwerkt.
sudo view /var/log/supervisor/dsmr_webinterface.log  
(Info)Worker exiting (pid 1595)
(Info)Booting worker with (pid 1605)
(Critical) Worker timeout (pid1605)
Daarna kan ik de website ook niet bekijken.

Dat hij traag is ,kan dat door dat de sd slecht aan het worden is?
Waar zit de fout dat het systeem stopt

Bedankt

dennissiemensma, 2019-08-17:
@eterpstra het project bestaat uit losse processen, dus dat MQTT werkt hoeft niet te betekenen dat de webinterface ook goed draait. Mocht je het niet kunnen vinden bij HTTPS, laat het even weten.

dennissiemensma, 2019-08-21:
@eterpstra heb je dit zelf nog kunnen oplossen?

eterpstra, 2019-08-23:
@dennissiemensma Ik heb het nog niet opgelost. Maar ik ga er wel uit komen denk ik. Het ticket mag dicht.

Thanks!

closed 2019-08-24 by dennissiemensma