dsmrreader/dsmr-reader #692: Fout bij lezen telegrammen na nieuwe slimmer meter (Enexis T210-D ESMR5)

erikdewildt, 2019-08-16:
### Jouw omgeving

Enexis T210-D ESMR5

* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.2.3`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Bug, wens of iets anders

Allereerst bedankt voor deze geweldige applicatie!

Ik heb vandaag een nieuwe slimme meter gekregen i.v.m. verzwaring aansluiting naar 3 fase. Met deze nieuwe meter lijken geen telegrammen meer van de P1 poort gelezen te kunnen worden.

Ik heb verder geen instelling o.i.d. gewijzigd.

In de logging zie ik deze foutmelding continu terugkomen:
`[2019-08-16 12:50:38,072] ERROR     [!] Exception raised in run(): month must be in 1..12`

De telegrammen lijken goed gelezen te kunnen worden vanaf de P1 poort, in de logging zie ik bijvoorbeeld:

> [2019-08-16 12:51:19,049] INFO
> /Ene5\T210-D ESMR5.0
> 
> 1-3:0.2.8(50)
> 0-0:1.0.0(190816125118S)
> 0-0:96.1.1(4530303438303030303335383536323139)
> 1-0:1.8.1(000001.200*kWh)
> 1-0:1.8.2(000001.766*kWh)
> 1-0:2.8.1(000000.092*kWh)
> 1-0:2.8.2(000000.000*kWh)
> 0-0:96.14.0(0002)
> 1-0:1.7.0(00.479*kW)
> 1-0:2.7.0(00.000*kW)
> 0-0:96.7.21(00007)
> 0-0:96.7.9(00003)
> 1-0:99.97.0(0)(0-0:96.7.19)
> 1-0:32.32.0(00002)
> 1-0:52.32.0(00002)
> 1-0:72.32.0(00002)
> 1-0:32.36.0(00000)
> 1-0:52.36.0(00000)
> 1-0:72.36.0(00000)
> 0-0:96.13.0()
> 1-0:32.7.0(241.0*V)
> 1-0:52.7.0(243.0*V)
> 1-0:72.7.0(242.0*V)
> 1-0:31.7.0(000*A)
> 1-0:51.7.0(000*A)
> 1-0:71.7.0(001*A)
> 1-0:21.7.0(00.186*kW)
> 1-0:41.7.0(00.012*kW)
> 1-0:61.7.0(00.281*kW)
> 1-0:22.7.0(00.000*kW)
> 1-0:42.7.0(00.000*kW)
> 1-0:62.7.0(00.000*kW)
> 0-1:24.1.0(003)
> 0-1:96.1.0(0000000000000000000000000000000000)
> 0-1:24.2.1(632525252525S)(00000.000)
> !162E
> 
> [2019-08-16 12:51:19,066] ERROR     [!] Exception raised in run(): month must be in 1..12
> [2019-08-16 12:51:19,067] DEBUG    Sleeping 0.5 sec(s)
> [2019-08-16 12:51:20,040] INFO
> /Ene5\T210-D ESMR5.0
> 
> 1-3:0.2.8(50)
> 0-0:1.0.0(190816125119S)
> 0-0:96.1.1(4530303438303030303335383536323139)
> 1-0:1.8.1(000001.200*kWh)
> 1-0:1.8.2(000001.766*kWh)
> 1-0:2.8.1(000000.092*kWh)
> 1-0:2.8.2(000000.000*kWh)
> 0-0:96.14.0(0002)
> 1-0:1.7.0(00.482*kW)
> 1-0:2.7.0(00.000*kW)
> 0-0:96.7.21(00007)
> 0-0:96.7.9(00003)
> 1-0:99.97.0(0)(0-0:96.7.19)
> 1-0:32.32.0(00002)
> 1-0:52.32.0(00002)
> 1-0:72.32.0(00002)
> 1-0:32.36.0(00000)
> 1-0:52.36.0(00000)
> 1-0:72.36.0(00000)
> 0-0:96.13.0()
> 1-0:32.7.0(241.0*V)
> 1-0:52.7.0(243.0*V)
> 1-0:72.7.0(241.0*V)
> 1-0:31.7.0(000*A)
> 1-0:51.7.0(000*A)
> 1-0:71.7.0(001*A)
> 1-0:21.7.0(00.186*kW)
> 1-0:41.7.0(00.012*kW)
> 1-0:61.7.0(00.283*kW)
> 1-0:22.7.0(00.000*kW)
> 1-0:42.7.0(00.000*kW)
> 1-0:62.7.0(00.000*kW)
> 0-1:24.1.0(003)
> 0-1:96.1.0(0000000000000000000000000000000000)
> 0-1:24.2.1(632525252525S)(00000.000)
> !4274
> 
> [2019-08-16 12:51:20,058] ERROR     [!] Exception raised in run(): month must be in 1..12
> [2019-08-16 12:51:20,060] DEBUG    Sleeping 0.5 sec(s)
> [2019-08-16 12:51:21,044] INFO
> /Ene5\T210-D ESMR5.0
> 
> 1-3:0.2.8(50)
> 0-0:1.0.0(190816125120S)
> 0-0:96.1.1(4530303438303030303335383536323139)
> 1-0:1.8.1(000001.200*kWh)
> 1-0:1.8.2(000001.766*kWh)
> 1-0:2.8.1(000000.092*kWh)
> 1-0:2.8.2(000000.000*kWh)
> 0-0:96.14.0(0002)
> 1-0:1.7.0(00.480*kW)
> 1-0:2.7.0(00.000*kW)
> 0-0:96.7.21(00007)
> 0-0:96.7.9(00003)
> 1-0:99.97.0(0)(0-0:96.7.19)
> 1-0:32.32.0(00002)
> 1-0:52.32.0(00002)
> 1-0:72.32.0(00002)
> 1-0:32.36.0(00000)
> 1-0:52.36.0(00000)
> 1-0:72.36.0(00000)
> 0-0:96.13.0()
> 1-0:32.7.0(240.0*V)
> 1-0:52.7.0(243.0*V)
> 1-0:72.7.0(242.0*V)
> 1-0:31.7.0(000*A)
> 1-0:51.7.0(000*A)
> 1-0:71.7.0(001*A)
> 1-0:21.7.0(00.185*kW)
> 1-0:41.7.0(00.012*kW)
> 1-0:61.7.0(00.281*kW)
> 1-0:22.7.0(00.000*kW)
> 1-0:42.7.0(00.000*kW)
> 1-0:62.7.0(00.000*kW)
> 0-1:24.1.0(003)
> 0-1:96.1.0(0000000000000000000000000000000000)
> 0-1:24.2.1(632525252525S)(00000.000)
> !43CF
> 
> [2019-08-16 12:51:21,062] ERROR     [!] Exception raised in run(): month must be in 1..12
> [2019-08-16 12:51:21,064] DEBUG    Sleeping 0.5 sec(s)

erikdewildt, 2019-08-16:
Na een aantal uren is het opeens gaan werken....

closed 2019-08-16 by erikdewildt

dennissiemensma, 2019-08-16:
Bedankt voor je terugkoppeling. Ik denk dat het lag aan dat de gasmeter zich nog niet goed geregistreerd had.

In je bericht staat namelijk:
```
0-1:96.1.0(0000000000000000000000000000000000)
0-1:24.2.1(632525252525S)(00000.000)
```

Dat is een leeg serienummer en op de tweede regel staat geen gasstand (dat kan), maar ook een datumtijd van `632525252525S`. 

De applicatie probeert dat te vertalen naar: `2063-25-25 25:25:25 zomertijd`. Daar klopt natuurlijk niets van en het is erg lelijk dat de slimme meter dat zo teruggeeft.

Kan het zijn dat het onderste deel van het telegram inmiddels wel goede waardes bevat?

erikdewildt, 2019-08-16:
Hoi Dennis,

Je hebt denk ik helemaal gelijk.

De telegrammen zien er nu beter uit, er staat nu inderdaad een juiste tijd in het 0-1:24.2.1 veld:

> /Ene5\T210-D ESMR5.0
> 
> 1-3:0.2.8(50)
> 0-0:1.0.0(190816193209S)
> 0-0:96.1.1(4530303438303030303335383536323139)
> 1-0:1.8.1(000001.200*kWh)
> 1-0:1.8.2(000005.901*kWh)
> 1-0:2.8.1(000000.092*kWh)
> 1-0:2.8.2(000000.000*kWh)
> 0-0:96.14.0(0002)
> 1-0:1.7.0(00.674*kW)
> 1-0:2.7.0(00.000*kW)
> 0-0:96.7.21(00007)
> 0-0:96.7.9(00003)
> 1-0:99.97.0(0)(0-0:96.7.19)
> 1-0:32.32.0(00002)
> 1-0:52.32.0(00002)
> 1-0:72.32.0(00002)
> 1-0:32.36.0(00000)
> 1-0:52.36.0(00000)
> 1-0:72.36.0(00000)
> 0-0:96.13.0()
> 1-0:32.7.0(231.0*V)
> 1-0:52.7.0(233.0*V)
> 1-0:72.7.0(233.0*V)
> 1-0:31.7.0(001*A)
> 1-0:51.7.0(000*A)
> 1-0:71.7.0(002*A)
> 1-0:21.7.0(00.228*kW)
> 1-0:41.7.0(00.011*kW)
> 1-0:61.7.0(00.434*kW)
> 1-0:22.7.0(00.000*kW)
> 1-0:42.7.0(00.000*kW)
> 1-0:62.7.0(00.000*kW)
> 0-1:24.1.0(003)
> 0-1:96.1.0(4730303538353330303039303137343137)
> 0-1:24.2.1(190816193000S)(01420.156*m3)
> !204D

Ik denk dat hier meer mensen tegen aan lopen na het vervangen van een slimme meter. Dus misschien kan deze case afgevangen worden naar een meer informatieve foutmelding.

Ik ben in ieder geval weer blij dat het werkt :-)

Prettig weekend!

dennissiemensma, 2019-08-16:
Bedankt voor je terugkoppeling! Mochten er veel mensen tegenaan lopen dan zal ik er zeker wat voor maken. Voor nu ben je (helaas) pas de eerste die het hier meldt. Jij ook goed weekend!

Woutch, 2019-09-06:
Hoi,

Ik heb sinds 4 dagen een digitale meter (België) en bijhorende gasmeter. Echter heb ik hetzelfde probleem als hierboven. Hieronder zijn de overeenkomende lijnen:

0-1:24.1.0(003)
0-1:96.1.1(37464C4F32313138303030383439)
0-1:24.4.0(0)
0-1:24.2.3(632525252525W)(00000.000)

Als ik een ander programma gebruik zoals P1 monitor verwerkt deze wel de telegrammen (buiten het gas gedeelte dan).

Vriendelijke groeten,
Wouter

dennissiemensma, 2019-09-06:
@Woutch bedankt voor je melding. Ik kan niet goed oordelen of je gasmeter niet goed aangesloten is of dat dit simpelweg in België anders is.

De applicatie kan wel een uitzondering krijgen op deze situaties, alleen is het wel zo dat dit afwijkt van de specificaties. Je slimme meter geeft nu een foute waarde door. Als er geen gasmeter is aangesloten, geeft die (in Nederland) uberhaupt geen data daarvan terug.

Weet je toevallig of je gasmeter ook is aangesloten op de slimme meter? Hij lijkt namelijk het tijdstip niet (goed) door te geven.

Woutch, 2019-09-06:
Hoi, 

Volgens mij is er iets mis met de connectie tussen beide. Ik heb ook een digitale gasmeter die draadloos connectie zou maken met de andere meter. Daar staat momenteel 3m³ ofzo op, dus zou wel iets zinnig moeten doorgeven.

Veel kan ik ook niet in de menu van de gasmeter doen, de configuratie delen zijn met een code afgeschermd :-)

Groetjes,
Wouter

dennissiemensma, 2019-09-06:
Ik denk dat je dan het beste contact kan opnemen met je netbeheerder. Grote kans dat zij namelijk ook niet je gasmeter op afstand kunnen uitlezen. 

Dit komt regelmatig voor, ook in Nederland. Soms kunnen ze het op afstand oplossen, soms moet er iemand voor langskomen, maar ik vind het lastig te zeggen wat er precies mis is.

Woutch, 2019-09-06:
Ik denk dat dit mss idd de beste oplossing is.

Ik zal ze eens contacteren volgende week of ze het kunnen nakijken.

Alvast bedankt,
Wouter

Woutch, 2019-09-15:
Ter info,

Doorgegeven naar netbeheerder en na een kleine 2 weken kwam de gas er ineens door. Remote kunnen oplossen dus en kan nu de software perfect gebruiken.

Groetjes,
Wouter

dennissiemensma, 2019-09-15:
Bedankt voor je terugkoppeling, fijn om te horen.