dsmrreader/dsmr-reader #694: Na maanden goed gewerkt te hebben ineens problemen.

dennissiemensma, 2019-08-17:
Na maanden goed gewerkt te hebben ineens problemen.
Ik had/heb hetzelfde probleem, na een restartall gaat hij even verder, ook erg traag,  als ik dan kijk op de webinterface , die het dan even doet. Zie ik dat  de verwerking achter gaat blijven.
Na restartall zijn de metingen wel weer verwerkt.
sudo view /var/log/supervisor/dsmr_webinterface.log  
(Info)Worker exiting (pid 1595)
(Info)Booting worker with (pid 1605)
(Critical) Worker timeout (pid1605)
Daarna kan ik de website ook niet bekijken.

Dat hij traag is ,kan dat door dat de sd slecht aan het worden is?
Waar zit de fout dat het systeem stopt

Bedankt

_Originally posted by @Arnaud48 in https://github.com/dennissiemensma/dsmr-reader/issues/691#issuecomment-522228289_

dennissiemensma, 2019-08-17:
@Arnaud48 ik heb wat meer informatie nodig. Welke hardware gebruik je? Heb je een DSMR v4 of een DSMR v5 slimme meter?

Arnaud48, 2019-08-17:
Systeem  Raspb 3b  ,Raspbian , Nginx 10.4. DSMR V4 ,landis en Gyr E350.
Systeem ontvangt nog steeds telegrammen, echter verwerkt deze niet.
Na poosje stopt het systeem,  en kan ik er niet meer bij .
Vandaag gezocht en gevonden om er weer bij te kunnen. 
Restart div processen.

dennissiemensma, 2019-08-17:
Oke dat zou verder goed moeten werken qua hardware.

Je kunt nog even kijken in de logs: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Arnaud48, 2019-08-17:
Dat alles zo traag ging lag aan de sd card, heb net kaart je geklooned en nu start alles snel op.
Echter het verwerken van de telegrammen gebeurt niet bij ongeveer 25 a 30 wordt het vak rood.

dennissiemensma, 2019-08-17:
Heb je nog bovenstaande debugging geprobeerd? In de log van de backend kun je misschien zien waar die dan exact zo lang mee bezig is.

Arnaud48, 2019-08-17:
In de log dsmr_backend.log staat alleen iets van vorige maand, van vandaag niks.

dennissiemensma, 2019-08-17:
Als het proces draait zou er wel iets in moeten komen. Weet je zeker dat het proces goed draait in supervisor?

Arnaud48, 2019-08-17:
Als ik "sudo supervisorctl restart all " uitvoer dan zie ik
 dsmr_mqtt: started
 dsmr_datalogger : started
 dsmr_backend : started
 dsmr_webinterface : started

 
 als die weer opgestart is dat alles is bijgewerkt, behalve dag en uur statistieken.

dennissiemensma, 2019-08-17:
De dag en uur statistieken pakt die 1x per dag op, ergens in de nacht. Je kunt dit proberen:

```
sudo su - dsmr
./manage.py dsmr_stats_regenerate_statistics
```

Je krijgt geen output te zien. Maar wellicht forceert dat de statistieken. En anders even een nachtje wachten om te zien of het zich vannacht oplost.

Arnaud48, 2019-08-18:
Goedemorgen,  de dag en uur statistieken zijn uitgevoerd. Daarna heeft hij alle telegrammen verwerkt.
Tot 05:03 en nu (09:32) staat de teller van onverwerkte telegrammen  op 800.
Nu zie ik dat in het menu bij  Api v2 docs  allemaal foutmeldingen
Server Error
Sorry, something unexpected happened.
Exception:
AttributeError: 'AutoSchema' object has no attribute 'get_link'
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/decorators/csrf.py", line 54, in wrapped_view
    return view_func(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/rest_framework/views.py", line 497, in dispatch
    response = self.handle_exception(exc)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/rest_framework/schemas/views.py", line 48, in handle_exception
    return super().handle_exception(exc)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/rest_framework/views.py", line 457, in handle_exception
    self.raise_uncaught_exception(exc)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/rest_framework/views.py", line 468, in raise_uncaught_exception
    raise exc

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/rest_framework/views.py", line 494, in dispatch
    response = handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/rest_framework/schemas/views.py", line 37, in get
    schema = self.schema_generator.get_schema(request, self.public)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/rest_framework/schemas/coreapi.py", line 164, in get_schema
    links = self.get_links(None if public else request)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/rest_framework/schemas/coreapi.py", line 151, in get_links
    link = view.schema.get_link(path, method, base_url=self.url) 

Heeft dat er misschien mee te maken?

>

dennissiemensma, 2019-08-18:
Bedankt voor je melding. Ik had dat zelf nog niet gezien, dat is gebeurd met de laatste upgrade. 

Het zijn echter alleen de API-docs, dus dat heeft verder geen invloed op de rest van het systeem. Kun je kijken of je nu wel recente informatie ziet in de dsmr_backend logs, met debugging aan?

Arnaud48, 2019-08-18:
Er staat iets in van check disk space deze sd is 32GB
2019-08-18 09:57:17,473] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4MTgwOTU3MjVTKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMDM0MzAzMDMxMzYzNzMwMzIzNDM3MzMzMTM1KQ0KMS0wOjEuOC4xKDAwNzY3My4wOTUqa1doKQ0KMS0wOjIuOC4xKDAwMTE1OS4IqQSkNCjEtMDoyMS43LjAoMDAuMTU5......KmtXKQ0KMS0wOjIyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITNGODcNCg=='
[2019-08-18 09:57:49,946] ERROR    @ apps | publish_json_dsmr_reading() failed: could not extend file "base/16385/16883.3": Geen ruimte meer over op apparaat
HINT:  Check free disk space.

[2019-08-18 09:57:50,407] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4MTgwOTU3MzVTKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMDM0MzAzMDMxMzYzNzMwMzIzNDM3M.......EzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITAzOTINCg=='
[2019-08-18 09:57:57,504] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4M......xMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KIUI2NjQNCg=='
[2019-08-18 09:58:07,555] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4MT..........xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITBEODENCg=='
[2019-08-18 09:58:17,574] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4MTgwOTU4MjVTKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMDM0MzAzMDMxMzYzNzMwMzIzNDM3Mz..........zMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITAzQ0QNCg=='
[2019-08-18 09:58:39,790] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4MTgwOTU4MzVTKQ0KMC0wOjk2LjEuMSg0NTMwMzAzMDM0MzAzMDMxMzYzNzMwMzIzNDM3M......i4wKDAwMDAwKQ0KMC0wOjk2LjEzLjEoKQ0KMC0wOjk2LjEzLjAoKQ0KMS0wOjMxLjcuMCgwMDIqQSkNCjEtMDoyMS43LjAoMDAuMTU0KmtXKQ0KMS0wOjIyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITJFODcNCg=='
[2019-08-18 09:58:47,813] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4M........QowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITU3QkINCg=='
[2019-08-18 09:59:28,894] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4M..........MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITk1MTcNCg=='
[2019-08-18 09:59:40,390] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4M..........EzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITI1QUYNCg=='
[2019-08-18 09:59:53,088] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4M...........xLjcuMCgwMDIqQSkNCjEtMDoyMS43LjAoMDAuMTQ2KmtXKQ0KMS0wOjIyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITA5NzENCg=='
[2019-08-18 10:00:46,736] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4M..................LTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITczOTANCg=='
[2019-08-18 10:01:08,825] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4M...........MzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITM4QjANCg=='
[2019-08-18 10:01:17,898] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4M...........wMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITc2QTINCg=='
[2019-08-18 10:01:27,972] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4M..........wMDIqQSkNCjEtMDoyMS43LjAoMDAuMTIwKmtXKQ0KMS0wOjIyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MDkwMDAwUykoMDQyMjguOTM4Km0zKQ0KITAxRDcNCg=='
[2019-08-18 10:02:43,598] INFO     @ services | Received telegram (base64 encoded): b'L1hNWDVMR0JCRkZCMjMxMjU0MjczDQoNCjEtMzowLjIuOCg0MikNCjAtMDoxLjAuMCgxOTA4MTgxMD................EtMDozMi4zNi4wKDAwMDAwKQ0KMC0wOjk2LjEzLjEoKQ0KMC0wOjk2LjEzLjAoKQ0KMS0wOjMxLjcuMCgwMDIqQSkNCjEtMDoyMS43LjAoMDAuMTEyKmtXKQ0KMS0wOjIyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MTAwMDAwUykoMDQyMjguOTM4Km [... cut ...] TEqcykoMTYwNzE1MTUwODAwUykoMDAwMDAwMTI1MipzKSgxNTA5MTYwOTE1MzFTKSgwMDAwMDAwMTgxKnMpDQoxLTA6MzIuMzIuMCgwMDAwMCkNCjEtMDozMi4zNi4wKDAwMDAwKQ0KMC0wOjk2LjEzLjEoKQ0KMC0wOjk2LjEzLjAoKQ0KMS0wOjMxLjcuMCgwMDIqQSkNCjEtMDoyMS43LjAoMDAuMTIwKmtXKQ0KMS0wOjIyLjcuMCgwMC4wMDAqa1cpDQowLTE6MjQuMS4wKDAwMykNCjAtMTo5Ni4xLjAoNDczMDMwMzMzMjM1MzYzMTMyMzMzMTMwMzEzMjMxMzEzNSkNCjAtMToyNC4yLjEoMTkwODE4MTAwMDAwUykoMDQyMjguOTM4Km0zKQ0KIUU1RUQNCg=='
[2019-08-18 10:03:16,409] INFO     @ services | Received telegram (base64 encod

Arnaud48, 2019-08-18:
Django.log    2019-07-15 23:29:34,436] ERROR    @ services | Discarded telegram with future timestamp: /XMX5LGBBFFB231254273

1-3:0.2.8(42)
0-0:1.0.0(190817104036S)
0-0:96.1.1(4530303034303031363730323437333135)
1-0:1.8.1(007667.018*kWh)
1-0:2.8.1(001157.014*kWh)
1-0:1.8.2(008571.477*kWh)
1-0:2.8.2(002508.499*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(02.254*kW)
1-0:2.7.0(00.000*kW)
0-0:96.7.21(00006)
0-0:96.7.9(00003)
1-0:99.97.0(3)(0-0:96.7.19)(190602193202S)(0000001211*s)(160715150800S)(0000001252*s)(150916091531S)(0000000181*s)
1-0:32.32.0(00000)
1-0:32.36.0(00000)
0-0:96.13.1()
0-0:96.13.0()
1-0:31.7.0(010*A)
1-0:21.7.0(02.254*kW)
1-0:22.7.0(00.000*kW)
0-1:24.1[2019-08-17 12:13:58,124] WARNING  @ log | Not Found: /favicon.ico
[2019-08-17 13:56:00,286] ERROR    @ log | Internal Server Error: /xhr/electricity
[2019-08-17 16:45:03,815] WARNING  @ log | Not Found: /favicon.ico
[2019-08-18 09:27:39,916] ERROR    @ log | Internal Server Error: /api/v2/docs/
[2019-08-18 09:28:54,858] ERROR    @ log | Internal Server Error: /api/v2/docs/

Arnaud48, 2019-08-18:
Geen nieuwe log in dsmr_backend.log

dennissiemensma, 2019-08-18:
Dat is een duidelijke foutmelding. En meestal zorgt dat voor hele vreemde fouten.

Kun je eens kijken wat dit aangeeft?

```
sudo df -h
```

Arnaud48, 2019-08-18:
pi@raspberrypi:~ $ sudo df -h
Bestandssysteem Grootte Gebruikt Besch Geb% Aangekoppeld op
/dev/root           15G      14G  9,9M 100% /
devtmpfs           460M        0  460M   0% /dev
tmpfs              464M      25M  440M   6% /dev/shm
tmpfs              464M      13M  452M   3% /run
tmpfs              5,0M     4,0K  5,0M   1% /run/lock
tmpfs              464M        0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1      44M      23M   22M  51% /boot
tmpfs               93M        0   93M   0% /run/user/1000

jeroenpeters1986, 2019-08-18:
In [dit comment](https://github.com/dennissiemensma/dsmr-reader/issues/694#issuecomment-522301894) staat het al eigenlijk:
`[2019-08-18 09:57:49,946] ERROR @ apps | publish_json_dsmr_reading() failed: could not extend file "base/16385/16883.3": Geen ruimte meer over op apparaat
HINT: Check free disk space.`

Je schijf is vol, je zult wat op moeten ruimen voordat Linux weer dingen kan ondernemen zoals nieuwe logs wegschrijven.

Arnaud48, 2019-08-18:
Vreemd, had gisteren een sd van 16gb, heb hem gekloond en weggeschreven naar 32gb, en dan nu al vol?

jeroenpeters1986, 2019-08-18:
Nou, doordat je hem hebt gekloond gebruikt je Pi nog steeds die 16GB, ook al staat deze nu op een fysiek grotere kaart.

Je moet eigenlijk je Pi nog laten weten dat je nu een grotere schijf hebt. Dat heet het expanden van je filesystem. Dat klinkt misschien moeilijk, maar is het niet.

Je kunt lezen hoe dat werkt op deze pagina: https://blog.jongallant.com/2017/11/raspberrypi-expand-file-system/

Arnaud48, 2019-08-18:
pi@raspberrypi:~ $ sudo df -h
Bestandssysteem Grootte Gebruikt Besch Geb% Aangekoppeld op
/dev/root           30G      14G   15G  50% /
devtmpfs           460M        0  460M   0% /dev
tmpfs              464M     4,0K  464M   1% /dev/shm
tmpfs              464M      13M  452M   3% /run
tmpfs              5,0M     4,0K  5,0M   1% /run/lock
tmpfs              464M        0  464M   0% /sys/fs/cgroup
/dev/mmcblk0p1      44M      23M   22M  51% /boot
tmpfs               93M        0   93M   0% /run/user/1000
pi@raspberrypi:~ $ sudo supervisorctl status
dsmr_backend                     RUNNING   pid 603, uptime 0:05:13
dsmr_datalogger                  RUNNING   pid 602, uptime 0:05:13
dsmr_mqtt                        RUNNING   pid 878, uptime 0:00:04
dsmr_webinterface                RUNNING   pid 601, uptime 0:05:13

Okee , het expanden is gelukt. 
Na opnieuw te hebben opgestart, zie ik bij status dat de telegrammen weer oplopen en niet verwerkt worden. 
Ik ga nu eerst een img maken , en dan weer verder zoeken. Vanmiddag weer verder zoeken.

jeroenpeters1986, 2019-08-18:
Fijn om te lezen dat dit overwonnen is!

Arnaud48, 2019-08-18:
!! Nu werkt het weer, geen idee wat er nu veranderd is, behoudens expanden sd schijf want daarna werkte het systeem nog niet , heb nog wat checks gdaan log files bekeken.
- Nu eerst img maken.
- Backups op andere sd /usb of synology proberen te krijgen.
- Uitzoeken welke bestanden weg mogen.
- Wat er ingevuld of nodig is voor MQTT te configureren.
Alle adviesen hulp is welkom.
Bedankt voor snelle reacties!!

dennissiemensma, 2019-08-18:
Werkt alles weer naar behoren nu? In je laatste reactie lees ik gemengde dingen namelijk.

Arnaud48, 2019-08-18:
Systeem werkt weer, maar wil voorkomen dat de sd card weer vol loopt.
Ik wil de telegrammen  bewaren op mijn synology ds115 en de backups. En de log.gz bestanden
Lukte me net nog niet.
Mqtt snap ik ook niet.

dennissiemensma, 2019-08-18:
* Voor back-ups kan ik je sowieso Dropbox aanraden, naast eventuele andere methodes.
* Synchronisatie naar je NAS kan ik je helaas niet mee helpen.
* Je kunt het beste eerst inlezen over MQTT: http://www.steves-internet-guide.com/mqtt/

Mocht je nog tegen fouten aanlopen, hoor ik het graag. :]

closed 2019-08-18 by dennissiemensma

Arnaud48, 2019-08-18:
Hartelijk dank voor de snelle reacties

Arnaud48, 2019-08-23:
Backup naar dropbox deed hij gelijk na het instellen, volgende dag fout melding.
Daar stond oa ook dropbox vol 3,45gb van de 2gb ( die zag ik even niet aankomen).
Dus zomaar ineens gestopt met werken klopt niet want het laatste wat ik ingesteld was ,backup maken .Dus raspberry was vol, geen ruimte meer om te werken.
Tot zover behoorlijk wat van opgestoken.
--Nu nog kijken hoe wat met backup. 
--En hoe opschonen zonder metingen te verliezen.

dennissiemensma, 2019-08-24:
Hoe groot zijn de DB-backups nu dan per stuk?

Arnaud48, 2019-08-24:
Dat was een .gz  bestand van 3,45 gb  met ook gegevens van maart dit jaar tot heden

dennissiemensma, 2019-08-24:
Wat krijg je hier te zien?

```
sudo su - postgres
psql dsmrreader

# Query
SELECT nspname || '.' || relname AS "relation",
    pg_size_pretty(pg_total_relation_size(C.oid)) AS "total_size"
  FROM pg_class C
  LEFT JOIN pg_namespace N ON (N.oid = C.relnamespace)
  WHERE nspname NOT IN ('pg_catalog', 'information_schema')
    AND C.relkind <> 'i'
    AND nspname !~ '^pg_toast'
  ORDER BY pg_total_relation_size(C.oid) DESC
  LIMIT 20;
```

Arnaud48, 2019-08-25:
relation                    | total_size 
------------------------------------------------+------------
 public.dsmr_mqtt_message                       | 4054 MB
 public.dsmr_consumption_electricityconsumption | 286 MB
 public.dsmr_datalogger_dsmrreading             | 271 MB
 public.dsmr_stats_hourstatistics               | 448 kB
 public.dsmr_consumption_gasconsumption         | 392 kB
 public.dsmr_weather_temperaturereading         | 368 kB
 public.dsmr_datalogger_meterstatistics         | 128 kB
 public.dsmr_stats_daystatistics                | 72 kB
 public.django_migrations                       | 64 kB
 public.auth_user                               | 64 kB
 public.django_admin_log                        | 64 kB
 public.django_session                          | 64 kB
 public.dsmr_backend_scheduledprocess           | 56 kB
 public.dsmr_weather_weathersettings            | 56 kB
 public.auth_permission                         | 56 kB
 public.dsmr_stats_electricitystatistics        | 56 kB
 public.django_content_type                     | 40 kB
 public.dsmr_stats_note                         | 40 kB
 public.dsmr_consumption_energysupplierprice    | 40 kB
 public.dsmr_backend_emailsettings              | 32 kB
(20 rows)

dennissiemensma, 2019-08-25:
Bedankt voor de informatie. Dat verklaart een hoop! 

Je hebt 4 GB aan MQTT-meldingen en dat is een bug in de applicatie, want zo is het nooit bedoeld.

Ik zou beginnen met het weggooien van al die meldingen:
```
sudo su - postgres
psql dsmrreader
truncate dsmr_mqtt_message;
```

dennissiemensma, 2019-08-25:
Daarnaast zul je het MQTT-proces moeten aanzetten, die alle berichten verstuurt: https://dsmr-reader.readthedocs.io/nl/latest/faq.html#how-do-i-retain-mqtt-support-when-upgrading-to-v1-23-0-or-higher

Arnaud48, 2019-08-25:
Van mij was het even kort antwoorden, moest nog telnet regelen voor deze pi, dat is nu gelukt. Nu kan de monitor weg uit de meterkast.

Ga ik uitvoeren, was aan het lezen mqtt  ga ik doorlezen.