dsmrreader/dsmr-reader #695: Tijdzone na installatie goedzetten

duiffie, 2019-08-17:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.2.3`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `VM (Ubuntu 18.04.3 LTS)`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v4`

### Bug, wens of iets anders
Na installatie van mijn VM ben ik vergeten de (systeem) tijdzone goed in te stellen, deze stond nog op Etc/UTC toen ik de DSMR-Reader installatie begon. Toen alles draaide kwam ik er achter dat tijden niet klopten, en heb ik de tijdzone alsnog aangepast. Nu doet alles het goed, behalve dat de gasmetingen niet juist zijn: deze loopt een uur achter. Data uit het telegram van 17:00u wordt in de grafiek getoond als data van 16:00u, en ik heb geen idee hoe ik dit nu recht moet trekken. Ik heb postgres ook al van de juiste tijdzone voorzien (in postgres.conf), maar dat mocht niet baten.

dennissiemensma, 2019-08-17:
Als het goed is converteert de datalogger alles naar de lokale tijdzone en daarna UTC. Kun je eens kijken wat er in de database staat?

```
sudo su - postgres

set timezone 'europe/amsterdam';
select read_at from dsmr_consumption_electricityconsumption order by read_at desc limit 5;
```

Klopt die output? Als in het huidige uur, relatief aan de tijdzone in het resultaat (hier `+02`).

dennissiemensma, 2019-08-17:
Idem voor deze, die is misschien wat makkelijker:

```
select timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;
```

duiffie, 2019-08-17:
Jep, die klopt:

`dsmrreader=# select read_at from dsmr_consumption_electricityconsumption order by read_at desc limit 5;
        read_at         
------------------------
 2019-08-17 18:22:00+02
 2019-08-17 18:21:00+02
 2019-08-17 18:20:00+02
 2019-08-17 18:19:00+02
 2019-08-17 18:18:00+02
(5 rows)
`

`dsmrreader=# select timestamp from dsmr_datalogger_dsmrreading order by timestamp desc limit 5;
       timestamp        
------------------------
 2019-08-17 18:22:46+02
 2019-08-17 18:22:36+02
 2019-08-17 18:22:26+02
 2019-08-17 18:22:16+02
 2019-08-17 18:22:06+02
(5 rows)
`

echter:
[image]

dennissiemensma, 2019-08-17:
Dat is fijn om te horen en nu zie ik wat je bedoelt. Ik heb in je initiele verhaal er overheen gelezen dat het over *gasmetingen* ging, sorry daarvoor!

In dat geval kan ik je vertellen dat dit werkt zoals bedoeld. Je moet de gasmetingen als volgt lezen:
* `17:00` betekent al het verbruik vanaf `17:00` tot `18:00`. 

Dit komt doordat de DSMR-v4 meters slechts 1x per uur de gasstand+tijdstip wijzigen. Ik moest kiezen om deze gegevens dan ofwel op het uur erna te plotten, ofwel het uur ervoor. Ik heb ooit voor het laatste gekozen.

duiffie, 2019-08-17:
Weet je dat zeker? Want ik heb nog een 2e instance draaien, en die laat nu dit zien (het is nu 19:27):

[image]

dennissiemensma, 2019-08-17:
Kan het zijn dat dat een DSMR-v5 meter is? DSMR-v4 heeft een nauwkeurigheid van 1x per uur. DSMR-v5 heeft elk kwartier (dacht ik). 

DSMR-v2 heeft geen gas, voor zover ik heb gehoord van anderen + voorbeelden tot nu toe (ik heb zelf alleen v4).

Kun je anders een telegram sturen van die niet-v4? Deze is onderaan deze pagina in te zien: `/admin/dsmr_datalogger/meterstatistics/1/change/`

De lange serienummers uit deze regels mag je weglaten voor je eigen privacy:
```
0-0:96.1.1( ...
0-1:96.1.0( ...
```

duiffie, 2019-08-17:
`0-0:96.1.1(################################)
1-0:1.8.1(14128.244*kWh)
1-0:1.8.2(07991.308*kWh)
1-0:2.8.1(00000.038*kWh)
1-0:2.8.2(00000.032*kWh)
0-0:96.14.0(0001)
1-0:1.7.0(0000.13*kW)
1-0:2.7.0(0000.00*kW)
0-0:17.0.0(0999.00*kW)
0-0:96.3.10(1)
0-0:96.13.1()
0-0:96.13.0()
0-1:24.1.0(3)
0-1:96.1.0(##################################)
0-1:24.3.0(190817190000)(00)(60)(1)(0-1:24.2.1)(m3)
(09689.892)
0-1:24.4.0(1)
!`

dennissiemensma, 2019-08-17:
Je hebt gelijk, er zit wel gas bij. Zit nota bene [in de testdata](https://github.com/dennissiemensma/dsmr-reader/blob/v2.2.3/dsmr_datalogger/tests/datalogger/test_iskra.py). Ik was in de war met een ander issue, over fasen, dat zit er niet in.

 Dan heb ik er nog geen verklaring voor. Zie je ook verschillen in de gastabel tussen de tijden van de instanties?

```
sudo su - postgres
psql dsmrreader

set timezone 'europe/amsterdam';
select read_at from dsmr_consumption_gasconsumption order by read_at desc limit 5;
```

Dat zou het in ieder geval visueel verklaren. Inhoudelijk moet ik dan in de code kijken wat de grenzen zijn.

dennissiemensma, 2019-08-17:
Aah, het zou kunnen zijn dat mijn versie-check nog aangepast mag worden. Ik zie dat ik speciaal voor v4 de tijd terugzet, omdat v5 toen nieuw was. Alleen heb ik nooit eerdere versies gecheckt.

https://github.com/dennissiemensma/dsmr-reader/blob/v2.2.3/dsmr_consumption/services.py#L151

duiffie, 2019-08-17:
`dsmrreader=# select read_at from dsmr_consumption_gasconsumption order by read_at desc limit 5;
        read_at         
------------------------
 2019-08-17 19:00:00+02
 2019-08-17 18:00:00+02
 2019-08-17 17:00:00+02
 2019-08-17 16:00:00+02
 2019-08-17 15:00:00+02
(5 rows)
`

Het enige nu merkbare verschil tussen de 2 is dat de v2 Postgres 11 heeft (gisteren geïnstalleerd op Rpi, Raspbian Buster Lite), en de v4 Postgres 10 (VM, Ubuntu).

dennissiemensma, 2019-08-17:
Ik denk dat het aan eerdergenoemde versiecheck ligt. Op een later moment kijk ik even aan wat ik er mee doe. 

Mocht het toch wijzigen voor v2, dan kun je het vrij makkelijk gelijktrekken naar de nieuwe situatie. Voor nu laat ik het even voor wat het is.

duiffie, 2019-08-17:
Duidelijk! Wat mij betreft zet je de tijd niet meer terug voor v4, ik vind het eerlijk gezegd ook wel zo logisch dat ik de laatste meting (tijdstip) in de grafiek kan zien, en de data van die laatste meting gelijk is aan wat ik in de grafiek zie. En volgens mij is dit ook nog de simpelste oplossing: 2 regeltjes code commenten en je bent er. Just my 2 cents... ;-)

dennissiemensma, 2019-08-17:
Ik begrijp dat het voor jou de voorkeur heeft, maar dat betekent ook een wijziging van 3 jaar geleden terugdraaien, net nu iedereen daar aan gewend is. ;-)

Ik denk dat de echte oplossing simpelweg in de interface zit. Als ik beter kan aangeven over welke tijdspanne het gaat, is er ook minder verwarring vermoed ik. Bijvoorbeeld: `17:00-18:00`.

dennissiemensma, 2019-09-10:
Ik heb hier #707 voor gemaakt.

closed 2019-09-10 by dennissiemensma