dsmrreader/dsmr-reader #697: Gas wordt niet verwerkt uit telegram bij digitale meters in België

floyson-reference, 2019-08-21:
DSMR-reader versie: v2.2.3
* Hardware: RaspberryPi 3
* DSMR-protocol versie van slimme meter v5

Bug:
Bij digitale meters in België wordt gas niet herkend, terwijl het wel in het telegram zit.
[image]

dennissiemensma, 2019-08-21:
Bedankt voor je melding! Heb je toevallig meer informatie over andere verschillen tussen de telegrammen?

dennissiemensma, 2019-08-21:
Heb je trouwens bovenstaand telegram in tekst-vorm? 

Je mag de twee serienummers wel weglaten. Dat zijn de regels die beginnen met:
```
0-0:96.1.1(
0-1:96.1.0(
```

floyson-reference, 2019-08-22:
Dag Dennis,

Telegram:

/FLU5\253769484_A

0-0:96.1.4(50213)
0-0:96.1.1(…)
0-0:1.0.0(190821210025S)
1-0:1.8.1(000260.129*kWh)
1-0:1.8.2(000338.681*kWh)
1-0:2.8.1(000000.010*kWh)
1-0:2.8.2(000000.425*kWh)
0-0:96.14.0(0002)
1-0:1.7.0(00.261*kW)
1-0:2.7.0(00.000*kW)
1-0:32.7.0(231.0*V)
1-0:52.7.0(000.0*V)
1-0:72.7.0(230.9*V)
1-0:31.7.0(000*A)
1-0:51.7.0(000*A)
1-0:71.7.0(000*A)
0-0:96.3.10(1)
0-0:17.0.0(999.9*kW)
1-0:31.4.0(999*A)
0-0:96.13.0()
0-1:24.1.0(003)
0-1:96.1.1(...)
0-1:24.4.0(1)
0-1:24.2.3(190821210011S)(00029.553*m3)
!BE13

Mvg
Frederic

dennissiemensma, 2019-08-22:
Bedankt! Ik heb er nu een generiekere oplossing voor gemaakt die ook werkt wanneer de gasmeter op bus 1,2 of 3 zit. Ik neem het mee voor de volgende release. Tot die tijd kun je gerust je eigen fix gebruiken.

closed 2019-08-22 by dennissiemensma

floyson-reference, 2019-08-22:
Ok dankjewel!

Mvg
Frederic Loyson

dennissiemensma, 2019-09-03:
Onderdeel van nieuwe release [v2.3](https://github.com/dennissiemensma/dsmr-reader/releases/tag/v2.3.0).

dennissiemensma, 2019-11-02:
In #750 heb ik dit anders opgelost. Zou moeten blijven werken gezien ik bovenstaand telegram als automatisch test heb gebruikt.