dsmrreader/dsmr-reader #700: [Question] Is it possible to update to a specific version?

janvanhelvoort, 2019-08-24:
Is it possible to update to a specific version, like the last version of 1.*.*?

Thank you, Jan

dennissiemensma, 2019-08-24:
Dat is zeker mogelijk. Voor elke release [maak ik een tag aan](https://github.com/dennissiemensma/dsmr-reader/tags). Je kunt die tag uitchecken met Git en dan blijft die daarop staan.

Ik weet trouwens niet zeker of je makkelijk kunt downgraden. Mocht het een nieuwe installatie of een installatie op een oude versie betreffen betreffen, dan is het sowieso geen probleem.

Je kunt wisselen met bijvoorbeeld: `git checkout tags/v1.28.0`

dennissiemensma, 2019-08-24:
Houd er wel rekening mee dat als je de installatie beschikbaar maakt op het Internet, er mogelijk security-lekken in kunnen zitten die in de nieuwere versies gefixt zijn. Als je dit alleen binnenshuis draait, is het minder een issue.

janvanhelvoort, 2019-08-25:
Het zijn 2 oude versies, 1.5.5 en 1.17.0. Moet ik na de checkout nog een script draaien? Bv './deploy.sh' 

Het systeem op versie 1.5.5 loopt qua verwerking zo’n 5000 telegrammen achter en verwerkt er ong 5 per minuut, zelfs als de data reader uit staat. Daarom wil ik een update proberen.

dennissiemensma, 2019-08-25:
Lees even goed de waarschuwingen door bij sommige versies: https://dsmr-reader.readthedocs.io/en/latest/changelog.html

Met name v1.23, 1.24 en 2.0. Na updaten (= tag uitchecken) hoef je alleen `post-deploy.sh` uit te voeren (ander script dus).

dennissiemensma, 2019-08-25:
Wat betreft de traagheid van metingen verwerken, welke hardware gebruik je?

janvanhelvoort, 2019-08-26:
Beide instanties draaien op een Raspberry pi 3 model B, beide begin 2017 geïnstalleerd. 

De pi met versie 1.5.5 hebben we gisteren naar 1.10.0 opgehoogd, tot nu toe lijkt hij alles goed bij te houden. Beide instanties kunnen de Configuratie `DATALOGGER - DSMR-uitlezing (alleen lezen)` niet opnemen, krijgen een `504 Gateway Time-out` foutmelding, kan zo 1-2-3 niet lezen of dat in een latere update opgelost is.

dennissiemensma, 2019-08-26:
Hmm, dat is niet een issue wat mij zo 1-2-3 iets zegt. iK kan je alleen aanraden om zo hoog mogelijk te upgraden. 

Als je voor elke upgrade een backup maakt van je database, kun je altijd makkelijk weer terug.

Kijk wel even goed uit of je:
* Vanaf `v1.23.0` MQTT uit hebt staan, tenzij je het dsmr_mqtt proces toevoegt zoals in de changelogs.
* Je minimaal Python 3.5 hebt als je naar `v2.0.0` of hoger gaat:
```
sudo su - dsmr
python -V
```

dennissiemensma, 2019-09-03:
Mocht je nog hulp nodig hebben, hoor ik het graag. Voor nu sluit ik dan deze.

closed 2019-09-03 by dennissiemensma