dsmrreader/dsmr-reader #701: NAS Backup

Erik050572, 2019-08-24:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.2.3`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3...`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

### Wens
Zou het mogelijk kunnen zijn dat er in een volgende versie een mogelijkheid komt om te backupen op een NAS?

dennissiemensma, 2019-08-24:
Ik ben er nog niet helemaal uit wat ik doe. Eigenlijk wil ik me het liefste focussen op wat de applicatie doet, want het is nu al aardig groot. En er zijn zoveel soorten NAS. 

Ik denk dat het wellicht makkelijker is dat je hier simpelweg zelf een cronjob voor maakt, met bijvoorbeeld rsync: https://originwww.synology.com/en-us/knowledgebase/DSM/tutorial/Backup/How_to_back_up_Linux_computer_to_Synology_NAS

dennissiemensma, 2019-09-10:
Ik ga dit sowieso niet in het project zelf integreren. Zie hierboven voor meer informatie. Mocht je er niet uitkomen dan kan ik je altijd nog wat verder op weg helpen.

closed 2019-09-10 by dennissiemensma