dsmrreader/dsmr-reader #703: API en Fases

Flytezero, 2019-08-26:
Op dit moment haal ik via de api /api/v2/consumption/electricity-live het huidige verbruik/gebruik op om in prtg te verwerken. Graag zou ik dit per fase op willen halen, is dit nu al mogelijk in de api of dien ik hier iets voor aan te passen ?

dennissiemensma, 2019-08-26:
Hi, je kunt dit volgens mij via de API doen. De lokale API-docs zijn momenteel stuk in de laatste release, maar je kunt ze altijd hier online vinden: https://dsmr-reader.readthedocs.io/en/latest/api.html#example-2-fetch-latest-reading

Als het goed is geven deze velden de fasen aan:
```
            "phase_currently_delivered_l1": "0.024",
            "phase_currently_delivered_l2": "0.054",
            "phase_currently_delivered_l3": "0.000",
            "phase_currently_returned_l1": "0.000",
            "phase_currently_returned_l2": "0.000",
            "phase_currently_returned_l3": "0.000",
```

Hiervoor moet je wel in de admin-interface instellen dat de fasen bijgehouden worden: https://dsmr-reader.readthedocs.io/en/latest/admin/datalogger.html

dennissiemensma, 2019-08-26:
Je kunt het ook met deze doen, hetzelfde verhaal: https://dsmr-reader.readthedocs.io/en/latest/api.html#get-consumption-electricity

Alleen worden deze records per minuut gegroepeerd, tenzij je dat expliciet uit hebt gezet.

De API-call in mijn eerdere comment pakt de informatie uit de metingen, dus die is dan wat nauwkeuriger. De applicatie houdt overigens geen kWh (verbruik( bij per fase, alleen kW (stroomvraag in Watt).

Flytezero, 2019-08-27:
Beste Dennis,

Dank je het is gelukt, de php code zal ik even delen mocht iemand anders de meter ook in prtg willen hebben.

`<?php

$api_token = 'jouw_api_token';
$ch = curl_init();

curl_setopt($ch, CURLOPT_URL, "http://dsmr.domein.nl/api/v2/datalogger/dsmrreading?ordering=-timestamp&limit=1");
curl_setopt($ch, CURLOPT_SSL_VERIFYPEER, false);
curl_setopt($ch, CURLOPT_HTTPHEADER, array('Authorization: Token ' . $api_token));
curl_setopt($ch, CURLOPT_RETURNTRANSFER, true);

$response = curl_exec($ch);
$result = json_decode($response);

curl_close($ch);

$l1 = str_replace(".", "",$result->results[0]->phase_currently_delivered_l1);
$l2 = str_replace(".", "",$result->results[0]->phase_currently_delivered_l2);
$l3 = str_replace(".", "",$result->results[0]->phase_currently_delivered_l3);

$data = [ 'prtg' => array ('result' => [
    array ('channel' => 'Stroom verbruik L1', 'value' => $l1),
    array ('channel' => 'Stroom verbruik L2', 'value' => $l2),
    array ('channel' => 'Stroom verbruik L3', 'value' => $l3),
]
)];

echo json_encode( $data );

?>`

dennissiemensma, 2019-08-27:
Fijn om te horen, bedankt voor je terugkoppeling!

closed 2019-08-27 by dennissiemensma