dsmrreader/dsmr-reader #704: Sdkaart crasht 1 a 2x per jaar, wat is wijsheid?

megapearl, 2019-09-08:
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `latest`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

De raspberry pi met postgresql database erop crasht ongeveer 1 a 2x per jaar waardoor ik alles weer opnieuw moet installeren en een backup moet terugzetten.
Wat is wijsheid? Ik heb een server met vmware draaien in de schuur die 30m van de meterkast geplaatst is waar de kabel in de P1 poort zit.

Is het handiger om dsmr-reader te installeren op 1 van de vm's en de p1 poort op afstand uit te lezen via netwerk (ser2net) of dsmr-reader weer op een raspberry pi 3 te installeren en enkel de postgresql database op een vm te zetten en de pi daarnaartoe laten connecten?
Wat zou de meest stabiele optie zijn?

Groeten,
Donald.

megapearl, 2019-09-08:
Ik zie net deze installatie:
https://dsmr-reader.readthedocs.io/nl/development/installation/datalogger.html#

Ga proberen de datalogger te installeren op een bestaande pi die al in de meterkast hangt, zag wel een klein foutje in de instructies:

`# Packages
sudo apt-get install -y supervisor python3 python3-pip python3-virtualenv virtualenvwrapper

# System user
sudo useradd dsmr --home-dir /home/dsmr --create-home --shell /bin/bash
sudo usermod -a -G dialout dsmr
sudo chown -R dsmr:dsmr /home/dsmr/

# Virtual env
sudo sudo -u dsmr mkdir /home/dsmr/.virtualenvs
sudo sudo -u dsmr virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3
sudo sh -c 'echo "source ~/.virtualenvs/dsmrreader/bin/activate" >> /home/dsmr/.bashrc'
sudo sh -c 'echo "cd ~/dsmr-reader" >> /home/dsmr/.bashrc'

# Requirements
# Packages
sudo apt-get install -y supervisor python3 python3-pip python3-virtualenv virtualenvwrapper

# System user
sudo useradd dsmr --home-dir /home/dsmr --create-home --shell /bin/bash
sudo usermod -a -G dialout dsmr
sudo chown -R dsmr:dsmr /home/dsmr/

# Virtual env
sudo sudo -u dsmr mkdir /home/dsmr/.virtualenvs
sudo sudo -u dsmr virtualenv /home/dsmr/.virtualenvs/dsmrreader --no-site-packages --python python3
sudo sh -c 'echo "source ~/.virtualenvs/dsmrreader/bin/activate" >> /home/dsmr/.bashrc'
sudo sh -c 'echo "cd ~/dsmr-reader" >> /home/dsmr/.bashrc'

# Requirements
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install pip install pyserial==3.2.1 requests==2.12.4
`

Volgens mij moet de laatste regel
`
# Requirements
sudo sudo -u dsmr /home/dsmr/.virtualenvs/dsmrreader/bin/pip3 install pyserial==3.2.1 requests==2.12.4
`
zijn.

dennissiemensma, 2019-09-08:
Bedankt voor je vraag:
> Is het handiger om dsmr-reader te installeren op 1 van de vm's en de p1 poort op afstand uit te lezen via netwerk (ser2net) of dsmr-reader weer op een raspberry pi 3 te installeren en enkel de postgresql database op een vm te zetten en de pi daarnaartoe laten connecten?
Wat zou de meest stabiele optie zijn?

Als het weinig moeite voor je is om `ser2net` werkend te krijgen, zou ik dat adviseren. DSMR-reader heb je dan volledig in je VM draaien, inclusief de database. De remote datalogger zou ook kunnen, net wat makkelijker is voor je.

dennissiemensma, 2019-09-08:
Wat betreft de remote datalogger, bedankt voor het vinden van die typo in de `development` versie van de documentatie. Ik heb het aangepast, maar ga het van de week nog uitproberen als dubbelcheck, want ik wacht nog een op nieuwe Pi.

megapearl, 2019-09-08:
Op zich werkt ser2net wel goed, alleen als de verbinding om wat voor reden ook verbreekt blijft die vaak verbroken totdat je ser2net met de hand stopt en start.
Ik heb nu de dsmr-reader volledig in de vm draaien en op de pi in de meterkast enkel de datalogger.
Nog tips om de database volledig terug te zetten?

dennissiemensma, 2019-09-08:
Wat betreft de database. Je zult die leeg moeten aanmaken en dan de backup terugzetten. Ik vermoed dat je in bovenstaande situatie misschien eerst al een `./manage.py migrate` hebt gedaan, waardoor ook al de (lege) tabellen aangemaakt worden.

Je kunt nu het best de applicatie helemaal stoppen, de DB verwijderen, opnieuw aanmaken en dan de backup inlezen. Je zou dan geen fouten moeten krijgen.

```
sudo supervisorctl stop all
sudo sudo -u postgres dropdb dsmrreader
sudo sudo -u postgres createdb -O dsmrreader dsmrreader

### Backup nu terugzetten, bijvoorbeeld:
zcat dsmrreader-postgresql-backup-Saturday.sql.gz | sudo sudo -u postgres psql dsmrreader

sudo supervisorctl start dsmr_webinterface

### Als je de oude data nu goed in de interface ziet, kan de rest ook aan.
sudo supervisorctl start all
```

megapearl, 2019-09-08:
That did the trick.

Database is volledig teruggezet en alles lijkt weer te werken als voorheen.
Ik heb de datalogger op de ubuntu vm uitgezet, en P1 poort uitlezen in de applicatie uitgezet.

Bedankt voor je supersnelle support!

closed 2019-09-08 by megapearl