dsmrreader/dsmr-reader #705: Data retention schoont niet op

allureGOT, 2019-09-10:
### Jouw omgeving
*(Indien van toepassing)*
* DSMR-reader versie *(rechtsbovenin applicatie zichtbaar)*: `v2.3.0`
* Hardware *(RaspberryPi 2, 3 of iets anders)*: `NUC i3`
* DSMR-protocol versie van slimme meter *(v2, v4 of v5)*: `v5`

Ik heb gezien dat mijn PostgresDB inmiddels (na twee en een half jaar) op 26GB grootte zit, ik wil hier de data retentie op toepassen, en heb deze op "discard most readings after one year" gezet.

Ik zie echter niets in de logging terug dat hij ook echt aan retentie doet, /var/log/supervisor/dsmr_backend.log heb ik nu meerdere dagen in de gaten gehouden.

Kan ik nog ergens anders zien dat de dataretentie wordt toegepast?

dennissiemensma, 2019-09-10:
Je kunt overigens deze instelling ook nog aanpassen: https://dsmr-reader.readthedocs.io/nl/latest/settings.html#dsmrreader-retention-until-this-hour

Want standaard schoont hij alleen op van middernacht tot 6 uur in de ochtend.

allureGOT, 2019-09-10:
Awh! Top, ik ging er vanuit dat het zowieso gelogd zou worden.
Ik zet m even op Debug om te zien of het dan wel langskomt.

dennissiemensma, 2019-09-10:
Je kunt het beste deze setting op 23 of 24 zetten, zodat die de hele dag doorgaat: https://dsmr-reader.readthedocs.io/nl/latest/settings.html#dsmrreader-retention-until-this-hour

Als je er performance issues van krijgt, kun je het weer weghalen (default die naar 6).

dennissiemensma, 2019-09-10:
Ik krijg trouwens geen mailtjes van comment edits, dus ik kan je aanraden om gewoon nieuwe comments te plaatsen hier. Dan krijg ik er sowieso een notificatie van. Ik zag je edit toevallig omdat ik nog een tab open had staan.

dennissiemensma, 2019-09-10:
Mocht de die de achterstand hebben weggewerkt, check dan even je DB-grootte, of die ook kleiner is geworden. Als dat niet zo is, kun je overwegen postgres dat te forceren (soort defragmenteren) met:

```
sudo su - postgres
vacuumdb -e -f -v -a
```

Dat kan even duren, afhankelijk van je hardware en resterende DB-grootte

allureGOT, 2019-09-11:
Na de eerste nacht opruimen, ik laat m in 1 run 168uur opruimen (1 week) wordt de retentie wel gelogd.
Ik heb de loggin weer uitgezet en zal elke week een "Vacuum" uitvoeren.

dennissiemensma, 2019-09-11:
Top, fijn om te horen! Er zit ook een auto-vacuum in postgres, maar ik weet dus niet hoe die omgaat wanneer je opeens je halve database omruimt, vandaar. 

Het kan dus zijn dat een handmatige, wekelijkse vacuum niet meer nodig is, zodra alle achterstand is weggewerkt. Mocht je toch nog issues hebben, dan hoor ik het graag.