dsmrreader/dsmr-reader #713: Sinds vannacht worden de inkomende telegrams niet meer verwerkt

Erik050572, 2019-09-17:
* DSMR-reader versie: `v2.3'
* Hardware: `RaspberryPi 3`
* DSMR-protocol versie van slimme meter: `v5`

### Bug
* Het programma wordt steeds langzamer:
De webpagina doet er steeds langer over om te laden en soms kan ik zelfs deze niet eens benaderen. Na een refresh van de webpagina wordt deze vervolgens wel geladen, maar dat gaat niet van harte.

[image]

En sinds vannacht komen wel de telegrams binnen, maar deze worden niet meer verwerkt.

[image]

Ik heb dit probleem in de vorige versie al 2 maal eerder gehad en toen heb ik het op nieuw geïnstalleerd. Buiten het feit dat ik de nog niet verwerkte telegrams ga verliezen, voel ik moet nog niet kundig genoeg om de backup terug te zetten.
(ik begrijp niet wat er in de handleiding bedoelt wordt met PostgreSQL & met MySQL: 
Welke zou ik moeten pakken als ik alleen de geschiedenis wil terugzetten)

Wat kan ik doen om dit probleem te verhelpen?

Alvast bedankt

Mvg,
Erik

dennissiemensma, 2019-09-18:
Bedankt voor je melding. Heb je toevallig al eerder uitgezocht wat de foutmeldingen waren destijds? En nu?

Zo nee, je kunt waarschijnlijk meer informatie vinden in de logfiles: https://dsmr-reader.readthedocs.io/nl/latest/troubleshooting.html

Ik hoor graag of je daar fouten ziet, vooral in `dsmr_backend`.

Erik050572, 2019-09-18:
Goedenavond Dennis,

Ik doe vast wat fout:

[image]

Erik050572, 2019-09-19:
Nieuwe ontwikkelingen..... 
Ik krijg nu deze fout als ik naar de website ga:

Server Error
Sorry, something unexpected happened.
Exception:
OperationalError: FATAL:  the database system is in recovery mode
FATAL:  the database system is in recovery mode
Traceback:
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/handlers/base.py", line 113, in _get_response
    response = wrapped_callback(request, *callback_args, **callback_kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 71, in view
    return self.dispatch(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 97, in dispatch
    return handler(request, *args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/views/generic/base.py", line 158, in get
    context = self.get_context_data(**kwargs)

  File "/home/dsmr/dsmr-reader/dsmr_frontend/views/status.py", line 12, in get_context_data
    context_data['status'] = dsmr_backend.services.backend.status_info()

  File "/home/dsmr/dsmr-reader/dsmr_backend/services/backend.py", line 176, in status_info
    capabilities = get_capabilities()

  File "/home/dsmr/dsmr-reader/dsmr_backend/services/backend.py", line 33, in get_capabilities
    'electricity': ElectricityConsumption.objects.exists(),

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/manager.py", line 82, in manager_method
    return getattr(self.get_queryset(), name)(*args, **kwargs)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/query.py", line 766, in exists
    return self.query.has_results(using=self.db)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/query.py", line 522, in has_results
    return compiler.has_results()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1070, in has_results
    return bool(self.execute_sql(SINGLE))

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/models/sql/compiler.py", line 1098, in execute_sql
    cursor = self.connection.cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 256, in cursor
    return self._cursor()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 233, in _cursor
    self.ensure_connection()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 217, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/utils.py", line 89, in __exit__
    raise dj_exc_value.with_traceback(traceback) from exc_value

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 217, in ensure_connection
    self.connect()

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/base/base.py", line 195, in connect
    self.connection = self.get_new_connection(conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/db/backends/postgresql/base.py", line 178, in get_new_connection
    connection = Database.connect(**conn_params)

  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/psycopg2/__init__.py", line 126, in connect
    conn = _connect(dsn, connection_factory=connection_factory, **kwasync)

dennissiemensma, 2019-09-19:
Kan het zijn dat je SD-kaartje corrupt is geraakt of iets? De melding `OperationalError: FATAL: the database system is in recovery mode` is in ieder geval geen goed teken.

Heb je al wel een backup kunnen veiligstellen?

Erik050572, 2019-09-20:
Ik heb half uur voor het fatale moment nog net een update gestuurd naar mijn dropbox.
Maar als ik nu een reinstall doe, ben ik dan ook alle telegrams kwijt van de afgelopen 3 dagen of is er een manier om die nog veilig te stellen?

dennissiemensma, 2019-09-20:
Het is wat lastig, maar niet onmogelijk. Je kunt beginnen met handmatig een backup te maken:

```
sudo su - dsmr
./manage.py dsmr_backup_create --full
```

In de output staat waar de backup is gemaakt. Zorg ervoor dat je die veiligstelt.

Daarna kun je een herinstallatie doen met de **oude** backup die je eerder had gemaakt. Als dat gelukt is hoor ik het graag. Je hebt dan wel tijdelijk een gat in je gegevens.

Erik050572, 2019-09-20:
Ik geloof dat het probleem zichzelf heeft opgelost:
ik heb een update gedaan naar versie 2.4.

Maar ik ga nog niet gelijk op de banken staan te juichen:
Want de Pi heeft er wat moeite mee, maar de algemene trend is dat de teller (van onverwerkte telegrammen) weer terugloopt.

Misschien kan je me nog helpen met het terugzetten van de 2 "oude" backups?

dennissiemensma, 2019-09-20:
Ik ben even in de war, waar heb je nu backups van? Als je je nieuwe installatie hebt ge-update heb je er nog maal eentje over toch? Degene met al je gegevens tot 3 dagen terug?

Heb je nog echt belang bij de metingen zelf of zijn de dag/uur-statistieken afdoende? Het is namelijk wat handmatig zoek en find-replace werk in de oude backup, want je kunt die niet zomaar importeren over een werkende installatie.

Erik050572, 2019-09-20:
Ik heb 2 keer eerder een dergelijk fout gehad.
Toen heb ik alles opnieuw geïnstalleerd en de back-up bewaard.
Ik heb deze nog nooit terug durven zetten omdat ik de uitleg in je werkinstructie niet begreep 
(ik ben een newby op het gebied van Linux) :)

dennissiemensma, 2019-09-20:
Hoewel je niet perse specifieke Linux kennis nodig hebt, zul je wel aardig diep moeten graven in de data.

Als je de backups uitpakt en opent met een text-editor, zul je op zoek moeten naar de metingen. Je kunt deze vinden op de term: `COPY public.dsmr_datalogger_dsmrreading`.

Als het goed is zie je daar op elke regel een meting staan. Bijvoorbeeld:
```
...

COPY public.dsmr_datalogger_dsmrreading (id, "timestamp", electricity_delivered_1, electricity_returned_1, electricity_delivered_2, electricity_returned_2, electricity_currently_delivered, electricity_currently_returned, extra_device_timestamp, extra_device_delivered, processed, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3) FROM stdin;

6745918 2018-02-11 06:59:53+01  2528.308        0.000   2501.346        0.000   0.051   0.000   2018-02-11 06:00:00+01  2339.089        t       0.017   0.033   0.000   \N      \N      \N

7310290 2018-04-30 16:47:26+02  2707.007        0.000   2736.528        0.000   0.132   0.000   2018-04-30 16:00:00+02  2525.584        t       0.098   0.033   0.000   \N      \N      \N
7

...

COPY public.dsmr_datalogger_meterstatistics

...
```

Het probleem is dat je:
* Alleen de metingen wilt hebben (in mij geval is alles tot aan het volgende blok `COPY public.dsmr_datalogger_meterstatistics`).
* Dit velen duizenden metingen kunnen zijn, afhankelijk van hoeveel je er had opgeslagen.
* En het eerste veld van elke meting (`id`) wilt weghalen, want die gaat garandeerd botsen met je nieuwe installatie.

Wat je ook doet. Zorg, vanzelfsprekend, dat je dit altijd probeert op een kopie van de backup. Als je het blok met metingen apart hebt gezet, kun je vrij makkelijk een find replace doen van de ID's. Ik kan daar ook eventueel wel een regex van voor je maken.

----

Mocht je dit allemaal te lastig vinden, dan kan ik je aanraden om de backup terug te zetten waarin de meeste data zit.

Erik050572, 2019-09-24:
> sudo su - dsmr
> ./manage.py dsmr_backup_create --full

> Het is wat lastig, maar niet onmogelijk. Je kunt beginnen met handmatig een backup te maken:
>
> ```
> sudo su - dsmr
> ./manage.py dsmr_backup_create --full
> ```
>
> In de output staat waar de backup is [...]

Ik moet zeker eerst het process stop zetten, want ik krijg de volgende foutmelding:

(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $ ./manage.py dsmr_backup_create --full
pg_dump: WARNING:  terminating connection because of crash of another server process
DETAIL:  The postmaster has commanded this server process to roll back the current transaction and exit, because another server process exited abnormally and possibly corrupted shared memory.
HINT:  In a moment you should be able to reconnect to the database and repeat your command.
pg_dump: Dumping the contents of table "dsmr_consumption_electricityconsumption" failed: PQgetCopyData() failed.
pg_dump: Error message from server: SSL SYSCALL error: EOF detected
pg_dump: The command was: COPY public.dsmr_consumption_electricityconsumption (id, read_at, delivered_1, returned_1, delivered_2, returned_2, currently_delivered, currently_returned, phase_currently_delivered_l1, phase_currently_delivered_l2, phase_currently_delivered_l3, phase_currently_returned_l1, phase_currently_returned_l2, phase_currently_returned_l3) TO stdout;
Traceback (most recent call last):
  File "./manage.py", line 10, in <module>
    execute_from_command_line(sys.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 381, in execute_from_command_line
    utility.execute()
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/__init__.py", line 375, in execute
    self.fetch_command(subcommand).run_from_argv(self.argv)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 323, in run_from_argv
    self.execute(*args, **cmd_options)
  File "/home/dsmr/.virtualenvs/dsmrreader/lib/python3.5/site-packages/django/core/management/base.py", line 364, in execute
    output = self.handle(*args, **options)
  File "/home/dsmr/dsmr-reader/dsmr_backup/management/commands/dsmr_backup_create.py", line 37, in handle
    folder=base_folder
  File "/home/dsmr/dsmr-reader/dsmr_backup/services/backup.py", line 134, in create_full
    on_backup_failed(process_handle=backup_process)
  File "/home/dsmr/dsmr-reader/dsmr_backup/services/backup.py", line 184, in on_backup_failed
    error_message = process_handle.stderr.read()
AttributeError: 'NoneType' object has no attribute 'read'
(dsmrreader) dsmr@raspberrypi:~/dsmr-reader $

dennissiemensma, 2019-09-25:
Je hoort geen foutmelding te krijgen. Krijg je deze elke keer?

Erik050572, 2019-09-25:
Helaas wel

dennissiemensma, 2019-09-26:
Je kunt dan nog proberen om de applicatie te stoppen en dan de backup te maken, maar eigenlijk belooft de foutmelding zelf ook niet veels goed.

```
sudo supervisorctl stop all

sudo su - dsmr
./manage.py dsmr_backup_create --full
CTRL+D

sudo supervisorctl start all
```

Erik050572, 2019-09-27:
Hoi Dennis,

Ik heb gisteren geprobeerd om een drop-box back-up terug te zetten op een andere SD-kaart.
Het gehele proces verliep eigen vrij simpel & de teruggezette data is ook weer te zien in het systeem.
Echter daarna kwam er geen nieuwe data binnen, terwijl het systeem wel aangaf dat alle systemen liepen. 

Ik ben gelukkig een nieuwe gebruiker van jouw software dus ik had niet zoveel data om te verliezen (zo'n 2 maanden). Dus ik heb het wederom opnieuw geïnstalleerd en nu met een lege database begonnen.

Ik hoop nu dat het probleem verholpen is en zal dit issue dan ook sluiten.

N.B.: Wat mij opviel was dat de database helemaal overschreven werd en dus de nieuwe data verloren gaat. Persoonlijk zou ik het handiger vinden als een back-up geïmporteerd zou worden in de huidige data in plaats van te worden overschreven. In mijn geval met meerdere halve back-ups zou dat ook beter uitkomen. 

Ik vind dat je een mooi product hebt neergezet en zal het ook bij meerdere personen gaan aanraden.

Met vriendelijke groet,
Erik

closed 2019-09-27 by Erik050572